edk2-porting/edk2-rk3588 #3: only see 7.8G ram on 16G ram variant.

microcai, 2023-01-23:
free outputs 
```
               total        used        free      shared  buff/cache   available
Mem:         7861584      170156     7503528        9664      187900     7466300
Swap:              0           0           0
```

with uboot, free output
```
               total        used        free      shared  buff/cache   available
Mem:        16104576     8322160     2042348       18720     5740068     7603336
Swap:              0           0           0

```

Marcinoo97, 2023-03-09:
check if changing 8192 to 16384 in PlatformSmbiosDxe.c file and edit rock5b.dsc file or something to replace 0x200000000 with 0x400000000 will do anything.

Googulator, 2023-03-17:
Known issue, memory size is hardcoded at compile time for now. Autodetection of installed memory is possible, but not implemented yet.

Googulator, 2023-03-23:
Should be fixed by https://github.com/edk2-porting/edk2-rk35xx/commit/2b04205689368f0dc7954bc062e10d9fc5b581d1 (not part of release v0.4, will be in next release).

closed 2023-03-24 by Googulator