edk2-porting/edk2-rk3588 #6: Cant boot ventoy udisk

amazingfate, 2023-03-16:
Got error output:
`Synchronous Exception at 0x0000000E2648D28`

mariobalanica, 2023-04-20:
Is this still an issue? It may have been related to simple-init.

NN708, 2023-04-21:
@mariobalanica I can confirm that this issue still exists with v0.6 on rock 5B.

amazingfate, 2023-04-21:
The reason I use [ventoy](https://www.ventoy.net/en/index.html) is that I can store multi os isos on one single udisk, and choose one to boot with out flashing the whole udiisk. I can boot ventoy on rk3568 with https://github.com/jaredmcneill/quartz64_uefi/.

mariobalanica, 2023-06-22:
Yeah, no idea what's going on. Someone else reported what looks to be the same issue, but on a different platform: https://github.com/ventoy/Ventoy/issues/1658

It's the only bootloader that does this and since the author claims that "ARM64 platform is hardware dependent", it might be better to ask them about it. Maybe it has some special requirements?

amazingfate, 2023-07-12:
Current ventoy release is using grub 2.04 and edk2 201911, so I tried to rebuild ventoy with grub 2.06 and edk2 202305. But I still get `Synchronous Exception`.
So this should be an cpu related issue.