edk2-porting/edk2-rk3588 #7: SD card preloaded with Windows 11 not visible to UEFI

Googulator, 2023-03-17:
I prepared an SD card with an installation of Windows 11 22H2 for ARM64 (GPT partitioning, bootmgfw.efi installed on FAT32 EFIESP).
When plugged in directly to the Rock 5B's SD card slot, the card is not detected by UEFI (not offered for boot, not browsable in UEFI shell). The same SD card plugged into the bottom USB 3.0 port through an adapter boots fine (the black USB 2.0 ports also detect the card, but boot fails with INACCESSIBLE_BOOT_DEVICE, probably due to an ACPI configuration issue),

Checking the code, it looks like pin muxing is never set up correctly to route the SDHCI controller to the SD socket.

champlainmorin, 2023-03-18:
I've had the same issue with this, as it doesn't seem like the SD card slot is recognized anywhere. I don't have much to contribute at the moment, but I'd like to bump this issue if that's a thing that happens on Github.

Googulator, 2023-04-20:
No, it's not - that commit only enables UEFI itself to see the card; it's still not presented via ACPI in any meaningful manner.

mariobalanica, 2023-04-20:
Right, but the original issue was "card not visible in UEFI". After that it's the job of the OS. There are no ACPI bindings in the Linux driver and no Windows drivers at all.

Googulator, 2023-04-20:
Looks like I misremembered - it's eMMC, not SD, which is missing ACPI representation (eMMC is PRP0001 right now)

closed 2023-04-20 by Googulator