edk2-porting/edk2-rk3588 #8: Linux 6.2 (pancsf-midstream) booted via this UEFI + matching DT bootloops on USB-PD

Googulator, 2023-03-17:
My most recent build of the pancsf-midstream kernel goes into a bootloop when loaded by UEFI, using the same DTB and boot options that work well with the stock u-boot.

The cause of the bootloop is a USB-PD hard reset received when the FUSB driver gets loaded. This issue is reproducible with all of my PD power supplies, including ones that work with u-boot.

This could simply be a timeout (as UEFI is much slower than u-boot at handing off to the kernel, by design), however the UEFI I2C code may also be writing to the FUSB and mangle its state before handoff.

strongtz, 2023-03-17:
To be honest I'm surprised that you can even boot upstream Linux with fusb302 driver, to me it has been an issue since playing with rk3399 board. I'm not sure if the problem has been fixed though. 
And I think there's nothing about FUSB in UEFI currently.
https://yhbt.net/lore/all/0ac6bbe7-6395-526d-213c-ac58a19d8673@fivetechno.de/t/#mca7df66b1a7b5c4aca139aeb8415ac4e442238c7

Googulator, 2023-03-17:
Midstream has some patches to the TCPM engine - by default, TCPM just straight up orders a simulated disconnect cycle on any new port it enumerates, with the assumption that the Type-C port is only a charging source for a battery. If you do that on a batteryless system, it's instant suicide. This is patched in midstream; port initialization now keeps the correct Rp/Rd resistance connected depending on the requested default port role.

mariobalanica, 2023-06-22:
> however the UEFI I2C code may also be writing to the FUSB and mangle its state before handoff.

Does the issue occur with the I2C driver disabled?