edk2-porting/edk2-rk3588 #13: Synchronous Exception at 0x00000000E20CDBC4

divaldo95, 2023-03-26:
On my most recent build and on older builds/releases when an USB storage (tried with USB card reader and pendrive) is inserted and an NVME drive is present, after a few seconds, most of the time at 3 seconds left out of 10 it throws 'Synchronous Exception at 0x00000000E20CDBC4' message. Also happens if I press F3 or select Simple Init from boot manager menu. 
I can reproduce the issue without an USB storage if I boot to Simple Init and wait 7 seconds. 
### Logs
[Boot without any interrupt when USB inserted](https://github.com/edk2-porting/edk2-rk35xx/files/11073089/edk2_r5b_pendrive.txt)
[Simple Init from boot manager menu without USB inserted](https://github.com/edk2-porting/edk2-rk35xx/files/11073081/edk2_r5b_wp_simpleinit.txt)

mariobalanica, 2023-03-26:
I can repro this.

mariobalanica, 2023-03-29:
simple-init is disabled by default now.

closed 2023-03-29 by mariobalanica