edk2-porting/edk2-rk3588 #19: SD Card Boot Error with Armbian EFI Image

champlainmorin, 2023-04-04:
Hello, after booting into the image once, I get an error that the filesystem cannot be mounted, and then after I try to reboot again into this image on my SD card, I get the following synchronous exception:

```bash
conf: no logfs or esp found
conf: loaded uefi://[VenHw(8500EF02-317D-11EC-95B4-F42A7DCB925D) /HD(1,GPT,478B363E-77D5-4641-9B94-A6C419A5A65C,0x2000,0x80000)]/uefi:/[VenHw(8500EF02-317D-11EC-95B4-F42A7DCB925D) /HD(1,GPT,478B363E-77D5-4641-9B94-A6C419A5A65C,0x2000,0x80000)]/ as default config path

Synchronous Exception at 0x00000000E2112F4
```

Any ideas would be greatly appreciated, or any pointers to any confirmed working Debian'ish images for this would be most useful.

Also, the other error I got was something like this:

```bash
Begin: Running /scripts/local-block ... done
```
For like twenty or so attempts and then the following line:

```bash
Gave up waiting for root file system device.
```
and
```bash
ALERT! UUID=9231d7a4-f805-4ec2-93ff-7110f09c6dc3 does not exist. Dropping to a shell!
```
and it opens an (initramfs) shell.

mariobalanica, 2023-04-04:
Please try the latest build artifact from here: https://github.com/edk2-porting/edk2-rk35xx/actions

There will be a new release with this fixed soon.

champlainmorin, 2023-04-05:
Thanks, this fixed the first issue I was having, but the other (initramfs) issue is still happening when I try to boot the Armbian image. Any other ideas or known working Debian based distros?

closed 2023-06-22 by mariobalanica