edk2-porting/edk2-rk3588 #26: Orange Pi5 only usb2 vertical port working

humsupboi20, 2023-06-30:
Orange Pi5 only usb2 vertical port working, If I plug anything into usb c port, there will be a "bugcode_usb_driver" error.

mariobalanica, 2023-06-30:
In what OS does this happen?

Note that the side-mounted USB-A 2.0 port shares the same pins with the
USB-C data port, connecting devices to both at the same time is not going
to work.

humsupboi20, 2023-07-01:
Windows 11, wor project. Even if i dont use the usb 2.0 port, if i plug in usb c it will crash and result to the error above. The 2 usb A 3.0 ports are not working at all...

mariobalanica, 2023-07-01:
What USB device are you trying to plug in?

USB 2.0 devices will only work on the USB-C port or the side-mounted USB-A.

The top USB 3.0 port only supports 3.0 devices.