edk2-porting/edk2-rk3588 #27: Any way to pass devicetree name to grub-efi?

amazingfate, 2023-07-07:
I'm working on uefi bootable iso. Now I can boot one with manually set command in grub like `devicetree /live/dtb/rockchip/rk3588-rock-5b.dtb`. I want to build an iso for all rk3588 boards, which will get devicetree name from efi vars with grub command like [efivar_get](https://www.mail-archive.com/grub-devel@gnu.org/msg30141.html).

mariobalanica, 2023-07-07:
Nice!
So, I've actually been thinking of bundling the board DTBs with the UEFI images. I'm not too familiar with this however.
Would including (for instance) the DTB from Radxa's kernel allow Armbian to work properly, or does it patch it in any way? The same question holds for all other supported platforms.

9tripod (Indiedroid)'s DeviceTree has some issues from what I've heard, preventing things like USB-C Alt mode to work at all. Do you know if that's something that got addressed/fixed in Armbian, and so would be a better idea to use their DTB instead?

amazingfate, 2023-07-08:
[image]
I finally did it.

closed 2023-07-08 by amazingfate