edk2-porting/edk2-rk3588 #40: Time is lost across reboots / power cycles

mariobalanica, 2023-07-26:
Most platforms have a HYM8563 RTC chip connected to I2C.

Unlike the other RK80x PMICs, 806 does not seem to include a RTC. Would've been useful for platforms that don't have a dedicated one, such as the Mixtile Blade 3.

mariobalanica, 2023-08-03:
Added support in: a5fdc7c...51f33c1

- Does not apply to Mixtile Blade 3 as it lacks the hardware.
- Waiting for clarification from Radxa on ROCK 5A RTC location.

@amazingfate please test on H88K.

@MollySophia, @S199pWa1k9r please test on FriendlyElec devices.

amazingfate, 2023-08-03:
I don't have a rtc battery atm. I wil buy one and test it on h88k.

mariobalanica, 2023-08-03:
You don't need a battery. Time persists across reboots without one too.

Btw, all OSes will be broken until I make the I2C driver work at runtime.

mariobalanica, 2023-08-05:
Update from Radxa: ROCK 5A production units do not have RTC.

I'm closing this since it now works fine in both UEFI and OS.

closed 2023-08-05 by mariobalanica