edk2-porting/edk2-rk3588 #42: Realtek network adapter has all-zero MAC address

mariobalanica, 2023-07-26:
The onboard Realtek PCIe NIC(s) on many platforms don't have a MAC address burnt in the eFuses (or external EEPROM). Linux and other OSes may generate a random one, but this is not ideal at all and the UEFI driver can't do it.

I've tried writing an address into the MAC registers starting at 0x0, but neither UEFI or Linux use it.
I'm going to try again later and check if it actually sticks, but if nothing comes out if this then we've got two options:
1) Figure out how and write an EFI app that can conveniently program a given MAC address into the eFuses. It's up to the user to generate an address and manually run this app.

2) Explain in the README how one can download and use the (rather obscure) Realtek PGtool app for Linux (also has aarch64 binaries!) to burn the MAC.

In both cases it's going to be a one-time, permanent action.

strygin, 2023-09-17:
Here's how to do this:
```bash
apt install -y linux-headers-legacy-rk35xx
git clone https://github.com/redchenjs/rtnicpg
cd rtnicpg
make
rmmod pgdrv # !!!
./pgload.sh
```
Make sure you don't have any Realtek modules loaded. 
If you have `r8125` builtin, you might have to reboot with `initcall_blacklist=rtl8125_init_module` as a kernel param (in grub).

How to burn MACs:

If you have only one Realtek PCIe NIC:
```bash
./rtnicpg-aarch64-linux-gnu /efuse /nodeid 00E04C001234
```
In case of two or more:
```bash
./rtnicpg-aarch64-linux-gnu /efuse /# 1 /nodeid 00E04C001234
./rtnicpg-aarch64-linux-gnu /efuse /# 2 /nodeid 00E04C001235
```

mariobalanica, 2023-09-17:
That's great, thanks!

mariobalanica, 2023-09-19:
Just tried on a board with RTL8111 and it worked wonders.

Updated README here: f6b305c7057f279190b74b36870be35c4e09f66a

closed 2023-09-19 by mariobalanica