edk2-porting/edk2-rk3588 #62: Pass a device tree to the OS

mariobalanica, 2023-07-28:
I'm not really sure what device trees to even pick, given that RK3588 is not fully mainlined in Linux yet (supported devices even less so).

Maybe the ones from Armbian for the legacy kernel would be good enough for the time being?

dtischler, 2023-07-28:
Is this the reason, perhaps, that I get different `lspci` results and a strange `[    0.315839] pci 0004:01:1f.0: unknown header type 7f, ignoring device` result when booting with a Radeon RX 550 sitting in the PCIE slot?    (I realize actually getting the GPU to **_work_** is a whole different rabbit hole consisting of cache coherency and this: https://lore.kernel.org/lkml/20230418142109.49762-2-sebastian.reichel@collabora.com/ ......... but I was curious why UEFI chokes on even posting the GPU, whereas Device Tree does not.)

**UEFI**

```
root@debian:/home/david# lspci
0004:01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
root@debian:/home/david# uname -a
Linux debian 6.1.0-9-arm64 #1 SMP Debian 6.1.27-1 (2023-05-08) aarch64 GNU/Linux

root@debian:/home/david# dmesg
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.1.0-9-arm64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP Debian 6.1.27-1 (2023-05-08)
[    0.000000] efi: EFI v2.70 by EDK II
[    0.000000] efi: SMBIOS=0xeabe0000 SMBIOS 3.0=0xeabc0000 ACPI 2.0=0xeac30018 MEMATTR=0xef2ed018 MEMRESERVE=0xeafc0d98 
[    0.000000] secureboot: Secure boot disabled
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000EAC30018 000024 (v02 RKCP  )
[    0.000000] ACPI: XSDT 0x00000000EAC3FE98 000074 (v01 RKCP   RK3588   00000000      01000013)
[    0.000000] ACPI: FACP 0x00000000EAC3FB98 000114 (v06 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: DSDT 0x00000000EAC3ED98 0005E2 (v02 RPIFDN RPI      00000002 INTL 20200925)
[    0.000000] ACPI: DBG2 0x00000000EAC3FA98 00005C (v00 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: GTDT 0x00000000EAC3FD18 000060 (v02 ARMLTD ARM-JUNO 20140727 ARM  00000099)
[    0.000000] ACPI: APIC 0x00000000EAC3E998 0002D4 (v04 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: MCFG 0x00000000EAC3F818 00007C (v01 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: SPCR 0x00000000EAC3FE18 000050 (v02 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: SSDT 0x00000000EAC3F418 00026B (v05 RKCP   PCIE21L0 00000002 INTL 20200925)
[    0.000000] ACPI: SSDT 0x00000000EAC37518 00026B (v05 RKCP   PCIE21L2 00000002 INTL 20200925)
[    0.000000] ACPI: SSDT 0x00000000EAC3DC98 000269 (v05 RKCP   PCIE34L0 00000002 INTL 20200925)
[    0.000000] ACPI: BGRT 0x00000000EAC3FF98 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.000000] ACPI: SPCR: console: uart,mmio32,0xfeb50000
[    0.000000] NUMA: Failed to initialise from firmware
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000003ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x3fdfb4380-0x3fdfb6fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000003ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000001fffff]
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000007bffff]
[    0.000000]   node   0: [mem 0x00000000007c0000-0x00000000007effff]
[    0.000000]   node   0: [mem 0x00000000007f0000-0x00000000083fffff]
[    0.000000]   node   0: [mem 0x0000000008400000-0x00000000093fffff]
[    0.000000]   node   0: [mem 0x0000000009400000-0x00000000e259ffff]
[    0.000000]   node   0: [mem 0x00000000e25a0000-0x00000000e266ffff]
[    0.000000]   node   0: [mem 0x00000000e2670000-0x00000000e277ffff]
[    0.000000]   node   0: [mem 0x00000000e2780000-0x00000000e278ffff]
[    0.000000]   node   0: [mem 0x00000000e2790000-0x00000000e281ffff]
[    0.000000]   node   0: [mem 0x00000000e2820000-0x00000000ea81ffff]
[    0.000000]   node   0: [mem 0x00000000ea820000-0x00000000ea827fff]
[    0.000000]   node   0: [mem 0x00000000ea828000-0x00000000ea888fff]
[    0.000000]   node   0: [mem 0x00000000ea889000-0x00000000eab6ffff]
[    0.000000]   node   0: [mem 0x00000000eab70000-0x00000000eac2ffff]
[    0.000000]   node   0: [mem 0x00000000eac30000-0x00000000eac3ffff]
[    0.000000]   node   0: [mem 0x00000000eac40000-0x00000000eafbffff]
[    0.000000]   node   0: [mem 0x00000000eafc0000-0x00000000effaffff]
[    0.000000]   node   0: [mem 0x00000000effb0000-0x00000000effdffff]
[    0.000000]   node   0: [mem 0x00000000effe0000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
[    0.000000]   node   0: [mem 0x00000003fc000000-0x00000003fc4fffff]
[    0.000000]   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
[    0.000000]   node   0: [mem 0x00000003fff00000-0x00000003ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000003ffffffff]
[    0.000000] cma: Reserved 64 MiB at 0x00000000d6400000
[    0.000000] psci: probing for conduit method from ACPI.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[... 2229 lines cut ...]
[   41.622164] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   41.622253] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[... cut ...]
[   41.624402] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   41.624493] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   41.624583] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   41.624661] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
[   41.624751] rockchip-i2s-tdm fddf8000.i2s: ASoC: error at snd_soc_dai_hw_params on fddf8000.i2s: -22
```

Thanks for the great project, btw!

mariobalanica, 2023-07-28:
@dtischler This is related to #52, but I suggest opening another issue for GPU. We could try making it work in UEFI first, if you're interested.

dtischler, 2023-07-28:
Hmmmm, well, I most definitely appreciate the offer, but I'm not a UEFI expert so my ability to **_truly_** assist would be limited to testing and posting logs, ha.   :-)

And do keep in mind, that based upon the cache coherency discussion / GIC / PCIe errata, there's no telling if a GPU will ever work anyways.  So I'd want to respect your time and efforts, as it's possibly a lost cause.  🤣

Marcinoo97, 2023-08-28:
is this any useful https://github.com/aarch64-laptops/edk2/tree/dtbloader-app ?

mariobalanica, 2023-10-06:
9b56a310f1413324f4c7baea6c71b456fabbbbc5

closed 2023-10-06 by mariobalanica