edk2-porting/edk2-rk3588 #68: Project Mu support

mariobalanica, 2023-07-29:
We don't have to give up on edk2, they can be both supported in the same codebase with minimal changes.

mariobalanica, 2023-10-01:
I was mainly interested in the new frontpage UI, but turns out it's unusable. HII support is really lacking and the (few) menus present are hard to navigate with a keyboard.

Project MU does have some nice security-related bits, but overall it doesn't seem worth the hassle. 
Sticking to plain EDK2 is better in case anyone feels like upstreaming the platform.

As such, I'm closing this.

closed 2023-10-01 by mariobalanica

BigfootACA, 2023-10-01:
yes, for project mu you are probably looking for https://github.com/woa-msmnile/msmnilePkg

sunflower2333, 2023-10-01:
if you want to have a try, i have a broken mu repo for my rock5b.
https://github.com/sunflower2333/mu-rk5b

sunflower2333, 2023-10-01:
[image]