edk2-porting/edk2-rk3588 #69: Secure Boot support

mariobalanica, 2023-07-29:


pbatard, 2023-09-28:
In case you haven't seen how we do it on the Pi, this is relatively easy to add during the EDK2 build process.

Basically you want to first get all the needed Secure Boot certificates and dbx, most of which can be downloaded directly:
https://github.com/pftf/RPi4/blob/master/.github/workflows/linux_edk2.yml#L50-L58

Note that, because we sure don't want any third party to have control over somebody else's machine when it comes to Secure Boot, we always generate a new PK as part of the build process and then discard the private key altogether.

Then, at EDK2 build time, you just need to feed the `-D SECURE_BOOT_ENABLE=TRUE` option along with something like `-D DEFAULT_KEYS=TRUE -D PK_DEFAULT_FILE=$WORKSPACE/keys/pk.cer -D KEK_DEFAULT_FILE1=$WORKSPACE/keys/ms_kek.cer -D DB_DEFAULT_FILE1=$WORKSPACE/keys/ms_db1.cer -D DB_DEFAULT_FILE2=$WORKSPACE/keys/ms_db2.cer -D DBX_DEFAULT_FILE1=$WORKSPACE/keys/arm64_dbx.bin`:
https://github.com/pftf/RPi4/blob/master/.github/workflows/linux_edk2.yml#L64-L65

And with this, you should have a UEFI firmware that both Windows and Linux are happy with when it comes to Secure Boot.

mariobalanica, 2023-10-01:
Thanks for input, Pete!

I've actually put this on hold because we don't have a secure var store, and there are more steps involved to ensure a trusted boot chain that I'm not sure many would bother with.

But yeah, some formal support in EDK2 to make Windows happy and satisfy one of the requirements is better than nothing.

pbatard, 2023-10-01:
Oh yeah, you're never going to get to a proper trusted boot chain on SoC systems without a lot of extra work, and we are well aware that, at least for the Pi, the Secure Boot feature we provide is more a showcase than anything that will actually help "secure" the system. It was indeed mostly added so that people could get `Secure Boot status: On` when running WoR...

mariobalanica, 2023-10-07:
Added initial support in: 8b6b3a6242af7ef37358d98b6918bfd11b52173e

Will keep this open to track further progress.

One obvious thing that needs to be fixed is disabling the recently added FDT override support when SecureBoot is on, since those files aren't signed. But it may be desirable to leave it this way until work on mainline Linux and DT settles, so folks won't have to rely on it as much.

hongkongkiwi, 2024-05-05:
Does the above method work? how can I sign a linux image I want to securely boot?

pbatard, 2024-05-05:
The above method is for adding the Secure Boot certificates to a UEFI firmware that you build yourself. It has nothing to do with signing a linux image to work with an existing UEFI firmware that you did not build. **If** you do build your own UEFI firmware from scratch you should however be able to install your own certificate by adding an extra `-D DB_DEFAULT_FILE3=my_cert.cer`, and then whatever you sign with the private key that matches the one from this public cert should be validated under Secure Boot.

**OR**, and I assume this is really what you actually want to accomplish, if you are not building a UEFI firmware from scratch, but just want to have something you sign work with an existing UEFI firmware, you should follow [this extensive guide](https://www.rodsbooks.com/efi-bootloaders/controlling-sb.html) to add your own certs using the existing UEFI interfaces.

hongkongkiwi, 2024-05-05:
Makes sense, thanks for the link. That clears up signing downstream.

But for upstream, I am interested to use the builtin eFuse on Rockchip to make sure that only my custom signed bootloader can be used.

I've seen this done successfully with U-Boot and the rockchip signing tools (they sign the idbloader.img file). In our case, is the idblock.bin an equivalent of that?

pbatard, 2024-05-05:
I'm afraid you're going to have somebody else about this, as I know nothing about `idblock.bin` and I am not involved in RK3588 development.

DualTachyon, 2024-05-05:
@hongkongkiwi you want this I guess: https://github.com/DualTachyon/rk3588-secure-boot

hongkongkiwi, 2024-05-06:
Very helpful! Thanks @DualTachyon

bexcran, 2024-11-27:
@pbatard I just noticed there are _two_ arm64 dbx files: the one listed on uefi.org is https://uefi.org/sites/default/files/resources/arm64_DBXUpdate.bin while the RPi4 script is fetching https://uefi.org/sites/default/files/resources/dbxupdate_arm64.bin - and they're different (though I haven't looked into how/why).

Also, I'm seeing an error trying to add the dbxDefault variable: the code expects it to be in a X509 format, not the raw data suitable for adding directly via SetVariable:
```
SecureBootFetchData: Invalid key format: 0
Content for dbxDefault not found
```

pbatard, 2024-11-27:
@bexcran, the file to use is **always** the one linked on https://uefi.org/revocationlistfile, therefore, **currently** that would be https://uefi.org/sites/default/files/resources/arm64_DBXUpdate.bin. At the time when we added the files for Pi 4, that link pointed to https://uefi.org/sites/default/files/resources/dbxupdate_arm64.bin (and we expected that the UEFI forum would not change URLs when they updated the DBXs, but it appears that they do), hence the discrepancy.

AFAIK, the DBX links have always been in the format needed for adding with `SetVariable()`, because it would make no sense to provide it otherwise (since it is meant to be used by the OS directly with `SetVariable()`), so your error must come from something else.

bexcran, 2024-11-27:
@pbatard RPi4 is also using SecureBootDefaultKeysDxe and SecureBootVariableProvisionLib so it's going to have the same problem I ran into: keys/arm64_dbx.bin is in a format for SetVariable while the other certificates are in DER format. SecureBootDefaultKeysEntryPoint tries to initialize PKDefault, KEKDefault, dbDefault, dbtDefault and dbxDefault.

SecureBootInitKEKDefault, SecureBootInitDbDefault and SecureBootInitDbxDefault all have the same pattern of code:
```
  Status = SecureBootFetchData (&gDefaultdbFileGuid, &SigListsSize, &EfiSig);
  if (EFI_ERROR (Status)) {
    return Status;
  }

  Status = gRT->SetVariable (
                  EFI_DB_DEFAULT_VARIABLE_NAME,
                  &gEfiGlobalVariableGuid,
                  EFI_VARIABLE_RUNTIME_ACCESS | EFI_VARIABLE_BOOTSERVICE_ACCESS,
                  SigListsSize,
                  (VOID *)EfiSig
                  );
```

SecureBootFetchData does the same thing in all cases: 
```
  while (1) {
    if (NewCertInfo == NULL) {
      Status = EFI_OUT_OF_RESOURCES;
      break;
    } else {
      CertInfo = NewCertInfo;
    }

    Status = GetSectionFromAnyFv (
               KeyFileGuid,
               EFI_SECTION_RAW,
               KeyIndex,
               &Buffer,
               &Size
               );

    if (Status == EFI_SUCCESS) {
      RsaPubKey = NULL;
      if (RsaGetPublicKeyFromX509 (Buffer, Size, &RsaPubKey) == FALSE) {
        DEBUG ((DEBUG_ERROR, "%a: Invalid key format: %d\n", __func__, KeyIndex));
        if (EfiSig != NULL) {
          FreePool (EfiSig);
        }

        FreePool (Buffer);
        Status = EFI_INVALID_PARAMETER;
        break;
      }

      CertInfo[KeyIndex].Data     = Buffer;
      CertInfo[KeyIndex].DataSize = Size;
      KeyIndex++;
      NewCertInfo = ReallocatePool (
                      sizeof (SECURE_BOOT_CERTIFICATE_INFO) * KeyIndex,
                      sizeof (SECURE_BOOT_CERTIFICATE_INFO) * (KeyIndex + 1),
                      CertInfo
                      );
    }

    if (Status == EFI_NOT_FOUND) {
      Status = EFI_SUCCESS;
      break;
    }
  }
```

So, since the DBX is in a different format than the KEK and DB it's going to cause a problem. I only noticed because I created  a RELEASE build with DEBUG_ERROR and DEBUG_WARN levels enabled so any errors being logged were easy to notice.

bexcran, 2024-11-27:
> Please see the UEFI specs. There's simply no way for DBX to be anything else than a custom list of element. It's **always** going to have to be in a different format than KEK or DB (well, as far as we're talking about adding individual certs to KEK and DB) because by nature we're dealing with completely different entities.

Then the code in EDK2 is wrong. It's assuming it's a list of certificates.

csillag, 2026-09-17:
Two data points on the dbx default and an offer. The dbx measurement below was on a ROCK 5B running a build of current master (824e6c12).

1. **The dbx URL in `.github/workflows/build.yml` no longer returns the file to a script.** From my (non-GitHub) host, `curl` on `https://uefi.org/sites/default/files/resources/dbxupdate_arm64.bin` gets **HTTP 403** with a 5.5 KB Cloudflare "Just a moment..." challenge page. The same happens for `arm64_DBXUpdate.bin`, and with a browser User-Agent. The workflow runs `curl -L ... -o keys/arm64_dbx.bin` without `-f`, so if GitHub's runners get the same answer, that HTML page becomes `DBX_DEFAULT_FILE1` and the build still succeeds. I have not checked what the runners actually receive. Microsoft also publishes the signed update at `microsoft/secureboot_objects` (`PostSignedObjects/DBX/arm64/DBXUpdate.bin`), which downloads fine.
2. **Confirming @bexcran's finding:** with a proper `DBXUpdate.bin` passed as `DBX_DEFAULT_FILE1`, `dbxDefault` is **absent** at runtime (checked via efivarfs from Linux). `PKDefault`, `KEKDefault` and `dbDefault` are all present. `SecureBootCreateDataFromInput()` wraps every default file as a single `EFI_CERT_X509_GUID` entry, so a dbx hash list cannot be supplied that way. dbx has to come from an authenticated update after enrollment.

**Offer:** for a downstream build I added a small DXE driver, behind an opt-in `-D SECURE_BOOT_ENROLL_DEFAULTS=TRUE`. At EndOfDxe, if the platform is in Setup Mode, it enrolls db, dbx, dbt, KEK and PK from the defaults (the same sequence as `SecurityPkg/EnrollFromDefaultKeysApp`). So a freshly flashed board boots with Secure Boot enforced without a trip through the setup menu.

Tested on the ROCK 5B:
- first boot enrolls, and the kernel reports `Secure boot enabled`;
- shim -> GRUB -> signed kernel boots;
- a kernel with its signature stripped is refused (`bad shim signature`).

I understand from the earlier comments that the missing secure variable store makes this more of a convenience than real protection. If it is still of interest, I can send it as a PR.

pbatard, 2026-09-17:
> **The dbx URL in `.github/workflows/build.yml` no longer returns the file to a script.**

That's because, even outside of https://uefi.org/revocationlistfile having introduced a cloudflare challenge to prevent script access, it also points to https://github.com/microsoft/secureboot_objects being **the** new official source for the DBX.

In other words, https://uefi.org/sites/default/files/resources/dbxupdate_arm64.bin should be updated to https://github.com/microsoft/secureboot_objects/raw/main/PostSignedObjects/SignedByKEK2023/dbx_aarch64.efiauth2.

Since it is signed by KEK 2023, you also need to make sure that you have KEK 2023 available (which currently the builds don't) and you also want the 2023 DBs.

I strongly suggest you use the following from the Pi4 builds as your reference to have a comprehensive and up to date Secure Boot databases for RK3588 builds: https://github.com/pftf/RPi4/blob/master/.github/workflows/linux_edk2.yml#L46-L66.