edk2-porting/edk2-rk3588 #74: OrangePi 5 Plus: "No Media" error when booting from a TF card reader

indygit, 2023-08-03:
Hello:

    I used orangepi-5plus_UEFI_Debug_4584499.img for OrangePi 5 Plus and hit the "No Media" error below.
    Does it mean the card in reader is not detected?

    The same TF card reader was used to write the Windows image.
    The reader was connected to the top USB 3 connector on the board for booting.
    I think that should be the right connector, can you confirm?
    I also tried the USB-C connector, but it didn't make a difference.

    I am using a Windows 10 build 19045.3271.230720-0819.22H2_RELEASE_SVC_PROD3_CLIENTMULTI_A64FRE_EN-US.ISO

    Should I get a different reader or even a different "disk"?

                                            Best regards, Indy Zhu, 8/3/2023

ConvertPages: Incompatible memory types, the pages to allocate have been allocated
XHCI: Initialize DWC3 at 0xFC000000
XHCI: Initialize DWC3 at 0xFC400000
XHCI: Initialize DWC3 at 0xFCD00000
XHCI: Don't support multi-TT feature for Hub now. (force to disable MTT)
UsbBootExecCmd: Success to Exec 0x0 Cmd (Result = 1)
UsbBootDetectMedia: UsbBootIsUnitReady (No Media)
UsbBootExecCmd: Success to Exec 0x0 Cmd (Result = 1)
UsbBootDetectMedia: UsbBootIsUnitReady (No Media)
UsbBootExecCmd: Success to Exec 0x0 Cmd (Result = 1)
UsbBootDetectMedia: UsbBootIsUnitReady (No Media)
OHCI started for controller @ EF49BC18, base address: 0xFC840000
OHCI started for controller @ EF49B198, base address: 0xFC8C0000

mariobalanica, 2023-08-03:
This is related to issue #39 

The workaround is to isolate the USB 3.0 SuperSpeed pins, either by using an USB 2.0 adapter/cable or by plugging in the connector only half-way through (so the SS pins do not make contact).

indygit, 2023-08-04:
Hi, Mario:

     Thanks a lot.
     Plugging the reader's USB cable half-way through worked around the "No Media" issue.

     But my TF card is probably not properly written, UEFI couldn't load Windows from it and eventually UEFI shell is shown.
     I know very little about UEFI, but the command below seems to show that there is no recognized file system.

    I used WoR.exe for writing the TD card, could there be something I didn't do correctly?
    On Windows, I do see this TD card has 2 partitions, 1 for UEFI and one NTFS partition (containing Windows)

                                             Best regards, Indy, 8/4/2023

Shell> map -v
Mapping table
     BLK0: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,00002CFE0000000000)/SD(0x0)

         Handle:      [78]
         Media Type:  Unknown
         Removable:   No
         Current Dir: BLK0:
     BLK1: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,00002CFE0000000000)/SD(0x0)
/HD(1,GPT,B750E44E-833F-4A30-C38C-B117241D84D4,0x800,0x4000)
         Handle:      [B4]
         Media Type:  HardDisk
         Removable:   No
         Current Dir: BLK1:
Shell> mount BLK1 C
map: No mappable target found - 'C'

mariobalanica, 2023-08-05:
>  But my TF card is probably not properly written, UEFI couldn't load Windows from it and eventually UEFI shell is shown.
 I know very little about UEFI, but the command below seems to show that there is no recognized file system.
 
Yes, that's what seems to be happening.

Does it see the FAT32 partition if you connect the SD card to the OPi's slot rather than through USB?

indygit, 2023-08-06:
Thanks Mario. I will try it as soon as possible.
Just to clarify, does that FAT32 partition contains Windows boot loader?
Windows itself is in the NTFS partition.

mariobalanica, 2023-08-06:
Yes, it contains the bootloader.

indygit, 2023-08-09:
Hi, Mario:

    I think plugging in the USB 3.0 TF card reader half-way through requires serious skills, I probably didn't do it right, at least I can't do it consistently. :-)

    I think my TF card is written correctly, I did 3 experiments:

1. Using diskpart.exe on Windows, I verified the TF card has 3 partitions: 1 for UEFI (FAT32 with a special GUID to make it hidden), 1 reserved (unused) partition and 1 with NTFS for Windows.

2. If I put TF card in the SD slot and boot, I can see the following:
>>Start HTTP Boot over IPv6..
  Error: Could not retrieve NBP file size from HTTP server.

  Error: Remote boot cancelled.
Process PlatformRecovery0000 (Default PlatformRecovery) ...
Installed Fat filesystem on EEE49798
add-symbol-file bootmgfw.pdb 0x10000400
Loading driver at 0x00010000000 EntryPoint=0x00010020508 bootmgfw.efi
ConvertPages: Incompatible memory types, the pages to allocate have been allocated

3. If I boot without the TF carder and then plug it all the way in when UEFI shell is running, I see the following:
Shell> map -v
map: No mapping found.
Shell> map -r
map: Status: Not Found
Shell> UsbBootExecCmd: Success to Exec 0x0 Cmd (Result = 1)
UsbBootDetectMedia: UsbBootIsUnitReady (No Media)
map: No mapping found.em on EED52B18
Shell> ExecCmd: Success to Exec 0x0 Cmd (Result = 1)
Shell> DetectMedia: UsbBootIsUnitReady (No Media)
Shell> ExecCmd: Success to Exec 0x0 Cmd (Result = 1)
Shell> DetectMedia: UsbBootIsUnitReady (No Media)
Shell> Device: read of part_lba failed No Media
Shell>
Shell> map -v
Shell>  table
Shell> cls
Shell> map -r
UsbBootExecCmd: Success to Exec 0x0 Cmd (Result = 1)
UsbBootDetectMedia: UsbBootIsUnitReady (No Media)
Mapping table
      FS0: Alias(s):HD0a0d0a1:;BLK1:
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,000040FC0000000000)/USB(0x0
,0x0)/USB(0x3,0x0)/Unit(0x0)/HD(1,GPT,96E421D4-9A41-4ECB-9542-B555D75E00CE,0x800
,0x40000)
     BLK0: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,000040FC0000000000)/USB(0x0
,0x0)/USB(0x3,0x0)/Unit(0x0)
     BLK4: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,000040FC0000000000)/USB(0x0
,0x0)/USB(0x3,0x0)/Unit(0x1)
     BLK2: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,000040FC0000000000)/USB(0x0
,0x0)/USB(0x3,0x0)/Unit(0x0)/HD(2,GPT,AAC15402-1D87-4139-A67A-2194732A80CC,0x408
00,0x8000)
     BLK3: Alias(s):
          VenHw(0D51905B-B77E-452A-A2C0-ECA0CC8D514A,000040FC0000000000)/USB(0x0
,0x0)/USB(0x3,0x0)/Unit(0x0)/HD(3,GPT,5CE10EDF-6316-405F-A727-12BC58143FF2,0x488
00,0x3B2F000)
Shell> ls fs0
ls: Current directory not specified.
Shell> fs0:
FS0:\>
FS0:\> ls
Directory of: FS0:\
06/05/2023  11:00              33,851  bcm2710-rpi-3-b-plus.dtb
06/05/2023  11:00              33,224  bcm2710-rpi-3-b.dtb
06/05/2023  11:00              30,923  bcm2710-rpi-cm3.dtb
06/05/2023  11:00              52,476  bootcode.bin
08/03/2023  11:57                 242  config.txt
08/03/2023  12:21 <DIR>         1,024  EFI
06/05/2023  11:00               7,266  fixup.dat
06/05/2023  10:52               3,593  Readme.md
08/02/2023  21:12                   9  release
06/05/2023  10:59           2,031,616  RPI_EFI.fd
06/05/2023  11:00           2,977,280  start.elf
         10 File(s)   5,170,480 bytes
          1 Dir(s)
FS0:\> ls EFI
Directory of: FS0:\EFI\
08/03/2023  12:21 <DIR>         1,024  .
08/03/2023  12:21 <DIR>             0  ..
08/03/2023  12:21 <DIR>         1,024  Boot
08/03/2023  12:21 <DIR>         1,024  Microsoft
          0 File(s)           0 bytes
          4 Dir(s)
FS0:\> ls EFI\Boot
Directory of: FS0:\EFI\Boot\
08/03/2023  12:21 <DIR>         1,024  .
08/03/2023  12:21 <DIR>         1,024  ..
08/02/2023  19:21           1,563,464  bootaa64.efi
          1 File(s)   1,563,464 bytes
          2 Dir(s)
FS0:\> ls EFI\Microsoft
Directory of: FS0:\EFI\Microsoft\
08/03/2023  12:21 <DIR>         1,024  .
08/03/2023  12:21 <DIR>         1,024  ..
08/03/2023  12:21 <DIR>         4,096  Boot
08/03/2023  12:21 <DIR>         1,024  Recovery
          0 File(s)           0 bytes
          4 Dir(s)
FS0:\> ls EFI\Microsoft\Boot
Directory of: FS0:\EFI\Microsoft\Boot\
08/03/2023  12:21 <DIR>         4,096  .
08/03/2023  12:21 <DIR>         1,024  ..
08/03/2023  12:21              20,480  BCD
08/03/2023  12:21 <DIR>         1,024  bg-BG
08/02/2023  19:21              10,717  boot.stl
08/02/2023  19:21           1,563,464  bootmgfw.efi
08/02/2023  19:21           1,546,624  bootmgr.efi
08/03/2023  12:21 <DIR>         1,024  cs-CZ
08/03/2023  12:21 <DIR>         1,024  da-DK

   I am perfectly happy to wait for you to get around and fix the root cause.
   Please let me know if there is anything I can help you with.

                                      Best regards, Indy, 8/8/2023

mariobalanica, 2023-08-11:
I can see you were able to list the directories on the USB card reader, but it's probably working intermittently due to the USB issue.

indygit, 2023-08-18:
It works reasonably well because I plugged it in after UEFI shell is ready.

mariobalanica, 2023-08-19:
It should work fine with the latest build: https://github.com/edk2-porting/edk2-rk3588/actions

Please let me know if you still have issues.

closed 2023-08-22 by mariobalanica