edk2-porting/edk2-rk3588 #76: [BUG] Not recognizing SSD on USB 3.0 port on NanoPi R6S

lesca, 2023-08-05:
Hello,
This is to let you aware it doesn't work on r6s.

I tested the UEFI firmware in artifacts [https://github.com/edk2-porting/edk2-rk3588/actions/runs/5768458137] for Nano Pi R6S but to find it doesn't recognize the SSD on USB 3 port but it can do on USB 2. 

However, the boot failed with blue screen with message "INACCESSIBLE_BOOT_DEVICE" and during the boot process, my SSD LED is not flashing. I assume there is no READ / WRITE operations during that boot on USB2 port.

Hopefully it could be fixed soon. I can wait to test WIN ARM on my board. :) 

Regards,
Lesca

lesca, 2023-08-07:
Hello,
I tested the UEFI firmware further. Here are some updates. 

### What I did
* Since the UEFI won't boot on USB3 SSD, I managed to flash the WIN11 ARM image to eMMC 
* Boot the board from SD card with UEFI
* The OS on eMMC booted successfully by selecting **eMMC User Data** option in UEFI 
* In Windows 11, the USB 2 port is not recognized, that's why I cannot boot the OS via SSD on USB 2 port.

### Issues on USB 3
The UEFI can recognized the USB 3 port after many tries. It's quite unstable that UEFI recognize the USB 3 port and its connected devices. 

Steps:
* Boot the board and press **ESC** to enter UEFI 
* unplug the USB hub / SSD, wait several seconds, plug in again 
* check if the SSD LED is blinking, if no, try above step again
* if yes, then the SSD is recognized, manually select it from the boot device list

mariobalanica, 2023-08-08:
I believe this could also be related to #39 and #74.
Does plugging in the USB device only half-way through (so the 3.0 pins don't touch) make it more reliable?

lesca, 2023-08-09:
> Does plugging in the USB device only half-way through (so the 3.0 pins don't touch) make it more reliable?

I read the issue #74. I understand "plugging in the USB device only half-way through" makes the port work as USB 2 mode.

I can confirm it's connected fully plugged in. After the SSD is recognized, the windows is booted successfully. I think it doesn't have USB2 driver, because Win11 won't recognize devices on the physical USB2 port.  

However, the windows OOBE is outstandingly slow in comparison that's installed on eMMC. Maybe it's working as USB 2 mode. How can I know that? 

Regards,
Lesca

lesca, 2023-08-09:
Now I back home, further information: 
* since R6S only have one USB3 port but USB2 doesn't work in Windows, I used a USB3 hub to connect the SSD and a USB dongle for mouse and keyboard. In this way, detection of usb devices on the hub (both SSD and usb dongle) is not stable. The workaround is try plug in and out in UEFI menu many times.
* Now I connect usb dongle to usb2 port (it works in UEFI menu before Windows boots), and SSD directly to usb3 port. When I check the boot device list, the SSD always lists there after many reboots (ctrl-alt-del) / power on and off again / plug out and in again. -- it's stable!!

Narrow down: 
it's more likely the UEFI cannot stably detect the devices connected to a USB hub.

mariobalanica, 2023-08-11:
Thanks for the details!

> it's more likely the UEFI cannot stably detect the devices connected to a USB hub.

Yeah, this is a known issue with USB 3.0 hubs. It's already being tracked in #39, so I will close this one.

closed 2023-08-11 by mariobalanica