edk2-porting/edk2-rk3588 #79: HOW  am I supposed to flash to sdcard?

derzahla, 2023-08-14:
no partitions are detected by losetup, nor when writing the img to sdcard. It would be nice to get some clarity here.  The resulting sdcard does not boot in my orangepi. This image appears to be some protective/hybrid mbr or some shit WTF i dont even

mariobalanica, 2023-08-14:
Have you tried the latest image from artifacts?

amazingfate, 2023-08-14:
Download the latest nightly build: https://github.com/edk2-porting/edk2-rk3588/actions/runs/5850363526
This has get fixed by https://github.com/edk2-porting/edk2-rk3588/commit/444ac8109bd1403f1bd54a0ff9d9dd5ac597548f.

derzahla, 2023-08-14:
thanks! I had tried the nightly early, maybe from last week but not this new of one. sorry for being cranky, i start to go bonkers from lack of sleep!

 So now gdisk doesnt complain about  about the partition being larger than the disk, and i am able to successfully repair the gpt table with gdisk, or so it says.  is the 'uboot' labeled partition any type of semi-filesystem that I should be able to mount from linux?  Cuz my system doesnt detect a valid filesystem.  For clarity, what I am hoping to do is use the UEFI capabilities to PXE boot my orangepi5plus through ipxe.

mariobalanica, 2023-08-14:
You should create another FAT32 EFI System Partition. The 8MB U-Boot one is reserved.

By the way, if your board does not have MAC addresses burnt into the Realtek NICs, you may not be able to use networking in UEFI.

derzahla, 2023-08-14:
hmm, I created a new fat partition on the sdcard i flashed with orangepi-5plus_UEFI_Debug_5bfd480.img, but its still not attempting to boot uefi on my opi5+. I know last week I flashed to spi and it would at least boot.  Had a countdown for pressing 'esc' to get into settings but keyboard didnt seem to be detected so i couldnt do much.  Any other suggestions?  Not sure about the realtek NICs but will cross that bridge if I can get to it. Thanks!

nicolho, 2023-08-14:
> @derzahla make sure the SPI is clear. I recall seeing OPi 5 Plus prefer it over any other boot media. Only one HDMI port is supported, check both.
Yes, of the three HDMI ports, that's the middle one (primary output).

derzahla, 2023-08-14:
oh damn, i was using hdmi0 all along.  I dont think i have ever used the middle HDMI port before but i guess I mustve last time I got the countdown or did 0.7.1 release work on more HDMI ports? Better yet this time, I am able to press esc get to device manager and  my keyboard is plugged into just a regular usb3 port(top),. I tried using an adapter last time to plug into usbC and keyboard didnt work that time either.   Time to try some pxe booting, this is awesome guys thanks!!!

derzahla, 2023-08-14:
@mariobalanica @nicolho  So pxe booting unfortuantly didnt work. I didnt even know it was built in and expected to use ipxe.  Well neither of them are working.  NIC with a link lights up, but neigh edk2 nor ipxe appear to detect the link, or at least do not try to get a DHCP address.  Both MAC addresses do show up in UEFI under network settings so I would guess that means they are burned in?  Would appreciate any other ideas you may have but then I will close this since this issue since it is venturing off topic. thanks!

mariobalanica, 2023-08-15:
@derzahla do the network interfaces have a valid MAC or it's all zeros?

mariobalanica, 2023-08-15:
I've just tested on my OPi5 Plus, it does have valid MAC addresses and networking works fine. Have not tried PXE but there's no reason why it shouldn't work assuming the server is configured correctly.

nicolho, 2023-08-15:
@mariobalanica FYI, I quickly set up a local PXE server to test PXE boot with my OPI5+, and still booting your latest firmware from sdcard, I was directly able to download a file through onboard ethernet. Some log excerpt (it loaded much bigger files too) :
````
[Bds]Booting UEFI PXEv4 (MAC:C0742BFF5D80)
>>Start PXE over IPv4.MnpSyncSendPacket: No network cable detected.
  Station IP address is 192.168.1.36
  Server IP address is 192.168.1.16
  NBP filename is bootaa64.efi
  NBP filesize is 1792512 Bytes
 Downloading NBP file...
  NBP file downloaded successfully.
````
(sorry, I'm a total noob with PXE... Suggestions of some simple "how-to" to load anything compatible with this board for testing are welcome)

derzahla, 2023-08-15:
i didnt get far but i did have to fix the config for dhcp match line on my server and it now is attempting to grab files from tftp.  The transfer always times out though before completion, a probably I dont have with any of my x86 PXE boots.  Where is your pxeboot code derived from?  Its not iPXE based, correct?

derzahla, 2023-08-15:
Continuing here https://github.com/edk2-porting/edk2-rk3588/issues/80

closed 2023-08-15 by derzahla