edk2-porting/edk2-rk3588 #80: OPI5+ PXE boot times out during transfer from tftp

derzahla, 2023-08-15:
Trying to PXE boot my orange pi 5 plus but  the  tftp transfer always times out before completion, a problem I  havent had with any of x86 PXE boots.  Any knowledge of similar issues or any ideas? Where is your pxeboot code derived from?  Its not iPXE based, correct?

One more slightly offtopic question, since your image is <8MB would I be able to add an additional 8M EFI partition for store some EFI tools etc, before writing to the 16MB spi flash?

derzahla, 2023-08-17:
Im just gonna close this.  Everything is working beautifully approximately since writing to SPI. Currently booting ubuntu via UEFI grub instead of uboot.  I really wanted to go back and test from SDcard because I would be surprised if that made the difference but  I need to find it again first.  Thank for the great work and help guys

closed 2023-08-17 by derzahla

mariobalanica, 2023-08-17:
Great to hear :-)