edk2-porting/edk2-rk3588 #87: 0.8 release problem with Orange Pi 5 plus and Rock 5b

harryzz, 2023-09-17:
Hello,

there are problem with latest release 0.8 with rock 5b and opi 5 plus.
[image]

mariobalanica, 2023-09-17:
Are you trying to boot some OS? How did you flash the image?

harryzz, 2023-09-17:
> Are you trying to boot some OS? How did you flash the image?

Yes. I have running centos 9 on both (installed on NVME). Also i have some usb sticks and sd cards with fedora, armbian etc .... With build from commit before 13-09-2023 all working fine. Probably problem is somewhere here: 
Updated EDK2 base to edk2-stable202308 OR Updated rkbin.
I have burned image to spi flash with RKDevTool v3.15. 

regards,
Zahari

mariobalanica, 2023-09-17:
To clarify, the Synchronous Exception only appears after you attempt to boot the OS, right? Are you able to enter firmware setup or the UEFI shell on v0.8?

Can you try this build right before the EDK2 update? https://github.com/edk2-porting/edk2-rk3588/commit/7a0363351f6c6418e20163bdf2e214bc09c64809

Also, are you using a debug or release variant? I'd try both to see if there's any difference.

harryzz, 2023-09-17:
> To clarify, the Synchronous Exception only appears after you attempt to boot the OS, right? Are you able to enter firmware setup or the UEFI shell on v0.8?
yes UEFI shell is availabe.

[image]
[image]

> 
> Can you try this build right before the EDK2 update? [7a03633](https://github.com/edk2-porting/edk2-rk3588/commit/7a0363351f6c6418e20163bdf2e214bc09c64809)
commit 7a0363351f6c6418e20163bdf2e214bc09c64809 (HEAD)
Author: Mario Bălănică <mariobalanica02@gmail.com>
Date:   Fri Sep 15 01:06:44 2023 +0300

    Silicon/Rockchip: I2cDxe: Connect all controller handles at end of Dxe

    Only the first handle was connected.

[image]

> 
> Also, are you using a debug or release variant? I'd try both to see if there's any difference.

I use release version from github artifacts.

mariobalanica, 2023-09-17:
Do you have a link for the CentOS image you're using? Is it CentOS Stream 9 aarch64 from: https://www.centos.org/download/ ?

harryzz, 2023-09-17:
> Do you have a link for the CentOS image you're using? Is it CentOS Stream 9 aarch64 from: https://www.centos.org/download/ ?

yes. iso image burned to usb and installed to nvme. I use modified acpi table to be able os to see NVME. (you have delete this branch :) - pcie-switch-testing )

mariobalanica, 2023-09-17:
Can you expand a bit on the PCIe matter? Does NVME not work without patching ACPI, or are you using a PCIe switch in between?

That branch was just a hack that happens to work on Linux but will not do so on any other OS.

harryzz, 2023-09-17:
> Can you expand a bit on the PCIe matter? Does NVME not work without patching ACPI, or are you using a PCIe switch in between?
> 
> That branch was just a hack that happens to work on Linux but will not do so on any other OS.

for sure not working ( may be for some patched kernels - armbian .. etc - can work i don't know ) Without acpi patch installer can't see the nvme.

mariobalanica, 2023-09-17:
Weird, it does work fine for me & others without any ACPI or kernel patches. What NVME drive are you using?

dmesg output before and after patching ACPI would be helpful, if you're interested in debugging this further.

harryzz, 2023-09-17:
> Weird, it does work fine for me & others without any ACPI or kernel patches. What NVME drive are you using?
> 
> dmesg output before and after patching ACPI would be helpful, if you're interested in debugging this further.

Hi
rock 5b: Samsung SSD 970 EVO Plus 2TB PCIe M.2 2280 SSD
orange pi 5 plus: Crucial P3 Plus 2TB PCIe M.2 2280 SSD

[edk2_rk3588_acpi.patch](https://github.com/edk2-porting/edk2-rk3588/files/12643485/edk2_rk3588_acpi.patch)
[dmesg.no_acpi_patch.txt](https://github.com/edk2-porting/edk2-rk3588/files/12643487/dmesg.no_acpi_patch.txt)
[dmesg.acpi_patched.txt](https://github.com/edk2-porting/edk2-rk3588/files/12643488/dmesg.acpi_patched.txt)

athre0z, 2023-09-17:
As another data point: with 0.8 on my Rock 5B I see NVMe just fine without any ACPI patches with a NixOS 6.1-rc1 kernel. I have pretty much the same SSD as OP, except that it's the 1TB variant.

harryzz, 2023-09-17:
the only difference if 6.6-rc1 is kernel version is this. But in pure ACPI this make no sense.

mariobalanica, 2023-09-18:
Workaround available in 842db13abdcf46a2cc45570806dd7c0e0bc00db8

As for NVME, I was not able to reproduce the issue on CentOS Stream 9.
In your case it does not see the PCIe device at all, so it seems related to #66, although this is a single device.

closed 2023-09-18 by mariobalanica