edk2-porting/edk2-rk3588 #91: Corsair Force MP600 SSD not recognized by OS under ACPI

byhat, 2023-10-10:
Not sure if it's because of MP600 being a PCIe 4.0 drive or some other reason, but a Samsung 970 EVO PCIe 3.0 drive can be recognized successfully under ACPI. On the other hand, MP600 can also be recognized with device tree mode.

Hardware: Rock 5b + Force MP600
Operating System: CoreOS testing (linux 6.5.5), also reproducible on CentOS Stream 9
UEFI: edk2-rk3588 v0.9 (ACPI, no device tree)

mariobalanica, 2023-10-10:
Would you be able to provide serial logs if I send some test builds of the firmware?

byhat, 2023-10-10:
I don't have a USB to TTL serial cable for now, but I can get one if that's necessary.

mariobalanica, 2023-10-10:
It may not be necessary. First of all, does the NVME drive show up in UEFI boot options?

byhat, 2023-10-10:
The drive does show up in UEFI. It's only the OS (only tried Linux) that's not recognizing the SSD.

mariobalanica, 2023-10-10:
[rock-5b-ssd-test1.zip](https://github.com/edk2-porting/edk2-rk3588/files/12861896/rock-5b-ssd-test1.zip)

Can you share the output of the `pci` command in UEFI Shell with the firmware image above?

NVME won't show up in the boot menu, but it should work in Linux with ACPI now. Please attach `dmesg` logs if you can too.

byhat, 2023-10-10:
CoreOS can indeed recognize the drive now.
dmesg:
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.5.5-300.fc39.aarch64 (mockbuild@959b1746e0784dcdb58143b363973403) (gcc (GCC) 13.2.1 20230918 (Red Hat 13.2.1-3), GNU ld version 2.40-13.fc39) #1 SMP PREEMPT_DYNAMIC Sat Sep 23 23:14:54 UTC 2023
[    0.000000] KASLR disabled due to lack of seed
[    0.000000] efi: EFI v2.7 by EDK II
[    0.000000] efi: SMBIOS=0xeac20000 SMBIOS 3.0=0xeac00000 ACPI 2.0=0xeac30018 MEMATTR=0xef179018 MOKvar=0xe9df0000 MEMRESERVE=0xeab41698 
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000EAC30018 000024 (v02 RKCP  )
[    0.000000] ACPI: XSDT 0x00000000EAC3FE98 000084 (v01 RKCP   RK3588   00000000      01000013)
[    0.000000] ACPI: FACP 0x00000000EAC3FB98 000114 (v06 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: DSDT 0x00000000EAC37518 000842 (v02 RKCP   RK3588   00000002 INTL 20200925)
[    0.000000] ACPI: DBG2 0x00000000EAC3FA98 00005C (v00 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: GTDT 0x00000000EAC3FD18 000060 (v02 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: APIC 0x00000000EAC3E998 0002D4 (v04 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: MCFG 0x00000000EAC3F818 00007C (v01 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: PPTT 0x00000000EAC3ED98 000220 (v01 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: SPCR 0x00000000EAC3FE18 000050 (v02 RKCP   RK3588   00000000 EDK2 00000000)
[    0.000000] ACPI: SSDT 0x00000000EAC3E318 00026B (v05 RKCP   PCIE21L0 00000002 INTL 20200925)
[    0.000000] ACPI: SSDT 0x00000000EAC3F018 00026B (v05 RKCP   PCIE21L2 00000002 INTL 20200925)
[    0.000000] ACPI: SSDT 0x00000000EAC3C798 000269 (v05 RKCP   PCIE34L0 00000002 INTL 20200925)
[    0.000000] ACPI: SSDT 0x00000000EAC3F698 000158 (v05 RKCP   USB2HOST 00000002 INTL 20200925)
[    0.000000] ACPI: BGRT 0x00000000EAC3FF98 000038 (v01 INTEL  EDK2     00000002      01000013)
[    0.000000] ACPI: SPCR: console: uart,mmio32,0xfeb50000
[    0.000000] NUMA: Failed to initialise from firmware
[    0.000000] NUMA: Faking a node at [mem 0x0000000000000000-0x00000003ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x3f9edf300-0x3f9ef5fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000003ffffffff]
[    0.000000]   Device   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x00000000001fffff]
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000007bffff]
[    0.000000]   node   0: [mem 0x00000000007c0000-0x00000000007effff]
[    0.000000]   node   0: [mem 0x00000000007f0000-0x00000000083fffff]
[    0.000000]   node   0: [mem 0x0000000008400000-0x00000000093fffff]
[    0.000000]   node   0: [mem 0x0000000009400000-0x00000000e9deffff]
[    0.000000]   node   0: [mem 0x00000000e9df0000-0x00000000e9dfffff]
[    0.000000]   node   0: [mem 0x00000000e9e00000-0x00000000e9edffff]
[    0.000000]   node   0: [mem 0x00000000e9ee0000-0x00000000e9f5ffff]
[    0.000000]   node   0: [mem 0x00000000e9f60000-0x00000000e9f8ffff]
[    0.000000]   node   0: [mem 0x00000000e9f90000-0x00000000ea77ffff]
[    0.000000]   node   0: [mem 0x00000000ea780000-0x00000000ea78cfff]
[    0.000000]   node   0: [mem 0x00000000ea78d000-0x00000000ea7edfff]
[    0.000000]   node   0: [mem 0x00000000ea7ee000-0x00000000eaafffff]
[    0.000000]   node   0: [mem 0x00000000eab00000-0x00000000eab3ffff]
[    0.000000]   node   0: [mem 0x00000000eab40000-0x00000000eab6ffff]
[    0.000000]   node   0: [mem 0x00000000eab70000-0x00000000eac2ffff]
[    0.000000]   node   0: [mem 0x00000000eac30000-0x00000000eac3ffff]
[    0.000000]   node   0: [mem 0x00000000eac40000-0x00000000eafbffff]
[    0.000000]   node   0: [mem 0x00000000eafc0000-0x00000000effaffff]
[    0.000000]   node   0: [mem 0x00000000effb0000-0x00000000effdffff]
[    0.000000]   node   0: [mem 0x00000000effe0000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
[    0.000000]   node   0: [mem 0x00000003fc000000-0x00000003fc4fffff]
[    0.000000]   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
[    0.000000]   node   0: [mem 0x00000003fff00000-0x00000003ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000003ffffffff]
[    0.000000] cma: Reserved 64 MiB at 0x00000000dfe00000
[    0.000000] psci: probing for conduit method from ACPI.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 33 pages/cpu s95528 r8192 d31448 u135168
[    0.000000] pcpu-alloc: s95528 r8192 d31448 u135168 alloc=33*4096
[... 805 lines cut ...]
[    8.452935] sd 0:0:0:0: [sda] Attached SCSI removable disk
[    8.459141] XFS (loop0): Mounting V5 Filesystem cfa4379e-5cc8-11ee-971d-ed7fb74d6e96
[    8.461476] XFS (loop0): Ending clean mount
[    8.463671] systemd[1]: Mounted run-ephemeral.mount - /run/ephemeral.
[    8.556465] systemd[1]: Found device dev-disk-by\x2dlabel-fedora\x2dcoreos\x2d39.20231006.1.0.device - SD_MMC fedora-coreos-39.20231006.1.0.
[... cut ...]
[    8.718317] ostree-prepare-root[755]: Resolved OSTree target to: /sysroot/ostree/deploy/fedora-coreos/deploy/8462105c406cb0abfcccdaa5fae0c2acc2275d23bc0eb5e1a6a6241faff954e7.0
[    8.730884] ostree-prepare-root[755]: sysroot.readonly configuration value: 0 (fs writable: 0)
[    8.731692] ostree-prepare-root[755]: composefs: Mounting with no digest or signature check
[    8.732439] ostree-prepare-root[755]: composefs: optional support failed: No such file or directory
[    8.920957] systemd[1]: Finished coreos-liveiso-persist-osmet.service - Persist Osmet Files (ISO).
```

I'm still figuring out how to install `lspci` on a live CoreOS system.

byhat, 2023-10-10:
Outputs from `lspci`:

```
⬢[root@toolbox ~]# lspci -vvv
0000:01:00.0 Non-Volatile memory controller: Phison Electronics Corporation E16 PCIe4 NVMe Controller (rev 01) (prog-if 02 [NVM Express])
	Subsystem: Phison Electronics Corporation E16 PCIe4 NVMe Controller
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 25
	NUMA node: 0
	Region 0: Memory at 910000000 (64-bit, non-prefetchable) [size=16K]
	Capabilities: [80] Express (v2) Endpoint, MSI 00
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s unlimited, L1 unlimited
			ExtTag+ AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag+ PhantFunc- AuxPwr- NoSnoop+ FLReset-
			MaxPayload 128 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr+ NonFatalErr- FatalErr- UnsupReq+ AuxPwr- TransPend-
		LnkCap:	Port #1, Speed 16GT/s, Width x4, ASPM L1, Exit Latency L1 unlimited
			ClockPM- Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM Disabled; RCB 64 bytes, Disabled- CommClk-
			ExtSynch- ClockPM- AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 8GT/s (downgraded), Width x4
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Range ABCD, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp+ 10BitTagReq- OBFF Not Supported, ExtFmt+ EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
			 FRS- TPHComp- ExtTPHComp-
			 AtomicOpsCap: 32bit- 64bit- 128bitCAS-
		DevCtl2: Completion Timeout: 50us to 50ms, TimeoutDis- LTR- 10BitTagReq- OBFF Disabled,
			 AtomicOpsCtl: ReqEn-
		LnkCap2: Supported Link Speeds: 2.5-16GT/s, Crosslink- Retimer+ 2Retimers+ DRS-
		LnkCtl2: Target Link Speed: 16GT/s, EnterCompliance- SpeedDis-
			 Transmit Margin: Normal Operating Range, EnterModifiedCompliance- ComplianceSOS-
			 Compliance Preset/De-emphasis: -6dB de-emphasis, 0dB preshoot
		LnkSta2: Current De-emphasis Level: -6dB, EqualizationComplete+ EqualizationPhase1+
			 EqualizationPhase2+ EqualizationPhase3+ LinkEqualizationRequest-
			 Retimer- 2Retimers- CrosslinkRes: Upstream Port
	Capabilities: [d0] MSI-X: Enable- Count=9 Masked-
		Vector table: BAR=0 offset=00002000
		PBA: BAR=0 offset=00003000
	Capabilities: [e0] MSI: Enable- Count=1/8 Maskable- 64bit+
		Address: 0000000000000000  Data: 0000
	Capabilities: [f8] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=0mA PME(D0-,D1-,D2-,D3hot-,D3cold-)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [100 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [110 v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=10us PortTPowerOnTime=300us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [128 v1] Alternative Routing-ID Interpretation (ARI)
		ARICap:	MFVC- ACS-, Next Function: 0
		ARICtl:	MFVC- ACS-, Function Group: 0
	Capabilities: [1e0 v1] Data Link Feature <?>
	Capabilities: [200 v2] Advanced Error Reporting
		UESta:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UEMsk:	DLP- SDES- TLP- FCP- CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP- ECRC- UnsupReq- ACSViol-
		UESvrt:	DLP+ SDES- TLP- FCP+ CmpltTO- CmpltAbrt- UnxCmplt- RxOF- MalfTLP+ ECRC- UnsupReq- ACSViol-
		CESta:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		CEMsk:	RxErr- BadTLP- BadDLLP- Rollover- Timeout- AdvNonFatalErr+
		AERCap:	First Error Pointer: 00, ECRCGenCap- ECRCGenEn- ECRCChkCap+ ECRCChkEn-
			MultHdrRecCap- MultHdrRecEn- TLPPfxPres- HdrLogCap-
		HeaderLog: 20000001 0000000f 00000009 001f8000
	Capabilities: [300 v1] Secondary PCI Express
[... 63 lines cut ...]
		HeaderLog: 00000000 00000000 00000000 00000000
	Capabilities: [148 v1] Virtual Channel
		Caps:	LPEVC=0 RefClk=100ns PATEntryBits=1
		Arb:	Fixed- WRR32- WRR64- WRR128-
		Ctrl:	ArbSelect=Fixed
		Status:	InProgress-
		VC0:	Caps:	PATOffset=00 MaxTimeSlots=1 RejSnoopTrans-
			Arb:	Fixed- WRR32- WRR64- WRR128- TWRR128- WRR256-
			Ctrl:	Enable+ ID=0 ArbSelect=Fixed TC/VC=ff
			Status:	NegoPending- InProgress-
	Capabilities: [168 v1] Device Serial Number 01-00-00-00-68-4c-e0-00
	Capabilities: [178 v1] Transaction Processing Hints
		No steering table available
	Capabilities: [204 v1] Latency Tolerance Reporting
		Max snoop latency: 0ns
		Max no snoop latency: 0ns
	Capabilities: [20c v1] L1 PM Substates
		L1SubCap: PCI-PM_L1.2+ PCI-PM_L1.1+ ASPM_L1.2+ ASPM_L1.1+ L1_PM_Substates+
			  PortCommonModeRestoreTime=150us PortTPowerOnTime=150us
		L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
			   T_CommonMode=0us LTR1.2_Threshold=0ns
		L1SubCtl2: T_PwrOn=10us
	Capabilities: [21c v1] Vendor Specific Information: ID=0002 Rev=4 Len=100 <?>
	Kernel driver in use: r8169
```

mariobalanica, 2023-10-10:
Thanks, that looks good so far. What about the `pci` command in UEFI shell?

byhat, 2023-10-10:
Here you go:
[image]

mariobalanica, 2023-10-11:
Good. Will follow up with a proper fix soon.

mariobalanica, 2023-10-12:
Done.

@harryzz, this is the same issue that you had with a Samsung drive.

closed 2023-10-12 by mariobalanica