edk2-porting/edk2-rk3588 #92: Compile-time default for config table mode

milas, 2023-10-14:
Is it possible to control the default config table mode?

I've got a fork of Talos Linux for Rock 5 devices ([milas/rock5-talos](https://github.com/milas/rock5-talos)), and I'm looking to drop my custom extlinux code and use UEFI (which is what upstream requires).

Since I'm building the full image, I'd like to force it to expose the device tree instead of ACPI for use with the bundled BSP kernel + appropriate DTB (& maybe overlays).

I think it's hardcoded to default to ACPI mode:
https://github.com/edk2-porting/edk2-rk3588/blob/98e10085990a4363a03e0225ff48f1e0d7984b77/edk2-rockchip/Silicon/Rockchip/RK3588/Drivers/RK3588Dxe/RK3588DxeHii.vfr#L467

Would you be open to exposing that as a build-time flag?

mariobalanica, 2023-10-14:
This enables both Device Tree mode and overlay support by default:
```bash
./build.sh --device rock-5b --release Release --build-flags "--pcd gRK3588TokenSpaceGuid.PcdConfigTableMode=0x00000002 --pcd gRK3588TokenSpaceGuid.PcdFdtSupportOverrides=1"
```

Our build process also enables Secure Boot here: https://github.com/edk2-porting/edk2-rk3588/blob/master/.github/workflows/build.yml#L82-L90

But I would actually advise against bundling a custom firmware with public OS images, for several reasons:
* It sort of defeats the purpose of using UEFI in the first place: users shouldn't need to download an OS image specific to their board, but rather let the firmware abstract that aspect. As an OS vendor, this also saves you from having to deal with multiple images.

* Your images will only boot seamlessly from SD/eMMC. People who prefer USB, NVME or SATA still have to go through the process of flashing the firmware to SPI NOR or SD/eMMC.

* Firmware updates.

milas, 2023-10-15:
_Definitely_ understood re: bundling a custom bootloader.

To be honest, my hope is to have my Talos Linux fork slowly cease to exist 🙃 I'm actually looking at UEFI now because it would bring the fork a LOT closer to upstream. However, given the current state of my fork & BSP situation overall...I think having a bundled bootloader overall is the right call for the moment. That is, upstream Talos Linux is UEFI native (supports secure boot, has UKI mode now, etc), but currently bundles U-Boot, so it's more 1:1 for me if I can swap U-Boot with edk2.

Anyway, that's just more context from my end, I'm closing this as a resolved - I appreciate the work you've done and the quick response! ❤️

closed 2023-10-15 by milas