edk2-porting/edk2-rk3588 #93: Trouble booting: no misc partition

milas, 2023-10-15:
My boot is failing pretty quickly:
```plain
U-Boot SPL board init
U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 0
spl: mmc init failed with error: -123
Trying to boot from MMC1
No misc partition
spl: partition error
```

I believe disk is partitioned correctly.

It's worth noting that I'd first tried to use the latest Collabora U-Boot branch, and can get GRUB (via EFI) booting, which then starts the BSP kernel, but quickly freezes. I am also adapting this from a previous Rockchip U-Boot + extlinux configuration, so I'm fairly confident it's sane.

Here's what the image looks like:
```plain
Disk metal-rock_5b-arm64.raw: 4 GiB, 4294967296 bytes, 8388608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 745B9664-44EB-4D00-AFF1-63E65DC6566D
First usable LBA: 34
Last usable LBA: 8388574
Alternative LBA: 8388607
Partition entries starting LBA: 2
Allocated partition entries: 128
Partition entries ending LBA: 33

Device                     Start     End Sectors Type-UUID                            UUID                                 Name      Attrs
metal-rock_5b-arm64.raw1   32768  237567  204800 C12A7328-F81F-11D2-BA4B-00A0C93EC93B 4F147E21-51CF-47B2-95B8-DF97948EF77F EFI       
metal-rock_5b-arm64.raw2  237568 2285567 2048000 0FC63DAF-8483-4772-8E79-3D69D8477DE4 A2A1612A-781A-47C5-89AC-98A56CE1316D BOOT      
metal-rock_5b-arm64.raw3 2285568 2287615    2048 0FC63DAF-8483-4772-8E79-3D69D8477DE4 47D37DA7-A920-4209-9A96-2CD38531DA98 META      
metal-rock_5b-arm64.raw4 2287616 2492415  204800 0FC63DAF-8483-4772-8E79-3D69D8477DE4 C6AC394C-9953-486B-9A62-F4E37458CF37 STATE     
metal-rock_5b-arm64.raw5 2492416 8386559 5894144 0FC63DAF-8483-4772-8E79-3D69D8477DE4 46794A78-EAC2-48DE-802B-EA0D03B23720 EPHEMERAL 
```

---

One thing that IS odd is that after it can't find the misc partition, it starts spewing malloc/general allocation errors...I'm guessing that this might just be misleading as it's already failed to find the EFI partition to boot at this point from what I can tell.

<details>
<summary><strong>Complete serial log</strong></summary>
<pre>
DDR V1.12 52218f4949 cym 23/07/06-19:46:50
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
Manufacturer ID:0x6
CH0 RX Vref:27.5%, TX Vref:24.8%,25.8%
CH1 RX Vref:26.7%, TX Vref:24.8%,24.8%
CH2 RX Vref:27.9%, TX Vref:25.8%,24.8%
CH3 RX Vref:27.1%, TX Vref:24.8%,24.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 0
spl: mmc init failed with error: -123
Trying to boot from MMC1
No misc partition
spl: partition error
Trying fit image at 0x4000 sector
Not fit magic
Trying fit image at 0x5000 sector
Not fit magic
sys malloc pool space exhausted
alloc_read_gpt_entries: ERROR: Can't allocate lX bytes for GPT Entries
GPT: Failed to allocate memory for PTE
part_get_info_efi: *** ERROR: Invalid GPT ***
sys malloc pool space exhausted
alloc_read_gpt_entries: ERROR: Can't allocate lX bytes for GPT Entries
GPT: Failed to allocate memory for PTE
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
sys malloc pool space exhausted
alloc_read_gpt_entries: ERROR: Can't allocate lX bytes for GPT Entries
GPT: Failed to allocate memory for PTE
part_get_info_efi: *** ERROR: Invalid GPT ***
sys malloc pool space exhausted
alloc_read_gpt_entries: ERROR: Can't allocate lX bytes for GPT Entries
GPT: Failed to allocate memory for PTE
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
Trying to boot from MTD2
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
No misc partition
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
Trying fit image at 0x4000 sector
Not fit magic
Trying fit image at 0x5000 sector
Not fit magic
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
part_get_info_efi: *** ERROR: Invalid GPT ***
part_get_info_efi: *** ERROR: Invalid Backup GPT ***
SPL: failed to boot from all boot devices
### ERROR ### Please RESET the board ###
# Reset the board to bootrom #
</pre>
</details>

Does this look familiar?

This is on a Radxa Rock 5B, using v0.9.1 and compiling with `--release "Debug"` and capturing output from a USB-TTL cable.

mariobalanica, 2023-10-15:
How did you flash the firmware image?

For UEFI and OS on the same disk, you need to flash UEFI first and then create any additional partitions, without touching the 8 MB reserved one at the beginning.

milas, 2023-10-15:
I'm partitioning it on the same disk.

My EFI partition starts at sector 0x8000, so 16MiB offset (that's the offset Rockchip expects for U-Boot/extlinux, so I kept it here as it gives enough space).

My disk/image looks a lot like the Arch Linux linked in the releases:
```
Disk outputs/ArchLinuxARM-Rock_5B-minimal-UEFI.img: 8 GiB, 8589934592 bytes, 16777216 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 517653F0-336E-4F27-8327-05133F991745

Device                                           Start      End  Sectors  Size Type
outputs/ArchLinuxARM-Rock_5B-minimal-UEFI.img1   43008  1091583  1048576  512M EFI System
outputs/ArchLinuxARM-Rock_5B-minimal-UEFI.img2 1091584 16777182 15685599  7.5G Linux filesystem
```
(The EFI offset is 21MiB in this image, so a bit more, but that should be fine.)

---

Or for the same disk do I need it to be MBR with a protective 8MiB GPT partition? That is how the `RK3588_NOR_FLASH.img` is setup:
```
GPT PMBR size mismatch (32767 != 13696) will be corrected by write.
Disk out/edk2/RK3588_NOR_FLASH.img: 6.69 MiB, 7012864 bytes, 13697 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x00000000

Device                         Boot Start   End Sectors  Size Id Type
out/edk2/RK3588_NOR_FLASH.img1          1 13696   13696  6.7M ee GPT
```

I guess I assumed that since I actually have EDK2 @ 0x4000 and the U-Boot SPL is starting, that it'd understand a normal GPT partition table for the same disk...?

mariobalanica, 2023-10-15:
This is the layout expected by the firmware: https://github.com/edk2-porting/edk2-rk3588#flash-layout

milas, 2023-10-15:
Yeah, that's how my layout is with the exception that I'm creating a GPT partition table instead of using the MBR-based one (that originally comes from `rk3588_spi_nor_gpt.img`).

I guess that's likely the problem, I guess the SPL is looking for magic in the partition table so that it can chainload EDK2.

mariobalanica, 2023-10-15:
The generated image is GPT, not MBR. I had a quick look at `rk3588_spi_nor_gpt.img` and couldn't find anything special.
It's hard to tell what the issue with your image might be without having a look at it.

milas, 2023-10-15:
It looks that way:
```shell
$ diff -q <(dd if=./metal-rock_5b-arm64.raw bs=1K skip=$((0x00100000/1024)) count=$((0x00500000/1024))) <(dd if=./rock-5b_UEFI_Release_v0.9.1.img bs=1K skip=$((0x00100000/1024)) count=$((0x00500000/1024))) && echo $?
5120+0 records in
5120+0 records out
5242880 bytes (5.2 MB, 5.0 MiB) copied, 0.00873112 s, 600 MB/s
5120+0 records in
5120+0 records out
5242880 bytes (5.2 MB, 5.0 MiB) copied, 0.00835427 s, 628 MB/s
0
```

Here's an image: https://drive.google.com/file/d/15Wk1-h-xq4mDrXmO_TbVVhBen0gwDnBJ/view?usp=drive_link

Don't feel pressured to take a look! You've been a huge help already.

Anyway, I'll keep debugging this when I have time and update this issue if I figure out what's going on

mariobalanica, 2023-10-15:
Yeah, you're missing the first 8 MB reserved partition with first LBA at sector 2048 (FIT at 0x100000). After that one you can create ESP and the rest.

milas, 2023-10-28:
(D'oh, thanks for the tip about the `.dtb` - I messed up the path when moving things around!)

Gotcha on the 8MiB reserved partition, I had the data from sector 64 onwards but didn't realize that it actually cared about the _existence_ of a partition @ LBA 2048. I guess I'm too used to Rockchip magic sector addresses 🙃 

Adding that and rebuilding an image got me to the point of booting:
```
U-Boot SPL board init
U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 0
spl: mmc init failed with error: -123
Trying to boot from MMC1
No misc partition
Trying fit image at 0x800 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(b40ce543bb...) + OK
## Checking uboot 0x00200000 ... sha256(dbb66b4235...) + OK
## Checking fdt 0x00700000 ... sha256(c07f4a4d71...) + OK
## Checking atf-2 0x000f0000 ... sha256(f202de33d9...) + OK
## Checking atf-3 0xff100000 ... sha256(af08976afd...) + OK
## Checking optee 0x08400000 ... sha256(6193f4aac7...) + OK
## Checking nvdata 0x007c0000 ... OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 589.509 ms

INFO:    Preloader serial: 2

                             GNU GRUB  version 2.06

  Booting `A - Talos v1.5.2-2-g89d7e7ba4-dirty'
...
```

In case anyone else ends up here because you're building images from scratch, the key I was missing was that `fdisk` was not giving enough output, so `gdisk -l` was helpful:
```
Disk ./rock-5b_UEFI_Release_v0.9.1.img: 13697 sectors, 6.7 MiB
Sector size (logical): 512 bytes
Disk identifier (GUID): 25A6920B-1CE1-4A88-9FA7-72ECBF641DC7
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 32734
Partitions will be aligned on 2048-sector boundaries
Total free space is 16317 sectors (8.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048           18431   8.0 MiB     8300  uboot
```

I changed up my image to match:
```
Disk ./metal-rock_5b-arm64.raw: 8388608 sectors, 4.0 GiB
Sector size (logical): 512 bytes
Disk identifier (GUID): 23B5CE58-7116-4C87-9B04-8898F0D01B48
Partition table holds up to 128 entries
Main partition table begins at sector 2 and ends at sector 33
First usable sector is 34, last usable sector is 8388574
Partitions will be aligned on 2048-sector boundaries
Total free space is 18365 sectors (9.0 MiB)

Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048           18431   8.0 MiB     8300  uboot
   2           32768          237567   100.0 MiB   EF00  EFI
   3          237568         2285567   1000.0 MiB  8300  BOOT
   4         2285568         2287615   1024.0 KiB  8300  META
   5         2287616         2492415   100.0 MiB   8300  STATE
   6         2492416         8386559   2.8 GiB     8300  EPHEMERAL
```

For completeness, in `fdisk`, this looks like:
```
Disk metal-rock_5b-arm64.raw: 4 GiB, 4294967296 bytes, 8388608 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 23B5CE58-7116-4C87-9B04-8898F0D01B48

Device                     Start     End Sectors  Size Type
metal-rock_5b-arm64.raw1    2048   18431   16384    8M Linux filesystem
metal-rock_5b-arm64.raw2   32768  237567  204800  100M EFI System
metal-rock_5b-arm64.raw3  237568 2285567 2048000 1000M Linux filesystem
metal-rock_5b-arm64.raw4 2285568 2287615    2048    1M Linux filesystem
metal-rock_5b-arm64.raw5 2287616 2492415  204800  100M Linux filesystem
metal-rock_5b-arm64.raw6 2492416 8386559 5894144  2.8G Linux filesystem
```

Sadly, my BSP kernel freezes shortly after booting, which is the same issue I ran into the last time I tried EDK2 😭 

Thanks again for the guidance!

closed 2023-10-28 by milas