edk2-porting/edk2-rk3588 #94: orangepi 5 plus: `ESC` and `F1` not functional

beldzhang, 2023-10-15:
orange pi 5 plus 16GB, just got one week. nothing changed with SPI flash, don't know any info (version?date?) about it.
tested debian 11 from official website and it's work prefect.

with edk2 firmware every thing works like the README describe, except `ESC` and `F1`, whenever or whatever press or hold this keys, it's not functional.

the firmwares was write to a 4GB micro sd, via dd on linux.

1. tested 0.8 0.9 and 0.9.1,
2. tested usb2/usb3 ports,
3. tested with different keyboards

it's all the same.

anything maybe I doing wrong? or any way could try?

thanks a lot.

mariobalanica, 2023-10-15:
What if you boot without any OS drive connected? It should take you to the setup menu.

Do any LEDs on the keyboard light up?

beldzhang, 2023-10-15:
> What if you boot without any OS drive connected? It should take you to the setup menu.

there is a nvme drive installed, only one xfs partition used whole disk as data storage.

> Do any LEDs on the keyboard light up?

no light up, but cap lock and num lock are responsive.

if no other storage plugged, will reboot after progree bar finish then repeat from start
and tried a ventoy prepared disk on usb 3.0, it works and could load linux iso on it.

mariobalanica, 2023-10-15:
Can you try flashing UEFI to SPI NOR instead? 

So far it sounds like it's not able to save variables on the SD card, I would need [serial logs](https://github.com/edk2-porting/edk2-rk3588#advanced-troubleshooting) in this case.

beldzhang, 2023-10-17:
> Can you try flashing UEFI to SPI NOR instead?

currently still other test pending, will try later
 
> So far it sounds like it's not able to save variables on the SD card, I would need [serial logs](https://github.com/edk2-porting/edk2-rk3588#advanced-troubleshooting) in this case.

no that cable and no any experience of this kind debug yet ... sorry.

beldzhang, 2023-10-24:
flashed to SPI NOR, `ESC` key working, `F1` not

Crypto-Gi, 2023-11-13:
v0.9.1 Version, Added img file to sdcard, ESC and F1 not working

S199pWa1k9r, 2023-12-29:
@flucas1 

Are you sure that start UEFI comes from the SD card?
Maybe the SPI flash contains some kind of data?
Which USB port is your keyboard connected to?

This error may not appear if your SD card was not filled with zeros and already contained NV_VARIABLE_STORE with default values.

This error does not occur if you connect the UART TTL cable and enter the menu. UEFI settings and save the default values to an area on the SD card.

This error is very annoying, for example, on NanoPC-T6, which has only one USB3 port and you already have NVME installed with the operating system.
 
After a UEFI update, 
if the NV_VARIABLE_STORE area is cleared and filled with zeros then

https://github.com/edk2-porting/edk2-rk3588/blob/dc1d2558552a8ff556328027801729ee8abc1c82/edk2-rockchip/Silicon/Rockchip/RK3588/RK3588Base.dsc.inc#L38

—
   DEFINE ACPI_USB2_STATE_DISABLED = 0
   DEFINE ACPI_USB2_STATE_ENABLED = 1

   DEFINE USBDP_PHY_USB3_STATE_DISABLED = 0
   DEFINE USBDP_PHY_USB3_STATE_ENABLED = 1
—

And although the default value may be ENABLE (1), 
the saved data area has the value (0) - DISABLE 
and your USB keyboard will never work.

It will work only when the default values (or the values you specify) 
are saved to the settings saving area NV_VARIABLE_STORE  on the SD card.
This error can be fixed with a simple hack.

  DEFINE ACPI_USB2_STATE_DISABLED = 1
   DEFINE ACPI_USB2_STATE_ENABLED = 0

   DEFINE USBDP_PHY_USB3_STATE_DISABLED = 1
   DEFINE USBDP_PHY_USB3_STATE_ENABLED = 0

YOU NEED TO OVERRIDE and rebuild UEFI yourself.

someplaceguy, 2024-01-09:
For what it's worth, I've been using a keyboard connected to the USB-C port of my Orange Pi 5 Plus and the ESC key has been working fine with the 0.9.1 release, both when it was flashed onto the SD Card and now that it's flashed onto the SPI instead.

mariobalanica, 2024-01-11:
Fixed in 72385b183a943c9c3da46beca9fe222afec6393e

closed 2024-01-11 by mariobalanica

Victuar98, 2024-03-06:
Hi! There is an Orange Pi 5B(w/o SPI) + Debian 12 (bookworm)/6.1.43-rockchip-rk3588, the OS is installed on eMMC. 
I want, without affecting the internal media in any way, to boot other operating systems using a bootloader on an SD card (GRUB/GRUB2) with rootfs on a USB SSD/HDD. 
In the current configuration, the Orange Pi 5B does NOT boot from USB media. The download takes place from the SD card, but this option does not suit me in principle. An attempt to boot using the EDK2 bootloader (orangepi-5 UEFI v0.9.1 on the SD card) results in a cyclic reboot. The ESC key only works if you connect the keyboard when the progressbar is displayed, but the UEFI settings are not saved. The F1 key is not available. Attempt to compile u-boot-rockchip according to the instructions from https://radix.pro/devices/orange-pi5 Unfortunately, it was not successful. Errors - "Image 'simple-bin' is missing external blobs and is non-functional: rockchip-tpl... Some images are invalid" and "make: *** [Makefile:1124: .binman_stamp] Error 103
-bash: CROSS_COMPILE=/opt/toolchains/aarch64-RK358X-linux-glibc/1.9.8/bin/aarch64-rk358x-linux-gnu-: There is no such file or directory"
To be honest, I just don't know what to do next... Probably the question can be put like this - how do I make a bootable SD card with a GRUB bootloader, which will then transfer the download to a USB media or an embedded eMMC? Any tips and links to working(!) instructions will be gratefully accepted!)))

alanbach, 2024-05-05:
I got a new Orange Pi5 (v 1.2) and was having this same issue. A quick build from source fixed the issue (Built on Ubuntu 24.04 LTS - Noble):
```
$ sudo apt-get install git gcc g++ build-essential gcc-aarch64-linux-gnu iasl python3-pyelftools uuid-dev
$ git clone https://github.com/edk2-porting/edk2-rk3588.git
$ cd edk2-rk3588
$ git submodule update --init --recursive
$ ./build.sh --device orangepi-5 --release Release
$ sudo dd if=RK3588_NOR_FLASH.img of=/dev/mmcblk0
```
I think it is worth to produce a new binary release to make it easier for the community to use this. 

Kudos for maintaining this, it works great!!

alanbach, 2024-05-07:
A quick question: Is the CPU clock supposed to downscale to 861MHz if a non-DST kernel is being used? The UEFI on my Orange Pi 5 has CPU settings set to maximum and the settings are persistent (as they are supposed to be) however the speed does not seem to change:

```
$ sudo dmidecode | grep "Current Speed"
	Current Speed: 816 MHz
```

mariobalanica, 2024-05-07:
dmidecode gets its data from SMBIOS, which does not update dynamically and won't reflect the actual clocks per cluster.

If the settings have persisted then you can be certain it's running at maximum clock rate.

alanbach, 2024-05-07:
Thanks much for the clarification! 

Hopefully the upcoming kernel work with better Rockchip CPUFreq driver would improve things...

mariobalanica, 2024-05-07:
Make sure you're booting with a compatible Device Tree blob. ACPI is severely limited on Linux.

alanbach, 2024-05-07:
Do you think  *rk3588s-rock-5a.dtb* would be comptible with Orange Pi 5?

```
$ ls /lib/firmware/6.5.0-28-generic/device-tree/rockchip/ -l | grep rk3588s
-rwxr-xr-x 1 root root 55993 Apr  4 06:18 rk3588s-indiedroid-nova.dtb
-rwxr-xr-x 1 root root 46295 Apr  4 06:18 rk3588s-khadas-edge2.dtb
-rwxr-xr-x 1 root root 47308 Apr  4 06:18 rk3588s-rock-5a.dtb
```

mariobalanica, 2024-05-07:
No. OPi 5 support was added in a more recent kernel version, but I'm not up-to-date on the actual status.

alanbach, 2024-05-07:
You are correct, the 6.8 kernel has it! Now I have to back-port it :-D
```
~/tmp/tmp2/lib/firmware/6.8.0-32-generic/device-tree/rockchip$ ls -lh | grep orangepi-5
-rwxr-xr-x 1 alan alan 72K May  1 07:02 rk3588-orangepi-5-plus.dtb
-rwxr-xr-x 1 alan alan 64K May  1 07:02 rk3588s-orangepi-5.dtb
```