edk2-porting/edk2-rk3588 #96: DTB Override Location Clarification

zchef2k, 2023-10-25:
Can you explain the decision to require DTB overrides to exist as such:

> 
> The firmware will now look for overrides in the partition of a selected boot option / OS loader. In most cases this will be the first FAT32 EFI System Partition.
> 
> The base DTB must be located at \dtb\base\PLATFORM-DT-NAME.dtb.

For example, Fedora 39 includes rk3588 DTBs on the second partition of their raw image at '/dtb/rockchip/'. I have booted their image on the Radxa Rock 5b after appending the default GRUB entries with:

`devicetree (hd0,msdos2)/dtb/rockchip/rk3588-rock-5b.dtb`

It would be nice for some combination of the following to be true:

1. OS vendors decide on a default location for their included DTBs (if they haven't already) and for firmware to reliably look there.
2. Firmware be able to scan devices and partitions for matching DTBs.

So it's noted, I've not been able to boot any kernel reliably with the "firmware provided" DTB- at least not anything 6.0+.

Great work everyone, by the way. Thank you for your effort to advance this platform.

mariobalanica, 2023-10-26:
Hi,

> Can you explain the decision to require DTB overrides to exist as such:

To keep it simple. Since all other boot files are usually in the same file system as the bootloader image, it made the most sense to place DTBs there as well.

> For example, Fedora 39 includes rk3588 DTBs on the second partition of their raw image at '/dtb/rockchip/'. I have booted their image on the Radxa Rock 5b after appending the default GRUB entries with:
>
> devicetree (hd0,msdos2)/dtb/rockchip/rk3588-rock-5b.dtb

Yes, that's another way to do it.

> It would be nice for some combination of the following to be true:
> 
> 1. OS vendors decide on a default location for their included DTBs (if they haven't already) and for firmware to reliably look there.

Location of DTB overrides and such is not standardized in any way. The EBBR spec only requires firmware to provide a config table (either ACPI or DT) and that's all - anything else is up to the FW vendor to decide (or not).

I've added support for this mainly so that end-users (not OS vendors!) can boot newer kernels by simply placing an appropriate DTB in a well-known location expected by the firmware, without having to tweak or rely on Grub. As a bonus, it also allows for user overlays.

The plan is not to depend on this feature (which is not ideal regardless of the file path), but rather wait on mainline support of these boards to mature (many are not even supported at all), then update the firmware to provide new DTBs.

> 2. Firmware be able to scan devices and partitions for matching DTBs.

This would complicate things for no real benefit, in my opinion. What would the search criteria be? What's the expected behavior when multiple DTBs are found on different devices?

But we could expand the current search to SD/eMMC too, in case it's only used for firmware and OS is booted from another drive.

> So it's noted, I've not been able to boot any kernel reliably with the "firmware provided" DTB- at least not anything 6.0+.

The provided DTB is only compatible with Rockchip's legacy 5.10 kernel for now.

> Great work everyone, by the way. Thank you for your effort to advance this platform.

Thanks, appreciate it!

zchef2k, 2023-10-27:
While I've been a Linux user for the better part of 20 years, I've only recently begun to dig this deeply into the boot process of SBCs. Your responses help advance my understanding, thank you.

I look forward to contributing to this project as I am able.

closed 2024-01-05 by mariobalanica