edk2-porting/edk2-rk3588 #97: Support EFI_DT_FIXUP_PROTOCOL

soyflourbread, 2023-10-25:
It's awesome that device tree support is finally enabled. However, the current device tree override experience is far from optimal. Problems mentioned in #96 signify this issue.

Instead of patching each operating system to use the "correct" DTB path (which is difficult to enforce to each individual Linux distributions in practice), it makes more sense to have a UEFI protocol to patch/replace the existing device tree.

Fortunately, U-Boot already came up with (and implemented) a UEFI protocol for DTB patching/overrides: [EFI_DT_FIXUP_PROTOCOL](https://github.com/U-Boot-EFI/EFI_DT_FIXUP_PROTOCOL). It has the following advantages:

* Being a UEFI protocol.
* Supported by U-Boot, the most popular EFI implementation on ARM systems.
* Ability for the bootloader (e.g. systemd-boot) to provide different DTB files for each individual kernel version.
* Supported by systemd-boot.

If EFI_DT_FIXUP_PROTOCOL is implemented, there would be no need for hardcoded DTB override file paths in the UEFI. Instead. specifying a custom DTB would be as easy as specifying `devicetree` in your systemd-boot entry:
```
title   Linux
linux   /vmlinuz-<your kernel version>
initrd  /initramfs-<your kernel version>
devicetree /dtb-<your kernel version>/rockchip/rk3588-rock-5b.dtb
```

Supporting EFI_DT_FIXUP_PROTOCOL would pretty much solve all headaches with device tree compatibilities in EDK2 by moving it downstream into the bootloader/operating system layer. It would also result in a much more elegant user experience.

Googulator, 2023-10-25:
I don't think this is a good idea.

The RK3588 is a general purpose SoC, and the goal of this UEFI firmware is to enable it to function as such, i.e. to be able to run generic operating systems without requiring OS-level board enablement for each RK3588 board design.

To achieve this, Arm has published the SystemReady standard, which this UEFI intends to follow. It primarily targets the SystemReady ES profile, which requires ACPI - to support operating systems that target SystemReady IR instead, DTB support is also provided.

However, SystemReady, in all of its profiles, is predicated on the idea of the _firmware_ providing platform description (in the form of ACPI or DTB), and the HLOS interpreting that to discover platform hardware. Since DTB bindings still aren't backwards compatible between kernels (and even between distributions in some cases), we provide the current override mechanism as a workaround.

But that's all it is - a _workaround_ for what we consider a kernel bug, namely the inability or unwillingness to interpret DTBs and/or ACPI tables in a SystemReady-compliant way, and instead expecting that the kernel provides its own DTBs for each supported board, tailor-made to that kernel.

This idea that the kernel is essentially expected to internally know the intricate details of each platform and board it supports is firmly part of the _embedded_ mindset, wholly inappropriate for a general purpose operating system running on a general purpose platform.

For this reason, I can't support any proposal to facilitate routinely replacing the firmware-provided platform description with a kernel- or distribution-provided one.

mariobalanica, 2023-10-26:
Hi,

In addition to Googulator's great points and my comment on #96, 

> Instead of patching each operating system to use the "correct" DTB path (which is difficult to enforce to each individual Linux distributions in practice), it makes more sense to have a UEFI protocol to patch/replace the existing device tree.

OS vendors are not expected to follow what we've done here. It's just for end-users of this specific firmware.

Regarding EFI_DT_FIXUP_PROTOCOL:
* It only seems to serve as a way for the firmware to apply its own fix-ups to a *random* DTB passed by the bootloader. Since you can pass literally anything to the firmware, its hardcoded internal fix-ups may as well end up not matching your DTB, at worst corrupting it.

  You're probably thinking that it can still be used by the bootloader to override the firmware DTB. But there's already an UEFI boot service for that and the EFI_DT_INSTALL_TABLE flag in this protocol was removed for the same reason.

* It's not part of any official specification or even draft, despite it using the reserved EFI_ prefix.

So it really doesn't solve much and I'm not sure why it was even adopted by U-Boot and others in this state.

closed 2024-01-05 by mariobalanica