edk2-porting/edk2-rk3588 #98: Rockchip Legacy 5.10 Kernel Freezes on Boot

milas, 2023-10-28:
Thanks again for all the help in #93.

I'm now successfully hitting GRUB and starting the kernel, but it quickly freezes.

Over HDMI, I see the Radxa logo & progress bar, then GRUB, and then the boot starts:
```
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
```

Nothing more appears there.

Over serial output:
<details>
<summary>bootloader</summary>
<pre>
U-Boot SPL board init
U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
Trying to boot from MMC2
MMC: no card present
mmc_init: -123, time 0
spl: mmc init failed with error: -123
Trying to boot from MMC1
No misc partition
Trying fit image at 0x800 sector
## Verified-boot: 0
## Checking atf-1 0x00040000 ... sha256(b40ce543bb...) + OK
## Checking uboot 0x00200000 ... sha256(dbb66b4235...) + OK
## Checking fdt 0x00700000 ... sha256(c07f4a4d71...) + OK
## Checking atf-2 0x000f0000 ... sha256(f202de33d9...) + OK
## Checking atf-3 0xff100000 ... sha256(af08976afd...) + OK
## Checking optee 0x08400000 ... sha256(6193f4aac7...) + OK
## Checking nvdata 0x007c0000 ... OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 591.281 ms

INFO:    Preloader serial: 2

                             GNU GRUB  version 2.06

  Booting `A - Talos v1.5.2-2-g89d7e7ba4-dirty'

EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services and installing virtual address map...
</pre>
</details>

<details>
<summary>kernel</summary>
<pre>
[   35.282631] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[   35.282661] Linux version 5.10.110-milas-g5426ee10da67-dirty (root@buildkitsandbox) (aarch64-none-linux-gnu-gcc (GNU Toolchain for the A-profile Architecture 10.3-2021.07 (arm-10.29)) 10.3.1 20210621, GNU ld3
[   35.282907] earlycon: uart8250 at MMIO32 0x00000000feb50000 (options '')
[   35.286870] printk: bootconsole [uart8250] enabled
[   35.287625] efi: EFI v2.70 by EDK II
[   35.287967] efi: SMBIOS=0xefd90000 SMBIOS 3.0=0xea670000 MEMATTR=0xee7c4298 ACPI 2.0=0xea690018 MEMRESERVE=0xea452e98 
[   35.289016] efi: memattr: Unexpected EFI Memory Attributes table version 2
[   35.655013] ACPI: Early table checksum verification disabled
[   35.655567] ACPI: RSDP 0x00000000EA690018 000024 (v02 RKCP  )
[   35.656131] ACPI: XSDT 0x00000000EA69FE98 000074 (v01 RKCP   RK3588   00000000      01000013)
[   35.656959] ACPI: FACP 0x00000000EA69FB98 000114 (v06 RKCP   RK3588   00000000 EDK2 00000000)
[   35.657789] ACPI: DSDT 0x00000000EA697518 000842 (v02 RKCP   RK3588   00000002 INTL 20200925)
[   35.658611] ACPI: DBG2 0x00000000EA69FA98 00005C (v00 RKCP   RK3588   00000000 EDK2 00000000)
[   35.659434] ACPI: GTDT 0x00000000EA69FD18 000060 (v02 RKCP   RK3588   00000000 EDK2 00000000)
[   35.660255] ACPI: APIC 0x00000000EA69E998 0002D4 (v04 RKCP   RK3588   00000000 EDK2 00000000)
[   35.661076] ACPI: MCFG 0x00000000EA69F818 00007C (v01 RKCP   RK3588   00000000 EDK2 00000000)
[   35.661897] ACPI: PPTT 0x00000000EA69ED98 000220 (v01 RKCP   RK3588   00000000 EDK2 00000000)
[   35.662716] ACPI: SPCR 0x00000000EA69FE18 000050 (v02 RKCP   RK3588   00000000 EDK2 00000000)
[... 69 lines cut ...]
[   36.220985] Console: colour dummy device 80x25
[   36.221449] printk: console [tty1] enabled
[   36.221850] printk: bootconsole [uart8250] disabled
I/TC: Secondary CPU 1 initializing
I/TC: Secondary CPU 1 switching to normal world boot
I/TC: Secondary CPU 2 initializing
I/TC: Secondary CPU 2 switching to normal world boot
I/TC: Secondary CPU 3 initializing
I/TC: Secondary CPU 3 switching to normal world boot
I/TC: Secondary CPU 4 initializing
I/TC: Secondary CPU 4 switching to normal world boot
I/TC: Secondary CPU 5 initializing
I/TC: Secondary CPU 5 switching to normal world boot
I/TC: Secondary CPU 6 initializing
I/TC: Secondary CPU 6 switching to normal world boot
I/TC: Secondary CPU 7 initializing
I/TC: Secondary CPU 7 switching to normal world boot
</pre>
</details>

My device is a Radxa Rock 5B, and I'm using the Radxa kernel sources which are based off Rockchip legacy 5.10 rkr3.4.

Nothing really seems to be going obviously wrong, but it just halts at this point and the activity LED remains solid, so nothing seems to be happening.

Is this something you've seen before?

mariobalanica, 2023-10-28:
You're booting in ACPI mode rather than DT.

closed 2024-01-05 by mariobalanica

avi951, 2024-07-10:
Hello, @mariobalanica
Can you explain how to solve if the kernel freezes in between?

```
[    2.278904][    T1] mpp_vdpu1 fdb51000.avsd-plus: Adding to iommu group 1
[    2.279138][    T1] mpp_vdpu1 fdb51000.avsd-plus: probe device
[    2.279249][    T1] mpp_vdpu1 fdb51000.avsd-plus: reset_group->rw_sem_on=0
[    2.279257][    T1] mpp_vdpu1 fdb51000.avsd-plus: reset_group->rw_sem_on=0
[    2.279389][    T1] mpp_vdpu1 fdb51000.avsd-plus: probing finish
[    2.279728][    T1] mpp_vdpu2 fdb50400.vdpu: Adding to iommu group 1
[    2.279766][    T1] mpp_vdpu2 fdb50400.vdpu: probe device
[    2.279851][    T1] mpp_vdpu2 fdb50400.vdpu: reset_group->rw_sem_on=0
[    2.279856][    T1] mpp_vdpu2 fdb50400.vdpu: reset_group->rw_sem_on=0
[    2.279957][    T1] mpp_vdpu2 fdb50400.vdpu: probing finish
[    2.280119][    T1] mpp_vepu2 jpege-ccu: probing start
[    2.280124][    T1] mpp_vepu2 jpege-ccu: probing finish
[    2.280220][    T1] mpp_vepu2 fdb50000.vepu: Adding to iommu group 1
[    2.280256][    T1] mpp_vepu2 fdb50000.vepu: probing start
[    2.280334][    T1] mpp_vepu2 fdb50000.vepu: reset_group->rw_sem_on=0
[    2.280339][    T1] mpp_vepu2 fdb50000.vepu: reset_group->rw_sem_on=0
[    2.280463][    T1] mpp_vepu2 fdb50000.vepu: probing finish
[    2.280542][    T1] mpp_vepu2 fdba0000.jpege-core: Adding to iommu group 5
[    2.280628][    T1] mpp_vepu2 fdba0000.jpege-core: probing start
[    2.280727][    T1] mpp_vepu2 fdba0000.jpege-core: attach ccu success
[    2.280832][    T1] mpp_vepu2 fdba0000.jpege-core: probing finish
[    2.280896][    T1] mpp_vepu2 fdba4000.jpege-core: Adding to iommu group 6
[    2.280974][    T1] mpp_vepu2 fdba4000.jpege-core: probing start
[    2.281066][    T1] mpp_vepu2 fdba4000.jpege-core: attach ccu success
[    2.281168][    T1] mpp_vepu2 fdba4000.jpege-core: probing finish
[    2.281232][    T1] mpp_vepu2 fdba8000.jpege-core: Adding to iommu group 7
[    2.281308][    T1] mpp_vepu2 fdba8000.jpege-core: probing start
[    2.281400][    T1] mpp_vepu2 fdba8000.jpege-core: attach ccu success
[    2.281504][    T1] mpp_vepu2 fdba8000.jpege-core: probing finish
[    2.281568][    T1] mpp_vepu2 fdbac000.jpege-core: Adding to iommu group 8
[    2.281648][    T1] mpp_vepu2 fdbac000.jpege-core: probing start
[    2.281738][    T1] mpp_vepu2 fdbac000.jpege-core: attach ccu success
[    2.281839][    T1] mpp_vepu2 fdbac000.jpege-core: probing finish
[    2.282076][    T1] mpp-iep2 fdbb0000.iep: Adding to iommu group 9
[    2.282160][    T1] mpp-iep2 fdbb0000.iep: probe device
[    2.282275][    T1] mpp-iep2 fdbb0000.iep: allocate roi buffer failed
[    2.282375][    T1] mpp-iep2 fdbb0000.iep: probing finish
[    2.282550][    T1] mpp_jpgdec fdb90000.jpegd: Adding to iommu group 4
[    2.282714][    T1] mpp_jpgdec fdb90000.jpegd: probe device
[    2.282922][    T1] mpp_jpgdec fdb90000.jpegd: probing finish
[    2.283300][    T1] mpp_rkvdec2 fdc30000.rkvdec-ccu: rkvdec-ccu, probing start
[    2.283357][    T1] mpp_rkvdec2 fdc30000.rkvdec-ccu: ccu-mode: 1
[    2.283361][    T1] mpp_rkvdec2 fdc30000.rkvdec-ccu: probing finish
[    2.283461][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: Adding to iommu group 12
[    2.283748][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: rkvdec-core, probing start
[    2.283860][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_a is not found!
[    2.283864][    T1] rkvdec2_init:1022: No niu aclk reset resource define
[    2.283870][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: shared_niu_h is not found!
[    2.283874][    T1] rkvdec2_init:1025: No niu hclk reset resource define
[    2.283893][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: no regulator, devfreq is disabled
[    2.283948][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: core_mask=00010001
[    2.283953][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: attach ccu as core 0
[    2.284190][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: sram_start 0x00000000ff001000
[    2.284194][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: rcb_iova 0x00000000fff00000
[    2.284199][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: sram_size 491520
[    2.284203][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: rcb_size 1048576
[    2.284209][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: min_width 512
[    2.284216][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: rcb_info_count 20
[    2.284220][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [136, 24576]
[    2.284225][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [137, 49152]
[    2.284229][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [141, 90112]
[    2.284233][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [140, 49152]
[    2.284237][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [139, 180224]
[    2.284241][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [133, 49152]
[    2.284245][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [134, 8192]
[    2.284249][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [135, 4352]
[    2.284253][    T1] mpp_rkvdec2 fdc38100.rkvdec-core: [138, 13056]
```