edk2-porting/edk2-rk3588 #100: Unable to boot Fedora with LUKS

NN708, 2023-11-02:
OS: Fedora Workstation 39 Beta (Kernel 6.5.2)
Board: Rock 5B
Firmware: v0.9.1 (ACPI Mode)

After successfully installing Fedora with LUKS from live ISO, it get stuck on booting:
```
EFI stub: Decompressing Linux Kernel...
EFI stub: EFI_RNG_PROTOCOL unavailable
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```
If installed without LUKS encryption, Fedora can boot normally.

Googulator, 2023-11-02:
> EFI stub: Using DTB from configuration table

That sounds like you're not actually using ACPI - either Fedora overrode it with a distro-provided DTB, or you have DTB enabled in the configuration.

NN708, 2023-11-02:
> That sounds like you're not actually using ACPI - either Fedora overrode it with a distro-provided DTB, or you have DTB enabled in the configuration.

How can I determine whether I’m using ACPI or not, or is there a way to force the OS to boot using ACPI?

NN708, 2023-12-17:
Tried Alpine Linux 3.19 (Kernel 6.6.7) and had the same issue, so it shouldn't be related to distro.

adam900710, 2023-12-19:
You need to connect to the TTL to give enough kernel message to pin down the error.

NN708, 2023-12-20:
> You need to connect to the TTL to give enough kernel message to pin down the error.

Thanks for the information! It turns out that the OS outputs the passphrase prompt to the serial port, not to the screen. Is there any option to make the system redirect the output to the screen by default?

adam900710, 2023-12-20:
I believe that's due to the default `console=` command line, the default console is the serial output.

I'm not sure what pts the HDMI is, as I mostly use my board as a headless server.
But after your locked in (and if it's not switching to GUI greeter), check the login screen which should show something like `(ttyX)`, then append `console=ttyX` to the kernel cmdline.

Another solution is to unlock your LUKS rootfs remotely using an ssh server inside the initramfs:

https://wiki.archlinux.org/title/Dm-crypt/Specialties#Remote_unlocking_of_root_(or_other)_partition

For fedora, I'm pretty sure there would be similar dracut hooks to do the same thing. (A quick google search leads me to dracut-crypt-ssh: https://github.com/dracut-crypt-ssh/dracut-crypt-ssh

NN708, 2023-12-20:
Thank you! I read the [Linux Kernel documentation](https://docs.kernel.org/admin-guide/serial-console.html) and it says that the foreground virtual console should be `tty0`. After adding `console=tty0` to the kernel cmdline, the output is now displayed on the screen.

However, what confuses me is that the documentation also says:
> If no console device is specified, the first device found capable of acting as a system console will be used. At this time, the system first looks for a VGA card and then for a serial port. So if you don't have a VGA card in your system the first serial port will automatically become the console.

The OS does not specify the `console` option by default, so it should use `tty0`. Why is the output on the serial port using this UEFI firmware?

adam900710, 2023-12-20:
I guess because of the sequence of initialization.

As ttyS0 get initialized very early, and the one on HDMI may be much later due to the whole DSDT layout?
Or due to the extra initialization needed for HDMI output?

Anyway you can close the case for now I guess, since it's not really a bug in the UEFI firmware.

closed 2023-12-20 by NN708