edk2-porting/edk2-rk3588 #105: Missing NvStorage GUID

izissise, 2023-11-24:
Hello,
I think there is a missing data section in https://github.com/edk2-porting/edk2-rk3588/blob/master/edk2-rockchip/Silicon/Rockchip/RK3588/RK3588.fdf#L53

Other fdf that use nvstorage that I could find usually define this section with the nvstorage guid, example https://github.com/tianocore/edk2/blob/master/ArmVirtPkg/VarStore.fdf.inc#L21

This useful for external tools to find nvdata section https://gitlab.com/kraxel/virt-firmware/-/blob/master/virt/firmware/varstore/edk2.py?ref_type=heads#L26

mariobalanica, 2023-11-24:
We don't include those headers in the final image in order to prevent overwriting them on flash updates. They are generated at runtime if needed.

izissise, 2023-11-24:
Thanks for the quick response!, Could you point me where the header is generated?, I'd like to generate it before the first flash

mariobalanica, 2023-11-24:
Headers are installed here: https://github.com/edk2-porting/edk2-rk3588/blob/master/edk2-rockchip/Silicon/Rockchip/Drivers/RkFvbDxe/RkFvbDxe.c#L935

> I'd like to generate it before the first flash

Then you need to do the FDF changes and revert https://github.com/edk2-porting/edk2-rk3588/commit/1bb2861929651850b762425ad3f20f1ee72e8d67.

izissise, 2023-11-29:
In the end I went another way, my goal for now is only to choose the default boot entry. I patched Bsdentry.c to achieve that, the patch can be found here https://edk2.groups.io/g/devel/message/111811?p=%2C%2C%2C20%2C0%2C0%2C0%3A%3ACreated%2C%2C%2C20%2C2%2C20%2C102851815%2Cd%253D3&d=3 if is useful for someone .

Thanks for the help