edk2-porting/edk2-rk3588 #115: Orange Pi 5 Plus - EMMC Not Working

PBXForums, 2024-01-14:
I have just installed Proxmox on to an Orange Pi 5 Plus.

Everything appears to be working except emmc. Nor was there an option to install to emmc.

mariobalanica, 2024-01-14:
Linux does not support the SD/eMMC controllers in ACPI mode. You need to use a compatible kernel and device tree combo.

PBXForums, 2024-01-14:
Thanks for the reply, I'll have a read up.

jzbor, 2024-01-18:
Hi,
From what I can tell, Linux Mainline itself does [support eMMC](https://gitlab.collabora.com/hardware-enablement/rockchip-3588/notes-for-rockchip-3588/-/blob/main/mainline-status.md). However I can't figure out where to get the appropriate DeviceTrees and how to apply them (do I just put them at the appropriate path in the boot partition?).

Would you consider also shipping a version with mainline DeviceTrees enabled by default instead of the SDK ones?

mariobalanica, 2024-01-18:
The process to apply compiled DTBs is described in the readme: https://github.com/edk2-porting/edk2-rk3588#device-tree-configuration

Mainline is under active development and only a few devices have some form of support. We do plan to ship mainline DTBs, but it's a bit early for that.