edk2-porting/edk2-rk3588 #118: Orange Pi 5 Plus - serial port not working in FreeBSD14 and 13.2

rdunkle34, 2024-01-21:
v0.9.1-19-gcc77cee8
when I boot efi the serial ports works. The speed can be changed from the efi firmware.  The terminal works OK up to the point where the system checks the processors.  I see this on console during boot:
Setting currdev to disk0p2:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
console comconsole failed to initialize
The processors are identified and then serial output stops.
Output looks like this:
output to serial always stops here-->   I/TC: Secondary CPU 7 switching to normal world boot.   
Speed can be set to 150000 or 115200 in efi.  the same results.
 dmesg | grep uart
uart0: <Non-standard ns8250 class UART with FIFOs> iomem 0xfeb50000-0xfeb50fff irq 21 on acpi0
---
On FreeBSD normally the next line of the dmesg output will show that the uart0 is then configured.
It normally looks like this:
uart0: console (115200,n,8,1)
---
I believe there is something missing in the DTB for the serial port on the Orange Pi 5 Plus.

mariobalanica, 2024-01-21:
FreeBSD is hardcoding a 200 MHz ref clock here: https://github.com/freebsd/freebsd-src/blob/main/sys/dev/uart/uart_dev_ns8250.c#L461, while on this platform the clock is running at 24 MHz.

So it does not look like a firmware issue. The UART console works fine in NetBSD, although it's being reconfigured at 115200 baud.

> I believe there is something missing in the DTB for the serial port on the Orange Pi 5 Plus.
DTB is only usable for Rockchip Linux. For anything else ACPI should be used, and it's the default choice.

jcmichot, 2024-05-16:
What is inside your loader.conf ?

With my Orange Pi5 plus, i have EDK2 0.9.1-7 in SPI, i boot on Nvme and loader.conf contain:

dtbfile_load="YES"
dtbfile_type="dtb"
dtbfile_name="rk3588.dtb"
kern.cfg.order="acpi"
boot_serial="YES"
boot_multicons="YES"
console="eficom"
if_re_load="YES"
if_re_name="/boot/modules/if_re.ko"
hw.re.max_rx_mbuf_sz="2048"

root@Orange-Pi-5-PlusBSD:~ # uname -a
FreeBSD Orange-Pi-5-PlusBSD 14.0-RELEASE-p6 FreeBSD 14.0-RELEASE-p6 #0: Tue Mar 26 20:37:02 UTC 2024     root@arm64-builder.daemonology.net:/usr/obj/usr/src/arm64.aarch64/sys/GENERIC arm64

I have both console AND serial usable. (cu -s 1500000 -l /dev/cuaU0) with USB Serial connected to OrangePi5+ serial PIN.

S199pWa1k9r, 2024-05-16:
In my opinion, it should be written like this
console="efi,comconsole"

I managed to get a uart console only after this patch
https://reviews.freebsd.org/D40407

jcmichot, 2024-05-16:
You are right, but with console="efi,comconsole" i get the message "console comconsole failed to initialize"
but in both case the serial console still working ;-)

Consoles: EFI console  
    Reading loader env vars from /efi/freebsd/loader.env
Setting currdev to disk0p1:
FreeBSD/arm64 EFI loader, Revision 1.1

   Command line arguments: loader.efi
   Image base: 0xe9e90000
   EFI version: 2.70
   EFI Firmware: EDK II (rev 1.00)
   Console: efi (0x1000)
   Load Path: \EFI\BOOT\BOOTAA64.EFI
   Load Device: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/NVMe(0x1,32-D0-A0-31-D9-
38-25-00)/HD(1,GPT,B0291D42-111C-11EF-B896-C0742BFE6EDC,0x8000,0x19000)
   BootCurrent: 0003
   BootOrder: 0003[*] 0004 0005 0006 0007 0008 0009 000a 000b 0002 000c 0000
   BootInfo Path: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/NVMe(0x1,32-D0-A0-31-D
9-38-25-00)
Ignoring Boot0003: Only one DP found
Trying ESP: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/NVMe(0x1,32-D0-A0-31-D9-38-2
5-00)/HD(1,GPT,B0291D42-111C-11EF-B896-C0742BFE6EDC,0x8000,0x19000)
Setting currdev to disk0p1:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/NVMe(0x1,32-D0-A0-31-D9-38-25-00
)/HD(2,GPT,B02DDF86-111C-11EF-B896-C0742BFE6EDC,0x21000,0x2000000)
Setting currdev to disk0p2:
Trying: PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x0,0x0)/NVMe(0x1,32-D0-A0-31-D9-38-25-00
)/HD(3,GPT,B032790A-111C-11EF-B896-C0742BFE6EDC,0x2021000,0x726E5D80)
Setting currdev to zfs:zroot/ROOT/default:
Loading /boot/defaults/loader.conf
Loading /boot/defaults/loader.conf
Loading /boot/device.hints
Loading /boot/loader.conf
console comconsole failed to initialize
Loading /boot/loader.conf.local

I have more problem with USB2, after FreeBSD boot USB2 are unusable, before FreeBSD boot, USB2 can be used in EDK2 with Keyboard but you loose the keyboard after OS boot :(
Hopefully after boot USB3 work perfectly with keyboard, hub or usb hard disk.

mariobalanica, 2024-05-16:
Use the latest build artifact on Github to get USB 2 working.

S199pWa1k9r, 2024-05-16:
This won't work because FreeBSD doesn't have a generic_ohci_acpi driver. Here's the patch
https://personalbsd.org/download/ForFriends/

closed 2024-06-22 by mariobalanica

rdunkle34, 2025-03-16:
The serial port/console is now working.
A firmware update for the UEFI corrected the serial port. 
This is the build I tested with:
https://github.com/edk2-porting/edk2-rk3588/commit/f8af61cbe1fa168a557a7176e68ff7da1af29444

The relevant lines in the /boot.loader.conf
boot_multicons="YES"
console="comconsole,efi"

The efi video console and the serial port console work.
Default serial port is connected to a usb-serial CH340 device.  The CH340 has 3 wires connected to the board as shown in the Orange Pi User Manual.
Putty setting is: 1500000, 8,1,none,XON/XOFF

This is FreeBSD 14.2-STABLE stable/14-09def8d654e7 GENERIC arm64