edk2-porting/edk2-rk3588 #119: 0.9.1 can not load win11 install iso, 0.9 is ok

alecthw, 2024-01-21:


mariobalanica, 2024-01-21:
What are you booting the ISO from?

I cannot reproduce this and there wasn't any change in v0.9.1 that could've broken Windows support.

alecthw, 2024-01-22:
ISO and installation built by Raspberry imager tool both not ok. The circle loading run a few seconds， then it stop loading.
Waiting a long time， no reaction……

mariobalanica, 2024-01-25:
Btw, Windows is expected to freeze if you have an NVME drive connected. A workaround is available in the Discord server.

v0.9.1 fixed a PCIe enumeration bug, so the reason why v0.9 works is likely because it didn't even detect your NVME drive to begin with.

alecthw, 2024-01-26:
> Btw, Windows is expected to freeze if you have an NVME drive connected. A workaround is available in the Discord server.
> 
> v0.9.1 fixed a PCIe enumeration bug, so the reason why v0.9 works is likely because it didn't even detect your NVME drive to begin with.

Thank you very much. I'll try it later.

closed 2024-02-15 by mariobalanica