edk2-porting/edk2-rk3588 #123: Is it possible to have a SATA drive + 4xSATA card + boot?

baxi10, 2024-02-15:
Is it possible to run Linux in the configuration? PCIe card with 4 SATA ports connected to Orange PI 5 M2 via adapter. A SATA drive is connected to the PCIe card. In this configuration, the system does not boot from the SATA disk. In EDK2 the disk is visible. Maybe there is some configuration that allows booting

mariobalanica, 2024-02-15:
What's the output of the `pci` command in the UEFI Shell?

baxi10, 2024-02-15:
[image]

mariobalanica, 2024-02-15:
What Linux distro are you trying to boot?

baxi10, 2024-02-15:
There is Armbian on the connected disk. I also tried the distribution from the link https://ftp.radix.pro/radix/system/releases/1.9.367/rk358x-glibc/orange-pi5/
and the description https://radix.pro/devices/orange-pi5/ doesn't work to.

mariobalanica, 2024-02-15:
EDK2 does not support U-boot images. You need to use Grub for them and add the boot entry manually.

I'm doing most of my Linux testing with official Ubuntu Server and Fedora UEFI images in ACPI mode.

baxi10, 2024-02-15:
OK, but Orange PI boot from SD Card.

mariobalanica, 2024-02-15:
What is it booting from SD card?

baxi10, 2024-02-15:
OK, thanks for the tip, so I will check on another distribution.

baxi10, 2024-02-16:
All ok, boot form sata disk connected via PCIe card.
Working correctly with Armbian from https://github.com/armbian/build/pull/5900

closed 2024-02-16 by baxi10

rpardini, 2024-03-05:
Yeah @mariobalanica I forgot to say something about https://github.com/armbian/build/pull/5900 - in general we've good feedback, but a bit short on time to push it a bit further and maybe replace u-boot eventually.

 I tried to add an additional GRUB entry that does some USB Gadget "UMS mode" for all disks in the initrd (inspired by [systemd's storagetm](https://www.freedesktop.org/software/systemd/man/latest/systemd-storagetm.html), but using multiple usb-gadgets, one for each disk; see also [diskomator](https://github.com/poettering/diskomator) - Could you suggest some edk2/EFI-level alternative? Use-case is mostly flashing the eMMC, but also NVMe if possible...