edk2-porting/edk2-rk3588 #124: Support for iCore-3588Q

dickolsson, 2024-02-16:
Hello, thanks for the amazing work on the rk3588 porting effort!

I would love support for the [Firefly iCore-3588Q](https://en.t-firefly.com/product/core/icore3588q) system-on-module.

The primary reason for wanting this, is that the open hardware computer [MNT Reform](https://mntre.com) is developing support for this module. I would love to run this laptop with EDK2! Here are a couple of references for MNT's development of this module:

* KiCAD files: https://source.mnt.re/reform/mnt-reform-rk3588-som
* Recent update: https://mastodon.social/@mntmn/111881004143691367

While the iCore-3588Q module for MNT Reform is still under heavy development, porting for this SOM is probably best done with the reference carrier board [Firefly AIO-3588Q](https://en.t-firefly.com/product/industry/aio3588q).

I would be open to buying/sending an AIO-3588Q board to one of the porting developers here, if that would help.

mariobalanica, 2024-02-17:
Hi! Please drop me an email at the address on the profile.

idigdoug, 2024-03-27:
AIO-3588Q platform added.

closed 2024-03-27 by mariobalanica

dickolsson, 2024-03-28:
@idigdoug and @mariobalanica Thanks a lot for adding initial support for this! Incredible stuff 😊