edk2-porting/edk2-rk3588 #128: Orange Pi5B + EDK2 booting E and beyond

Victuar98, 2024-03-16:
Hello to the esteemed community! There is an Orange Pi 5B(w/o SPI) + Debian 12 (bookworm)/6.1.43-rockchip-rk3588, the OS is installed on eMMC. I want to boot other operating systems from a USB SSD/HDD without affecting the internal media in any way. 
In the current configuration, the Orange Pi 5B does NOT boot from USB media. The OS (Armbian, Ubuntu) is loaded from the SD card, but this option does not suit me in principle.  
The firmware EDK 2 0.9.1 for Orange Pi 5 (w/o letter!) written to the SD card is cyclically rebooted, the ESC and F1 keys do not work. 
According to the instructions from https://github.com/edk2-porting/edk2-rk3588 a modified firmware RK3688_NOR_FLASH.img was assembled, where the values of ACPI_USB2_STATE_... and USBDP_PHY_USB3_STATE_... were changed.
After that, the u-boot menu became available (F1 is still not available), but I cannot save the changed values of the parameters, for example, the ACPI / Device Tree is changed - F10 does not work. The UART connection is working, the menu is available, but the changes are not saved either. In addition, there are some errors (see below)
Questions - how do I get F10 to work and what should I do next so that EDK2 passes the download on to boot from a USB SSD/HDD or eMMC? Thanks to all who answered!)))

UART Debug:
DDR d5483af87d cym 23/11/23-16:15:24,fwver: v1.15
LPDDR4X, 2112MHz
channel[0] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[1] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[2] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
channel[3] BW=16 Col=10 Bk=8 CS0 Row=17 CS1 Row=17 CS=2 Die BW=8 Size=4096MB
Manufacturer ID:0x6
CH0 RX Vref:29.7%, TX Vref:22.8%,21.8%
CH1 RX Vref:27.9%, TX Vref:22.8%,21.8%
CH2 RX Vref:30.1%, TX Vref:20.8%,20.8%
CH3 RX Vref:30.1%, TX Vref:22.8%,21.8%
change to F1: 528MHz
change to F2: 1068MHz
change to F3: 1560MHz
change to F0: 2112MHz
out
U-Boot SPL board init
U-Boot SPL 2017.09-orangepi (Feb 03 2024 - 01:08:23)
Trying to boot from MMC1
Trying fit image at 0x800 sector
'## Verified-boot: 0
'## Checking atf-1 0x00040000 ... sha256(b40ce543bb...) + OK
'## Checking uboot 0x00200000 ... sha256(37dac54871...) + OK
'## Checking fdt 0x00700000 ... sha256(c07f4a4d71...) + OK
'## Checking atf-2 0x000f0000 ... sha256(f202de33d9...) + OK
'## Checking atf-3 0xff100000 ... sha256(af08976afd...) + OK
'## Checking optee 0x08400000 ... sha256(6193f4aac7...) + OK
'## Checking nvdata 0x007c0000 ... OK
Jumping to U-Boot(0x00200000) via ARM Trusted Firmware(0x00040000)
Total: 829.985 ms

INFO:    Preloader serial: 2
NOTICE:  BL31: v2.3():v2.3-616-gdc1125f48:finley.xiao
NOTICE:  BL31: Built : 16:41:17, Jul 13 2023
INFO:    spec: 0x13
INFO:    ext 32k is valid
INFO:    ddr: stride-en 4CH
INFO:    GICv3 without legacy support detected.
INFO:    ARM GICv3 driver initialized in EL3
INFO:    valid_cpu_msk=0xff bcore0_rst = 0x0, bcore1_rst = 0x0
INFO:    system boots from cpu-hwid-0
INFO:    idle_st=0x21fff, pd_st=0x11fff9, repair_st=0xfff70001
ERROR:   dfs get fsp_params[0] error, 0xfead0004 != 0xfead0003
ERROR:   dfs get fsp_params[1] error, 0x1111 != 0xfead0003
ERROR:   dfs get fsp_params[2] error, 0x0 != 0xfead0003
ERROR:   dfs get fsp_params[3] error, 0x61251520 != 0xfead0003
ERROR:   loader&trust unmatch!!! Please update trust if need enable dmc
INFO:    BL31: Initialising Exception Handling Framework
INFO:    BL31: Initializing runtime services
INFO:    BL31: Initializing BL32
I/TC:
I/TC: OP-TEE version: 3.13.0-698-g7f1ea6d6e #hisping.lin (gcc version 10.2.1 20201103 (GNU Toolchain for the A-profile Architecture 10.2-2020.11 (arm-10.16))) #53 Wed Apr 19 14:30:22 CST 2023 aarch64
I/TC: Primary CPU initializing
I/TC: Primary CPU switching to normal world boot
INFO:    BL31: Preparing for EL3 exit to normal world
INFO:    Entry point address = 0x200000
INFO:    SPSR = 0x3c9
UEFI firmware (version v0.9.1-30-g3e5a3c8c built at 13:20:27 on Mar 11 2024)

mariobalanica, 2024-03-16:
Can you try the latest build from https://github.com/edk2-porting/edk2-rk3588/actions/workflows/nightly.yml and see if that helps?

Victuar98, 2024-03-17:
Thanks for the quick response!))) If I understand correctly, then I should download the latest u-boot image for my board - orangepi-5_UEFI_Debug_779b6dc.img and write it to the SD card. Unfortunately, everything is the same - the settings are not saved...(((But unlike the previous version, the keyboard works right away!)))

mariobalanica, 2024-03-17:
Can you dump the full UART log? The one above is truncated..

But from what I can see, it's storing the user settings on eMMC rather than SD. Do you have some bootloader on the eMMC as well?

Victuar98, 2024-03-18:
Hi! Yes, I apologize, it turns out there are many more (O.O) lines!))) At the same time there is UEFI Shell, but I do not guarantee its completeness...
[UART_UEFI_Debug.log](https://github.com/edk2-porting/edk2-rk3588/files/14632139/UART_UEFI_Debug.log)
[UEFI_Shell.txt](https://github.com/edk2-porting/edk2-rk3588/files/14632141/UEFI_Shell.txt)

mariobalanica, 2024-03-18:
You have orangepi's U-Boot on eMMC, please erase that first. 

UEFI stores user settings on the boot device. U-Boot on eMMC is chain-loading UEFI on SD card, so it can't determine which one is the actual boot device.

mariobalanica, 2024-03-18:
Settings end up being loaded from SD card, but they're saved on eMMC, since that's the initial boot device.

Victuar98, 2024-03-19:
Hi!) 
1 - The meaning of my question is that I want to boot from a USB HDD/SSD/Flash drive without affecting eMMC in any way(!). For this, I assume to use the u-boot loader located on the SD card. Judging by your answers, such a scheme is impossible and, of course, you will have to place the u-boot on eMMC media, but this will be the next step.
2 - Now a Debian image from the official Orange Pi website is installed on eMMC via MaskROM, it may contain a u-boot, but I could not detect it, and it did not manifest itself in any way. There are two partitions on the eMMC - fat32 and ext4. On /dev/mmcblk0p1 with a size of 1 Gb, there is a bootfs with a size of 118 Mb. However, there is still an unmarked area of 30 Mb from 0 to 61349 sectors, and perhaps the u-boot hid there.)))
3 - Here I do not understand anything at all...((( Why can't we specify which device will be the primary one and save the settings where we need them?
However, it does not matter. I sincerely thank you for your answers and time spent on me!)))

ARC-MX, 2024-07-06:
I tried, but I found that the page told me that I don't have access to Jira on armbian.atlassian.net.

mariobalanica, 2024-12-19:
This has been fixed, but it's still recommended to have only one bootloader at a time to avoid any confusion.