edk2-porting/edk2-rk3588 #137: H96 MAX V58

manaboy, 2024-04-16:
Is the H96 max v58 which the original manufacturer has released final firmware with release keys going to be supported by this custom firmware?

Googulator, 2024-04-16:
While I'm not familiar with the specifics of the H96 Max, do note that this project isn't "firmware" in the same sense it's usually understood in relation to TV boxes, i.e. not a custom or modified operating system image. Instead, it's a UEFI-compatible SPI boot image (similar to what's called a bootloader in TV box parlance) that enables loading generic, general-purpose operating systems intended for ARM SystemReady and similar systems.

manaboy, 2024-04-16:
Would that include bootloading windows or Linux in that manner?

closed 2024-06-22 by mariobalanica