edk2-porting/edk2-rk3588 #141: Built-in ethernet not working on generic Linux image in ACPI mode on Orange Pi 5.

monomycelium, 2024-05-29:
When I booted the [Network Install image for Debian stable](https://cdimage.debian.org/debian-cd/current/arm64/iso-cd/debian-12.5.0-arm64-netinst.iso), no ethernet card was detected. However, the ethernet card worked fine when using booting [Armbian rk3588-live-iso](https://github.com/amazingfate/rk3588-live-iso) in Device Tree mode.

The README file states that ethernet is "tested/supported" in Linux in ACPI mode. I will try booting Ubuntu 22.04, with kernel 5.15.0-75-generic.

mariobalanica, 2024-05-29:
Right, it says PCIe Ethernet, e.g. Realtek on some boards.

monomycelium, 2024-05-30:
I see. Is this a work-in-progress or out of scope for the project?

mariobalanica, 2024-05-30:
No plans for it. Device Tree is the way to go on Linux, but you need to use a fairly recent kernel.

closed 2024-05-30 by mariobalanica