edk2-porting/edk2-rk3588 #142: Board request: NanoPc CM3588 NAS

NomisIV, 2024-06-06:
What would be required to add support for the [FriendlyElec CM3588 NAS kit](https://www.friendlyelec.com/index.php?route=product/product&path=60&product_id=294) board? I have very little experience in this area, but I am motivated to get this implemented. I do own this board myself, so I would be able to perform testing on it.

What I've figured out myself is that the device tree (dts) file needs to be included [here](https://github.com/edk2-porting/edk2-rk3588/tree/master/edk2-rockchip-non-osi/Platform/Rockchip/DeviceTree). The proper device tree for this board is probably [this one](https://github.com/armbian/linux-rockchip/blob/rk-6.1-rkr1/arch/arm64/boot/dts/rockchip/rk3588-nanopc-cm3588-nas.dts).

Are there any more files that need to be included for proper support?

Looking forward for input!

jjpe, 2024-06-13:
I too would like to know how support for the CM3588 would be added.

mariobalanica, 2024-06-22:
You can use one of the existing platforms as a template.
They're all fairly similar, with notable differences being power enable GPIOs (e.g. for USB) and I2C peripheral locations. These details can be found in the device tree and board schematics.

Board-specific code resides in the .DSC file and RockchipPlatformLib.

The DTB should indeed be imported from Armbian's rk 6.1 branch. Right now it's only used for booting Linux, UEFI itself does not care about it.

By the way, the PCIe code does not currently support bifurcation (which the CM3588 appears to be using), so things like NVME drives won't be usable in UEFI.

jjpe, 2024-06-25:
Thank you for the response.

> By the way, the PCIe code does not currently support bifurcation (which the CM3588 appears to be using), so things like NVME drives won't be usable in UEFI.

What exactly would be needed for PCIe bifurcation support? I ask because my primary use case for the board is as a NAS, with... you guessed it, NVMe drives.

mariobalanica, 2024-06-29:
Bifurcation missing in UEFI means you can't boot from NVME. Maybe one of the slots would work at x1 link depending on how things are wired, but I don't know for sure.

It shouldn't be difficult to implement. It mostly needs changes to the PCIe 3 PHY driver and some nice user-configurable settings for lane mapping. I haven't had the hardware to test this and hence no interest really, but I'm going to accept pull requests if anyone's interested in taking on this task.

If your use-case is booting Linux with a device tree off SD/eMMC/USB, then you probably don't care about any of the above.

navarro967, 2024-07-04:
I have begun simple work to get it ported to the device [here](https://github.com/navarro967/edk2-rk3588/tree/master-CM3588-nas).

The fork/branch produces a bootable build of EDK2. This device is almost identical to the T6 which I used as the base. I don't particularly require NVMe boot however the first NVMe device did show as a boot option.

I'm relatively new to the ARM scene and not very adept at reading device schematics however if I can reverse engineer the values from the T6 to understand how this is mapped I should be able to clean this up and submit a PR. @mariobalanica Maybe you can give me a simple example of how this mapping done?

Otherwise, if anyone want's to take a crack at it you can build from my fork

NomisIV, 2024-07-11:
I tried your fork @navarro967, but my keyboard wouldn't get detected. If I was lucky I could get into the boot options by pressing Escape, but once I got in I couldn't navigate anything.

Were you able to get this working on your board, or are you experiencing the same issue?

jjpe, 2024-07-12:
I just got my hardware.
What can I do to help, short of bricking the device?

Rippanda12, 2024-07-12:
> short of bricking the device?

@jjpe 
These devices aren't "brickable" since you have access to flashing modes etc which aren't really present on phones so don't worry about something happening to your device

jjpe, 2024-07-13:
> These devices aren't "brickable"

Not even if a wrong image is flashed, which at least in theory, could cause electrical damage by driving e.g. some I/O ports incorrectly? If that's true, then that is very useful to know indeed.

Rippanda12, 2024-07-13:
> > These devices aren't "brickable"
> 
> Not even if a wrong image is flashed, which at least in theory, could cause electrical damage by driving e.g. some I/O ports incorrectly? If that's true, then that is very useful to know indeed. 

Software wise they can't be bricked just like a regular pc, ofc physical damage can be done i.e plugging stuff where they don't belong but that rarely happens

navarro967, 2024-07-23:
> I tried your fork @navarro967, but my keyboard wouldn't get detected. If I was lucky I could get into the boot options by pressing Escape, but once I got in I couldn't navigate anything.
> 
> Were you able to get this working on your board, or are you experiencing the same issue?

I was able to, I plugged my Keychron K10 into a USB A port and was able to enter the BIOS and make changes.

Jeidnx, 2024-07-23:
I have tried 5 different Keyboards, and the only one that worked was a Microsoft Sculpt (notably, the only one which uses a wireless dongle). Every one of them worked, once booted into the OS. I couldn't find any error messages, the only thing i noticed was that all keyboards get power when plugged in, but only for a second before turning off again.

jneem, 2024-08-27:
Thanks for kicking this off @navarro967! I rebased your fork on the latest master to bring in the PCIe bifurcation support, and I can confirm that the firmware loads and sees my NVMe drive.

I haven't actually managed to boot into an OS yet. I have a working Ubuntu installation on the eMMC -- I know it's working because if I remove the sd card with edk2 then it boots -- but selecting it in the boot configuration menu just flashes the screen black and brings me back to edk2. If anyone has suggestions about how to debug it, I'd be grateful. Otherwise, I'm going to poke around with GPIO settings.

serpent213, 2024-09-04:
Got my NAS board today and tried @navarro967's fork, flashed to eMMC via the FriendlyWrt web flasher. Console is serial, connected to the three pin header.

Looks pretty good, menu works fine, and it boots into Grub from USB or microSD. I used a NixOS ISO (`nixos-minimal-24.05.4469.6e99f2a27d60-aarch64-linux.iso`).

After selecting the serial console in Grub and progressing to the kernel it gets stuck, though:

```
EFI stub: Booting Linux Kernel...
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```

Also tried an ISO from another distro with very similar results.

Did not get any output on HDMI, but I didn't try very hard.

Any idea on how to move on?

serpent213, 2024-09-05:
I built the NixOS image from @ryan4yin: https://github.com/ryan4yin/nixos-rk3588/tree/main, the rock5a variant, needed a small patch:

```diff
diff --git a/flake.nix b/flake.nix
index 5102bbc..ecffe1a 100644
--- a/flake.nix
+++ b/flake.nix
@@ -31,7 +31,7 @@
     ...
   }: let
     # Local system's architecture, the host you are running this flake on.
-    localSystem = "x86_64-linux";
+    localSystem = "aarch64-linux";
     pkgsLocal = import nixpkgs {system = localSystem;};
     # The native system of the target SBC.
     aarch64System = "aarch64-linux";
diff --git a/pkgs/mali-firmware/default.nix b/pkgs/mali-firmware/default.nix
index f2bb1c0..bc48e64 100644
--- a/pkgs/mali-firmware/default.nix
+++ b/pkgs/mali-firmware/default.nix
@@ -7,7 +7,7 @@ stdenv.mkDerivation {
   version = "g21p0-01eac0";
 
   src = fetchurl {
-    url = "https://github.com/JeffyCN/mirrors/raw/e08ced3e0235b25a7ba2a3aeefd0e2fcbd434b68/firmware/g610/mali_csffw.bin";
+    url = "https://github.com/JeffyCN/mirrors/raw/libmali/firmware/g610/mali_csffw.bin";
     hash = "sha256-jnyCGlXKHDRcx59hJDYW3SX8NbgfCQlG8wKIbWdxLfU=";
   };
 
```

Copied to SD; throws errors due to hardware mismatch but boots fully with network working (on serial console, did not try HDMI).

NomisIV, 2024-09-17:
> I built the NixOS image from @ryan4yin: https://github.com/ryan4yin/nixos-rk3588/tree/main, the rock5a variant, needed a small patch:
>
> ```diff
> diff --git a/flake.nix b/flake.nix
> index [...]

Could you elaborate on the steps you took to get this working? I am unable to reproduce. The rawEfiImage fails to build, and the sdImage seems to have a broken partition table (I cannot mount the second partition, and the first one is empty)

jneem, 2024-09-18:
My local system isn't an aarch64-linux, so I only changed the mali firmware url. (But it doesn't look like it matters; `localSystem` is only used in the cross-compiling targets.)  I'm on an x86-64 system with `boot.binfmt.emulatedSystems = [ "aarch64-linux" ]` in my nixos configuration (otherwise it will probably fail with an error about needing to build on an aarch64 system).

Then I run `nix build .#rawEfiImage-rock5a`, dd the result to an sdcard, and plugging it into my cm3588 with edk2 already installed just works.

(I've been struggling with updating to a more recent kernel, but that's another story...)

zatevakhin, 2024-09-18:
@jneem What kernel versions worked for you?

jneem, 2024-09-19:
- the 6.1.43 kernel from orangepi (the one that's currently used in nixos-rk3588) boots and has ethernet
- I tried updating to the armbian/linux-rockchip kernel version 6.1.75. It boots but doesn't have ethernet (the r8169 module is available, it just doesn't get loaded)
- I tried updating to mainline 6.10.9 with config from armbian, but it doesn't boot at all. I get nonsense like this on the serial console:

```
EFI stub: Booting Linux Kernel...
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
I nCiagI dPwi alt
                 I dPilg/SaUwnnld
/eaUinnld
I dC ag/SaPsi aro
                 I nCianISdPsioaro
                                  I dCiagISdPsioaro
                                                   I dCianISdPsioaro
                                                                    [    0.702728] rtc-hym8563 6-0051: hctosys: unable to read the hardware clock
[    0.705401] rockchip-thermal fec00000.tsadc: failed to register sensor 0: -19
[    0.706080] rockchip-thermal fec00000.tsadc: error -ENODEV: failed to register sensor[0].
ISaUwn ldt
          I nCilgI dPwi alt
/SaUwnnldt                 I dPil
          I dPilgISaUwn ldt
                           I dCiag/SaPwi alt
                                            I dCiag/SdPwiomro
                                                             I nC anI dCshtmob
                                                                              I dPilgISaPwi llt
                                                                                               I dCilgISdPwi alt
                                                                                                                I dCilgISaPwi alt
                                                                                                                                 I dCilgISaPwi alo
                                                                                                                                                  I dCianI nC htmoo
                                                                                                                                                                   I nC inI n  c ow
I nC anISdPshtmo
```

I tried booting the second two options (armbian/linux-rockchip and mainline 6.10) using armbian (instead of nixos) to build an sdcard image. The bootable armbian sdcards skip edk2 and they boot with working ethernet.

jneem, 2024-09-19:
Btw, I have an updated branch [here](https://github.com/jneem/edk2-rk3588/tree/cm3588). Compared to @navarro967 's branch, it's based on a more recent master and it comes with some updates to the pin definitions (but I didn't actually observe any differences caused by these changes)

Jeidnx, 2024-09-19:
Thanks for your work @jneem! Did you use mainline with just the armbian config or did you also apply their kernel patches? I have tried to build the 6.10.{5,9} Kernels with the armbian patches and a modified version of their config (with compressed firmware loading enabled), but haven't been able to successfully build it yet. I am pretty sure it is an issue with my build process though. I have also tried the armbian 6.1.75 and the friendlyarm 6.1.57 kernels, both boot but no Ethernet. The chip doesn't show in lspci, i haven't done more digging yet. Curiously the friendlyarm kernel with almost the same config (i had to enable efi and dmiid)  and the same dts has working ethernet when used in the vendor debian 12 image.

I have made a flake to test nixos on the cm3588 a while ago, but just now gotten around to cleaning it up. [Here](https://github.com/Jeidnx/nixos-cm3588) is a link for anyone that's interested.

jneem, 2024-09-19:
I've been trying without the armbian patches so far, but maybe I should give them a try. Btw, 6.11 now has [mainline](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/arch/arm64/boot/dts/rockchip/rk3588-friendlyelec-cm3588-nas.dts?h=v6.11) support for the cm3588! I'm building that now.

Working ethernet is a priority for me, but I was hoping to get a new kernel booting before digging into the ethernet issue...

jneem, 2024-09-19:
> but haven't been able to successfully build it yet

If it's the same issue I had, it's because you need to set `CONFIG_CRYPTO_AEGIS128_SIMD=n`, because the assembler nix is using to build the kernel thinks that the rk3588 doesn't support some instructions. I don't know how armbian gets away with this setting; maybe their assembler thinks differently...

NomisIV, 2024-09-19:
I got 6.11 working (with ethernet) on my cm3588. Here is the minimal config needed:
```nix
  boot.kernelPackages = pkgs.linuxPackagesFor pkgs.linux_latest;
  hardware = {
    enableRedistributableFirmware = true;
    deviceTree = {
      enable = true;
      name = "rockchip/rk3588-friendlyelec-cm3588-nas.dtb";
    };
  };
```

DualTachyon, 2024-09-19:
> If it's the same issue I had, it's because you need to set `CONFIG_CRYPTO_AEGIS128_SIMD=n`, because the assembler nix is using to build the kernel thinks that the rk3588 doesn't support some [...]

The Crypto instructions on RK3588 are disabled by default. Whether they are enabled in your environment depends on what build of BL31 is being used. See this patch for reference: https://review.trustedfirmware.org/c/TF-A/trusted-firmware-a/+/29363

jneem, 2024-09-20:
> boot.kernelPackages = pkgs.linuxPackagesFor pkgs.linux_latest;

Amazing! All that fiddling with armbian forks and custom configs, and it turns out that latest mainline just works! (For anyone following along, it's important here to be using `nixos-unstable` or else `linux_latest` is too old.) Now I have edk2 on my eMMC, booting into nixos on an nvme drive.

samyak-jain, 2024-09-21:
I am trying to flash the firmware from this branch (https://github.com/jneem/edk2-rk3588/tree/cm3588), however, I am unable to enter the boot options when pressing escape at the boot splash screen. I've tried 2 different keyboards. It boots into openmediavault by default no matter what I do. Any suggestions?

jneem, 2024-09-22:
It looks like only one USB port was being powered up (the upper USB 3.0 port). The latest commit should enable them all.

jjpe, 2024-09-25:
For the NixOS users here, how did you cross compile jneem's [EDK2 branch](https://github.com/jneem/edk2-rk3588/tree/cm3588)?
I'd rather not have to muck about with Debian and the like.

sanderhollaar, 2024-09-25:
@jneem Thanks a lot for your work! 🤗

> For the NixOS users here, how did you cross compile jneem's [EDK2 branch](https://github.com/jneem/edk2-rk3588/tree/cm3588)? I'd rather not have to muck about with Debian and the like.

Until someone actually answers your question ..

I did not want to trial and error the required tools and went the Debian route (on `x68_64`):

Enable `podman` with `virtualisation.podman.enable = true;`

Create `/tmp/mnt/`: `mkdir /tmp/mnt/`
Start Debian: `podman run --rm -it --network=host --volume=/tmp/mnt:/root debian:trixie /bin/bash`

In Debian:
`apt update && apt upgrade -y`
`apt install -y git gcc g++ build-essential gcc-aarch64-linux-gnu acpica-tools python3-pyelftools uuid-dev python-is-python3 device-tree-compiler`
`cd ~`
`git clone --recursive https://github.com/jneem/edk2-rk3588.git`
`cd edk2-rk3588`
`git switch cm3588`
`./build.sh --device nanopc-cm3588-nas --release Release`

Find `RK3588_NOR_FLASH.img` in `/tmp/mnt/`

Have not tried the UEFI image yet.

I also have a `NanoPi R6C` running NixOS.

jneem, 2024-09-25:
There were a few moving parts, and I haven't packaged it all up nicely.

1) You need to rebuild the `mkimage` binary because it isn't nix-friendly. I have a flake [here](https://github.com/jneem/edk2-nix) for building it, and then you can copy it to `misc/tools/<arch>/mkimage`
2) I needed to apply this patch (attached) to the `edk2/` submodule
3) I used the attached `shell.nix`
4) I had some trouble with OpenTFA that I didn't feel like debugging, so I build without it: `bash build.sh --device nanopc-cm3588-nas --open-tfa 0 --release RELEASE`

[edk2-patch.txt](https://github.com/user-attachments/files/17133746/edk2-patch.txt)
[shell.nix.txt](https://github.com/user-attachments/files/17133783/shell.nix.txt)

jjpe, 2024-09-25:
> I had some trouble with OpenTFA that I didn't feel like debugging, so I build without it: bash build.sh --device nanopc-cm3588-nas --open-tfa 0 --release RELEASE

One thing I'm running into ATM is that when I try this step, it fails to find brotli headers:
```
[nix-shell:~/dev/edk2-rk3588]$ bash build.sh --device nanopc-cm3588-nas --open-tfa 0 --release DEBUG
make: Entering directory '/home/j/dev/edk2-rk3588/edk2/BaseTools'
make -C Source/C
make[1]: Entering directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C'
Attempting to detect HOST_ARCH from 'uname -m': x86_64
Detected HOST_ARCH of X64 using uname.
mkdir -p .
make -C Common
make[2]: Entering directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C/Common'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C/Common'
make -C BrotliCompress
make[2]: Entering directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C/BrotliCompress'
gcc  -c -I ./brotli/c/include -I .. -I ../Include/Common -I ../Include/ -I ../Include/IndustryStandard -I ../Common/ -I .. -I . -I ../../../../MdePkg/Include/X64/ -I ../../../../MdePkg/Include -MD -fshort-wchar -fno-strict-aliasing -fwrapv -fno-delete-null-pointer-checks -Wall -Werror -Wno-deprecated-declarations -Wno-stringop-truncation -Wno-restrict -Wno-unused-result -nostdlib -g -O2  BrotliCompress.c -o BrotliCompress.o
BrotliCompress.c:20:10: fatal error: ./brotli/c/common/constants.h: No such file or directory
   20 | #include "./brotli/c/common/constants.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [../Makefiles/footer.makefile:21: BrotliCompress.o] Error 1
make[2]: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C/BrotliCompress'
make[1]: *** [GNUmakefile:79: BrotliCompress] Error 2
make[1]: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Source/C'
make: *** [GNUmakefile:19: Source/C] Error 2
make: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools'

```
I tried adding `brotli` to the `nativeBuildInputs` in `shell.nix` but that made no difference. 
Other than that, I don't directly see a place to add brotli support 

So I ended up just cloning the [brotli repo](https://github.com/google/brotli) to 
`edk-rk3588/edk2/BaseTools/Source/C/BrotliCompress/brotli/` and `edk2-rk3588/edk2/MdeModulePkg/Library/BrotliCustomDecompressLib/brotli/`, as that was the path of least resistance (though it technically breaks Nix's compilation model). I'm not sure how the upstream `edk-rk3588` expects developers to deal with that .

That seems to work, however the build then fails because it can't find openssl:
```
# SNIP

Ran 303 tests in 0.724s

OK
make[1]: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools/Tests'
make: Leaving directory '/home/j/dev/edk2-rk3588/edk2/BaseTools'
Loading previous configuration from /home/j/dev/edk2-rk3588/workspace/Conf/BuildEnv.sh
Using EDK2 in-source Basetools
WORKSPACE: /home/j/dev/edk2-rk3588/workspace
EDK_TOOLS_PATH: /home/j/dev/edk2-rk3588/edk2/BaseTools
CONF_PATH: /home/j/dev/edk2-rk3588/workspace/Conf
Build environment: Linux-6.6.52-x86_64-with-glibc2.39
Build start time: 18:44:42, Sep.25 2024

WORKSPACE        = /home/j/dev/edk2-rk3588/workspace
PACKAGES_PATH    = /home/j/dev/edk2-rk3588/edk2:/home/j/dev/edk2-rk3588/edk2-platforms:/home/j/dev/edk2-rk3588/edk2-rockchip:/home/j/dev/edk2-rk3588/edk2-rockchip-non-osi:/home/j/dev/edk2-rk3588/edk2-non-osi:/home/j/dev/edk2-rk3588
EDK_TOOLS_PATH   = /home/j/dev/edk2-rk3588/edk2/BaseTools
CONF_PATH        = /home/j/dev/edk2-rk3588/workspace/Conf
PYTHON_COMMAND   = python3

Processing meta-data Architecture(s)  = AARCH64
.Build target     = RELEASE
Toolchain        = GCC

Active Platform          = /home/j/dev/edk2-rk3588/edk2-rockchip/Platform/FriendlyElec/NanoPC-CM3588-NAS/NanoPC-CM3588-NAS.dsc

build.py...
/home/j/dev/edk2-rk3588/edk2/MdePkg/MdePkg.dec(32): error 000E: File/directory not found in workspace
        Library/MipiSysTLib/mipisyst/library/include is not found in packages path:
        /home/j/dev/edk2-rk3588/edk2
        /home/j/dev/edk2-rk3588/edk2-platforms
        /home/j/dev/edk2-rk3588/edk2-rockchip
        /home/j/dev/edk2-rk3588/edk2-rockchip-non-osi
        /home/j/dev/edk2-rk3588/edk2-non-osi
        /home/j/dev/edk2-rk3588

- Failed -
Build end time: 18:44:43, Sep.25 2024
Build total time: 00:00:01

```

Have you perhaps figured out any way to make the build use the dependencies from nixpkgs where possible?

jjpe, 2024-09-25:
I follow your steps, cloning the branch using `git clone --recursive https://github.com/jneem/edk2-rk3588.git`, and then executing `cd edk2-rk3588;  git switch cm3588`, after which I build, but I get an error: 

```
[nix-shell:~/dev/edk2-rk3588]$ bash build.sh --device nanopc-cm3588-nas --open-tfa 0 --release RELEASE

# SNIP

build.py...
 : error 7000: Failed to execute command
        make -s tbuild [/home/j/dev/edk2-rk3588/workspace/Build/NanoPC-CM3588-NAS/RELEASE_GCC/AARCH64/CryptoPkg/Library/OpensslLib/OpensslLib]

build.py...
 : error F002: Failed to build module
        /home/j/dev/edk2-rk3588/edk2/CryptoPkg/Library/OpensslLib/OpensslLib.inf [AARCH64, GCC, RELEASE]

- Failed -
Build end time: 19:35:20, Sep.25 2024
Build total time: 00:00:03

```

The `OpensslLib.inf` file does exist, it somehow just doesn't build. 
Any ideas?

jjpe, 2024-09-25:
Well, ultimately I got the image to build.
I wrote it to an sd card using `sudo dd bs=4M if=~/Downloads/RK3588_NOR_FLASH.img of=/dev/sda`.
I put it into my CM3588, plugged in a monitor and keyboard, and... 

It boots into openmediavault, and I can't see any obvious EFI configuration options. 
I also can't login from the CLI prompt, but I can navigate to it using my browser (url: `http://openmediavault`), which opens a WebGUI.

Then, when I remove the sd card from the CM3588, and reboot it, it still boots to openmediavault.

But I was expecting an EFI options menu. So is this the above supposed to happen?

Jeidnx, 2024-09-26:
@jjpe OpenZFS doesn't support the 6.11 kernel yet, that's why the package is marked as broken. I am still trying to get the 6.10 Kernel to work, but no luck so far.

jneem, 2024-09-26:
@samyak-jain did that patch end up getting your USB to work? If so, I might go ahead and submit this as a PR, because I think the OpenZFS issue is separate.

picnoir, 2024-09-26:
@jneem I've been using your branch without any major issue. I think it's at least ready to get PR-d.

I have one remaining issue wrt. thermals. `lm_sensors` does not report back any temperature data for the SOC on a 6.11 live NixOS installer. All the required sensor device trees should be there, they've been merged part of 6.10. Not sure if it's related to the edk2 boot though.

samyak-jain, 2024-09-27:
Hey @jneem. I don't have an answer on what the root cause of my problem is so I can't say for sure. Unfortunately, the patch did not fix my issue. I've tried each usb port with all my keyboards after fetching the latest version of the branch. I am taken to openmediavault every single time and my keyboard is working then so it's unclear to me if this is a usb issue or if the problem lies elsewhere. I've also noticed that some symbols like "^]" will appear on the top when I spam escape at the start so it looks like something might be registering even if I'm not taken to boot options.

jjpe, 2024-09-28:
> @jjpe OpenZFS doesn't support the 6.11 kernel yet, that's why the package is marked as broken. I am still trying to get the 6.10 Kernel to work, but no luck so far.

@Jeidnx  Yeah I figured as much. It makes me feel a bit better about not prioritizing my NAS build a bit more highly though, since I would have hit this (temporary) roadblock anyway.

@jneem I should mention that while I got EDK2 installed, and could boot into the EFI settings menu, the AArch64 `nixos-unstable` image I `dd`'d onto an SD card got as far as displaying the image's boot menu, but none of the available options booted successfully.

Instead I got a black screen with this text on it:
```
EFI stub: Booting Linux Kernel...
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Generating empty DTB
EFI stub: Exiting boot services...
```

Now granted, that was just a very recent image I downloaded from the servers, and not any of the forks people here have been working on. 
I'll try one of those next.

picnoir, 2024-09-28:
@jjpe I used the `iso_minimal_new_kernel_no_zfs.aarch64-linux` image (to get linux 6.11) flashed on a usb drive.

You can download the latest one there: https://hydra.nixos.org/build/273957457

sanderhollaar, 2024-09-29:
@jneem I'm booting eMMC installed NixOS on my CM3588 NAS with your UEFI image.

Thanks again for your work!

NomisIV, 2024-09-30:
@jjpe @Jeidnx There is unofficial support for linux 6.11 in zfs_unstable. I have successfully installed NixOS on ZFS on 4 SSDs. The package does complain that it is broken when trying to build, but as far as I can tell it is working anyway. Just give nix the `NIXPKGS_ALLOW_BROKEN=1` flag, and it should work.

I'm still having problems with HDMI on 6.11 though (no output after kernel starts). Anyone got any ideas for why?

sanderhollaar, 2024-09-30:
After UEFI I have no HDMI output either. I did not look into it as I run the server headless.

jjpe, 2024-10-03:
I'm having some trouble with the nixos live image mentioned by @picnoir.
I can boot it, but the image doesn't seem to detect:
- The eMMC storage
- The 4 NVMe drives I have hooked up to the board.

By default, the EFI NVMe settings look like this:
- `PCI express 3.0`:
  - `Support State`: **enabled**
  - `PHY mode`: **x4**
  
If I change `PHY mode` to **x1x1 x1x1**, and boot the usb disk, then I only see 2 out of 4 NVMe SSDs listed:
[image]

I think I'm missing something, but what?

jjpe, 2024-10-04:
> There is unofficial support for linux 6.11 in zfs_unstable ([link](https://github.com/openzfs/zfs/pull/16400)). I have successfully installed NixOS on ZFS on 4 SSDs.

@NomisIV Could you tell me what EFI settings you used to make nixos recognize all 4 drives?

jjpe, 2024-10-06:
Thank you, that will be most helpful.

One tricky aspect:
In the EFI setup menu, the emmc device is detected (so it's definitely present).
But when I boot in ACPI mode, the emmc drive isn't detected.
However, In Device Tree mode, I have no HDMI output (as mentioned in earlier posts). In addition, my keyboard is turned off, which means I can't (blindly) type `passwd`, enter a password, confirm it, in order to then ssh into the device.

So one (hopefully) last question:  How did you manage to install the OS to the emmc drive? 
If I can manage to get there, the rest will be pretty easy thanks to your help.

sanderhollaar, 2024-10-06:
> However, In Device Tree mode, I have no HDMI output (as mentioned in earlier posts). In addition, my keyboard is turned off, which means I can't (blindly) type `passwd`, enter a password, confirm it, in order to then ssh into the device.

With EFI set to `DeviceTree` my usb keyboard works in one of the two blue usb 3.0 ports. I used it to blindly configure the network to be able to ssh in. In my initial configuration during install, I had the wrong name of the interface. It is `enP4p65s0` on my device.

> So one (hopefully) last question: How did you manage to install the OS to the emmc drive?

From https://download.friendlyelec.com/CM3588 found on https://wiki.friendlyelec.com/wiki/index.php/CM3588 I downloaded `rk3588-sd-debian-bookworm-core-6.1-arm64-20240818.img` and `dd`'ed it to a microsd card.

In Debian I installed `nix` to be able to start a shell with tools. `dd`'ed the UEFI image to emmc, partitioned the emmc (keeping the first partition from the UEFI image untouched), and installed NixOS as usual (`nixos-generate-config` /`nixos-install`):
```
curl -L https://nixos.org/nix/install | sh -s -- --daemon
echo 'extra-experimental-features = flakes nix-command' | tee -a /etc/nix/nix.conf

nix shell -- nixpkgs#{coreutils-full,dosfstools,f2fs-tools,fscrypt-experimental,gptfdisk,nixos-install-tools,util-linux,vim}
```
```
$ sudo gdisk -l /dev/mmcblk0
[..]
Found valid GPT with protective MBR; using GPT.
Disk /dev/mmcblk0: 488554496 sectors, 233.0 GiB
[..]
Number  Start (sector)    End (sector)  Size       Code  Name
   1            2048           18431   8.0 MiB     8300  uboot
   2           18432         1067007   512.0 MiB   EF00  hamster_uefi
   3         1067008       488554462   232.5 GiB   8300  hamster
```
`uboot` is the UEFI image, `hamster_uefi` is `/boot`

jjpe, 2024-10-12:
This definitely helped, as it allowed me to partition and format the nvme and internal emmc drives, and then install NixOS on the emmc drive.
Even the systemd bootloader seems to work, as the usual NixOS generation boot menu now pops up.

The only thing I'm wondering is: how did you blindly configure the network?
And for that matter, how did you figure out that the network interface should be `enP4p65s0`?

I'm asking because after selecting a NixOS generation to boot, the screen (expectedly) goes blank, the keyboard lights stay lit, but when I try to log in with the root account (for which a password was entered during install), nothing seems to happen.

sanderhollaar, 2024-10-12:
> The only thing I'm wondering is: how did you blindly configure the network? And for that matter, how did you figure out that the network interface should be `enP4p65s0`?
> 
> I'm asking because after selecting a NixOS generation to boot, the screen (expectedly) goes blank, the keyboard lights stay lit, but when I try to log in with the root account (for which a password was entered during install), nothing seems to happen. And I can't tell the difference except for running `shutdown now` at the end, which always fails. So from that I infer that I was never logged in at all.

First thing I did was also power off the system to verify successful login. I use `systemctl poweroff` and see the leds on the CM switch off. `shutdown now` should work too, so it seems you are indeed not reaching login.

During install, I use a stripped down `configuration.nix` from another system with accounts, ssh and keys. And I use filesystem labels in `hardware-configuration.nix`, with `ext4` filesystem for `/boot` (and `f2fs` for `/`). Would this help your case, or is this already similar to your setup?

For the interface: I constructed a oneliner to fetch the name of the interface, and assign an ip to it:

`ip a a 192.168.178.3/24 dev $(ip a s|grep -m1 MULTICAST | awk -F ':' '{print $2}')`

I carefully typed the command and verified success with ping running on my laptop.

I then could ssh in with the non-root account already configured during install, `ip route add` a default route to the internet, correct the interface name in `configuration.nix` and `nixos-rebuild`.

Did you try `enP4p65s0` in your `configuration.nix`?

I later learned of the kernel parameter `earlycon=efifb`. Can you try if that continues output after the systemd bootloader?

jjpe, 2024-10-13:
> Would this help your case, or is this already similar to your setup?

I'll post my configuration for clarity, as viewed from the booted Debian SD card image.

I use 2 partitions on the emmc drive. 
The mount command (actually a one-liner, broken up for easier reading): 
```bash
mkdir -p /mnt-nixos/boot  \
  &&  mount /dev/disk/by-label/NIXROOT /mnt-nixos  \
  &&  mount /dev/disk/by-label/NIXBOOT /mnt-nixos/boot/ -o umask=077,fmask=0077,dmask=0077  \
  &&  systemctl daemon-reload
```

`mount` cmd output:
```
root@CM3588:~# mount
...  # snip irrelevant Debian-related mount details
/dev/mmcblk2p3 on /mnt-nixos type ext4 (rw,relatime)
/dev/mmcblk2p2 on /mnt-nixos/boot type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=936,iocharset=utf8,shortname=mixed,errors=remount-ro)
```
The reason I made these formatting choices is that the NixOS manual uses them. 
So I figured that these should be uncontroversial and reliable options for now (Of course, I could be wrong about that ...). 
Once there's a working baseline, making changes is easy.

`configuration.nix`:
```nix
{ config, lib, pkgs, ... }:
{
  imports =
    [ # Include the results of the hardware scan.
      ./hardware-configuration.nix
    ];

  boot.kernelPackages = pkgs.linuxPackages_latest;
  boot.loader = {
    efi = {
      canTouchEfiVariables = true;
    };
    systemd-boot = {
      enable = true;
      configurationLimit = 10;
    };
  };

# TODO: configure ZFS
#  boot.supportedFilesystems.zfs = true;
#  boot.initrd.supportedFilesystems.zfs = true;
#  boot.zfs.package = pkgs.zfs_unstable;
#  services.zfs = {
#    autoScrub.enable = true;
#    autoSnapshot.enable = true;
#  };
  networking.hostId = "54e492c4"; # See https://nixos.org/manual/nixos/unstable/options#opt-networking.hostId
  nixpkgs.config.allowBroken = true; # allow broken ZFS for now

  hardware.enableRedistributableFirmware = true;
  hardware.deviceTree = {
    enable = true;
    name = "rockchip/rk3588-friendlyelec-cm3588-nas.dtb";
  };

  networking.hostName = "cumulus"; # Define your hostname.

  # Set your time zone.
  time.timeZone = "Europe/Amsterdam";

  # Configure network proxy if necessary
  # networking.proxy.default = "http://user:password@proxy:port/";
  # networking.proxy.noProxy = "127.0.0.1,localhost,internal.domain";

[... 154 lines cut ...]
}
```
I also tried disabling home manager for now, but that turns out not to make a difference.

> For the interface: I constructed a oneliner to fetch the name of the interface, and assign an ip to it:
> 
> ip a a 192.168.178.3/24 dev $(ip a s | grep -m1 MULTICAST | awk -F ':' '{print $2}')
> 
> I carefully typed the command and verified success with ping running on my laptop.

Once I can log in, this will be most useful, thanks.

> I then could ssh in with the non-root account already configured during install

This is where things currently break for me. I tried logging in with both `root` and the non-root account I configured (called `j`), and can successfully log in with neither. I don't see anything obvious wrong in my config that would prevent login, but perhaps you see something wrong in the config files above?

> Did you try enP4p65s0 in your configuration.nix?
I tried both `enP4p65s0` and `eth0` (see `configuration.nix` above), and neither seems to work.

> I later learned of the kernel parameter earlycon=efifb. Can you try if that continues output after the systemd bootloader?

Just tried that. The outcome is that it prints a few lines (apologies for the quality):
[image]

After that the screen goes blank as usual.

sanderhollaar, 2024-10-13:
> ```shell
> mkdir -p /mnt-nixos/boot  \
>   &&  mount /dev/disk/by-label/NIXROOT /mnt-nixos  \
>   &&  mount /dev/disk/by-label/NIXBOOT /mnt-nixos/boot/ -o umask=077,fmask=0077,dmask=0077  \
>   &&  systemctl daemon-reload
> ```
This will not work if you `mkfs`'ed the filesystems just before these commands. The `/mnt-nixos/boot` directory created with `mkdir` will be hidden by the `mount` at `/mnt-nixos`. So I assume you are mounting a previous install here, with a directory structure that includes `/boot`.

Can you show the contents of your `/boot` directory with `find /mnt-nixos/boot`? Could it be that it is missing `rockchip/rk3588-friendlyelec-cm3588-nas.dtb` ? Maybe because during the initial install, the `mount` of `/boot` failed due to missing `/boot` directory?

> `mount` cmd output:
> ```
> root@CM3588:~# mount
> ...  # snip irrelevant Debian-related mount details
> /dev/mmcblk2p3 on /mnt-nixos type ext4 (rw,relatime)
> /dev/mmcblk2p2 on /mnt-nixos/boot type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=936,iocharset=utf8,shortname=mixed,errors=remount-ro)
> ```

I'm surprised by the `2` in `/dev/mmcblk2` - mine is `/dev/mmcblk0` - but with labels or uuid that should not matter.

>   networking.hostName = "cumulus"; # Define your hostname.

Nice hostname :sunglasses: I named my company Humilis, after Cumulus Humilis.

>   networking = { # I also tried the eth0 interface, as that's how it's enumerated in Debian. Alas, to no avail.

I'm using `systemd.network`, but that does not matter.

>   networking.firewall.allowedTCPPorts = [ 22 ];  # shouldn't be necessary but can't hurt during setup.

> `hardware-configuration.nix`:
>   networking.useDHCP = lib.mkDefault true;

I do however, disable both firewall and dhcp:
```
    networking = {
        dhcpcd.enable    =  false;
        firewall.enable  =  false;
        hostName         = "hamster";
        useDHCP          =  false;
        useNetworkd      =  true;
    };
```
Likely does not matter.

> As for the drives, I mount them by uuid (which I also do on my other nixos hosts): I have 4 identical NVMe drives so it seems like there's less chance of confusion when mounting by label. I expect mounting by uuid to be at least as stable as mounting by label.

Agreed. Although the NVMe drives are not in play yet.

> I installed nixos using `nixos-install --root /mnt-nixos --flake .#cumulus`, so here is my `flake.nix` as well:

> I also tried disabling home manager for now, but that turns out not to make a difference.

Yeah home-manager has no part in the OS itself. I have my home-manager entirely in my home directory btw, but unrelated.

> This is where things currently break for me. I tried logging in with both `root` and the non-root account I configured (called `j`), and can successfully log in with neither. I don't see anything [...]

Nothing obvious. It is a long shot but could be missing files in `/boot`, or long boot time if somehow it tries to DHCP.

> > I later learned of the kernel parameter earlycon=efifb. Can you try if that continues output after the systemd bootloader?
> 
> Just tried that. The outcome is that it prints a few lines (apologies for the quality):
> After that the screen goes blank as usual.

I guess this requires upstream HDMI support then. Next step could be, find the patch that enables HDMI in Armbian and include that in your NixOS kernel stanza.

If you can't make it work I'll recreate the stripped-down `configuration.nix`.

jjpe, 2024-10-13:
> This will not work if you mkfs'ed the filesystems just before these commands. The /mnt-nixos/boot directory created with mkdir will be hidden by the mount at /mnt-nixos. So I assume you are mounting a previous install here, with a directory structure that includes /boot.

Indeed, what I did to gather this information was:
1. Boot Debian with the same SD card I used initially
2. Enter a nix shell with `nix shell -- nixpkgs#{coreutils-full,dosfstools,f2fs-tools,fscrypt-experimental,gptfdisk,nixos-install-tools,util-linux,neovim,emacs,nix,nixos-rebuild,nix-tree}`
3. Execute `mkdir -p /mnt-nixos/boot  &&  mount /dev/disk/by-label/NIXROOT /mnt-nixos  &&  mount /dev/disk/by-label/NIXBOOT /mnt-nixos/boot/ -o umask=077,fmask=0077,dmask=0077  &&  systemctl daemon-reload`
4. Execute `nixos-install --root /mnt-nixos --flake .#cumulus` from the git repo containing the nix configuration files.

In particular, the partitions I'm mounting were created just before I created the first NixOS generation with the first execution step 4.
As presented above, the idea is to take the extant partitions as they are, and just bluntly install NixOS over the previous installation. This results in the creation of a new generation, just as if the system had been updated using `nixos-rebuild`. 

In order to create and format the partitions with the proper flags I followed the steps outlined in the [NixOS manual](https://nixos.org/manual/nixos/stable/#sec-installing-from-other-distro). Except I used a flake, which *should* work just as well as direct usage of `configuration.nix`.

> Can you show the contents of your /boot directory with `find /mnt-nixos/boot`? 
Sure, here's the output:
```
root@CM3588:~# find /mnt-nixos/boot/
/mnt-nixos/boot/
/mnt-nixos/boot/EFI
/mnt-nixos/boot/EFI/systemd
/mnt-nixos/boot/EFI/systemd/systemd-bootaa64.efi
/mnt-nixos/boot/EFI/BOOT
/mnt-nixos/boot/EFI/BOOT/BOOTAA64.EFI
/mnt-nixos/boot/EFI/Linux
/mnt-nixos/boot/EFI/nixos
/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-Image.efi
/mnt-nixos/boot/EFI/nixos/d19pz28xk6h2v3mfv2w2g1rwqj3dfqxr-initrd-linux-6.11.2-initrd.efi
/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi
/mnt-nixos/boot/EFI/nixos/.extra-files
/mnt-nixos/boot/EFI/nixos/f0f8n5zwk3kixg722ahdldp3jd40yp4z-initrd-linux-6.11.2-initrd.efi
/mnt-nixos/boot/EFI/nixos/n0lc1vg2fdvypnl8icas8y72hxgmfs65-initrd-linux-6.11.2-initrd.efi
/mnt-nixos/boot/EFI/nixos/anzs9381sh15m21f6ba4c3y0d6hwgr59-initrd-linux-6.11.2-initrd.efi
/mnt-nixos/boot/EFI/nixos/c7x9rrris4w298s43hgwd1sbriq1vbc2-initrd-linux-6.11.2-initrd.efi
/mnt-nixos/boot/loader
/mnt-nixos/boot/loader/entries
/mnt-nixos/boot/loader/entries/nixos-generation-11.conf
/mnt-nixos/boot/loader/entries/nixos-generation-12.conf
/mnt-nixos/boot/loader/entries/nixos-generation-3.conf
/mnt-nixos/boot/loader/entries/nixos-generation-4.conf
/mnt-nixos/boot/loader/entries/nixos-generation-5.conf
/mnt-nixos/boot/loader/entries/nixos-generation-6.conf
/mnt-nixos/boot/loader/entries/nixos-generation-7.conf
/mnt-nixos/boot/loader/entries/nixos-generation-8.conf
/mnt-nixos/boot/loader/entries/nixos-generation-9.conf
/mnt-nixos/boot/loader/entries/nixos-generation-10.conf
/mnt-nixos/boot/loader/loader.conf
/mnt-nixos/boot/loader/random-seed
/mnt-nixos/boot/loader/entries.srel
root@CM3588:~#
```
In particular, `/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi` pops out as what should / could be (a modified version of) the`.dtb` file.

> Could it be that it is missing rockchip/rk3588-friendlyelec-cm3588-nas.dtb ? Maybe because during the initial install, the mount of /boot failed due to missing /boot directory?

If my assumption is correct that `/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi` is essentially the `.dtb` file specified in `configuration.nix`, then it's present for sure.

> I'm surprised by the 2 in /dev/mmcblk2 - mine is /dev/mmcblk0 - but with labels or uuid that should not matter.

Yeah it's definitely a little weird - also that there's no `/dev/mmcblk1` device.
It turns out that `/dev/mmcblk0` is the external SD card on my board.  I found that out the hard way and then had to reflash the Debian image.

> Nice hostname 😎 I named my company Humilis, after Cumulus Humilis.

Thank you! That's also where I got the name from :)
Seems fitting as I'll essentially be building a private cloud with my CM3588.

> I'm using systemd.network, but that does not matter.

Did you only use `systemd.network.enable = true`, or is there more to your systemd network config?

> I do however, disable both firewall and dhcp:
> 
>     networking = {
>         dhcpcd.enable    =  false;
>         firewall.enable  =  false;
>         hostName         = "hamster";
>         useDHCP          =  false;
>         useNetworkd      =  true;
>     };
> 
>  Likely does not matter.

It's worth a shot, so I tried it. But yeah, these options (including disabling DHCP) doesn't seem to make a difference.

> If you can't make it work I'll recreate the stripped-down configuration.nix I used.

That's starting to look pretty good, thanks.

> I guess this requires upstream HDMI support then. Next step could be, find the patch that enables HDMI in Armbian and include that in your NixOS kernel stanza.

In the meantime I'll explore this option.

sanderhollaar, 2024-10-14:
>     3. Execute `mkdir -p /mnt-nixos/boot  &&  mount /dev/disk/by-label/NIXROOT /mnt-nixos  &&  mount /dev/disk/by-label/NIXBOOT /mnt-nixos/boot/ -o umask=077,fmask=0077,dmask=0077  &&  systemctl daemon-reload`

As you are posting this sequence of commands again: `mkdir -p /mnt-nixos/boot && mount [..] /mnt-nixos && mount [..] /mnt-nixos/boot/` does not do what you expect it to do.

The directory `boot` created in `/mnt-nixos` immediately gets hidden below the filesystem that gets mounted at `/mnt-nixos`.

The only reason that second `mount [..] /mnt-nixos/boot` can succeed, is because there happens to already be a directory `/boot` created in the filesystem mounted at `/mnt-nixos`. The sequence will fail with just created empty filesystems.

>     4. Execute `nixos-install --root /mnt-nixos --flake .#cumulus` from the git repo containing the nix configuration files.

> In order to create and format the partitions with the proper flags I followed the steps outlined in the [NixOS manual](https://nixos.org/manual/nixos/stable/#sec-installing-from-other-distro). Except I used a flake, which _should_ work just as well as direct usage of `configuration.nix`.

Yes, I install using flakes the same way you do.

> ```
> root@CM3588:~# find /mnt-nixos/boot/
> [..]
> /mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi
> [..]
> ```
> 
> In particular, `/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi` pops out as what should / could be (a modified version of) the`.dtb` file.
> 
> > Could it be that it is missing rockchip/rk3588-friendlyelec-cm3588-nas.dtb ? Maybe because during the initial install, the mount of /boot failed due to missing /boot directory?
> 
> If my assumption is correct that `/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi` is essentially the `.dtb` file specified in `configuration.nix`, then it's present for sure.

Looks good. For reference:
```
$ find /boot/ -name '*.dtb*'
/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi
/boot/EFI/nixos/vabylncn2iiiqld92ncza7vwrl7dp8pb-linux-6.11.3-rk3588-friendlyelec-cm3588-nas.dtb.efi
```
```
$ grep dtb /boot/loader/entries/nixos-generation-34.conf
devicetree /EFI/nixos/vabylncn2iiiqld92ncza7vwrl7dp8pb-linux-6.11.3-rk3588-friendlyelec-cm3588-nas.dtb.efi
```

> > I'm using systemd.network, but that does not matter.
> 
> Did you only use `systemd.network.enable = true`, or is there more to your systemd network config?

I have - stripped ipv6 details - but as I had the wrong interface name it was of no use:
```
    systemd = {
        network = {
            enable                  = true;
            networks = {
                enP4p65s0 = {
                    address         = [ "192.168.178.3/24" ];
                    matchConfig     = { Name = "enP4p65s0"; Type = "ether"; };
                    gateway         = [ "192.168.178.1" ];
                };
            };
        };
    };
```

> > If you can't make it work I'll recreate the stripped-down configuration.nix I used.
> 
> That's starting to look pretty good, thanks.

`hamster-configuration.nix` (I've removed the `users.users` stanza and some not relevant bits):
```
{ config, pkgs, ... }:

{
    imports = [ ./hamster-hardware.nix ];

    boot = {
        kernelPackages                  =  pkgs.linuxPackages_latest;
        loader = {
            efi.canTouchEfiVariables    =  true;
            systemd-boot = {
                enable                  =  true;
                configurationLimit      =  16;
            };
        };
    };

    hardware = {
        deviceTree = {
            enable                      =  true;
            name                        = "rockchip/rk3588-friendlyelec-cm3588-nas.dtb";
        };
    };

    networking = {
        dhcpcd.enable                   =  false;
        firewall.enable                 =  false;
        hostName                        = "hamster";
        useDHCP                         =  false;
        useNetworkd                     =  true;
    };

    services = {
        openssh = {
            enable                      =  true;
            settings = {
                KbdInteractiveAuthentication  =  false;
                PasswordAuthentication  =  false;
                PermitRootLogin         = "no";
            };
        };
    };

    system = {
        stateVersion                    = "24.11";
    };

    systemd = {
        network = {
            enable                      = true;
            networks = {
                enP4p65s0 = {
                    address             = [ "192.168.178.3/24" ];
                    matchConfig         = { Name = "enP4p65s0"; Type = "ether"; };
                    gateway             = [ "192.168.178.1" ];
                };
            };
        };
    };
}
```
`hamster-hardware.nix`:
```
{ config, lib, pkgs, modulesPath, ... }:

{
    imports = [ (modulesPath + "/installer/scan/not-detected.nix") ];

    boot.initrd.availableKernelModules = [ ];
    boot.initrd.kernelModules = [ ];
    boot.kernelModules = [ ];
    boot.extraModulePackages = [ ];

    fileSystems."/" = {
        label = "hamster";
        fsType = "f2fs";
    };

    fileSystems."/boot" = {
        label = "hamster_efi";
        fsType = "vfat";
    };

    nixpkgs.hostPlatform = lib.mkDefault "aarch64-linux";
    powerManagement.cpuFreqGovernor = lib.mkDefault "powersave";
}
```

jjpe, 2024-10-14:
> As you are posting this sequence of commands again: mkdir -p /mnt-nixos/boot && mount [..] /mnt-nixos && mount [..] /mnt-nixos/boot/ does not do what you expect it to do.
>
> The directory boot created [...]

Yeah IIRC the first time I had to mkdir that directory manually on `/mnt-nixos` after mounting it, which makes the mkdir in the compound mount command redundant altogether.
But all I really (think I) need is that `/mnt-nixos/boot` is available to `nixos-install` while being backed by `/dev/disk/by-label/NIXBOOT`; I don't manually manipulate the `/mnt-nixos/boot` dir (or the backing device) manually at all after creating and formatting the partitions. 
Therefore my reasoning was that the empty `/mnt-nixos/boot` directory on the mounted `/mnt-nixos` drive should work for the purposes of mounting `/dev/disk/by-label/NIXBOOT`), after which the installer would just do the right thing.
But maybe you know something I don't know, and more is needed than just making it available to `nixos-install`?  How I'd need to change the mount process would depend on the answer.

> Looks good. For reference:
>
>  $ find /boot/ -name '*.dtb*'
> /boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi [...]

I get similar output for both queries: 

```
root@CM3588:~# find /mnt-nixos/boot -name '*.dtb*'
/mnt-nixos/boot/EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi
```

```
root@CM3588:~# grep dtb /mnt-nixos/boot/loader/entries/nixos-generation-11.conf
devicetree /EFI/nixos/g78kk77p63yj65m885p578swjy19grp7-linux-6.11.2-rk3588-friendlyelec-cm3588-nas.dtb.efi
```

So I'm thinking that part is probably ok.

> `hamster-configuration.nix` (I've removed the users.users stanza and some not relevant bits):
> 
> `hamster-hardware.nix`:

Thank you for taking the effort.

Bafflingly, and the strongest hint yet something else might be going on here:
This minimal config (after adding my own `users.users` block) yields the same observable results as my own: It'll appear to boot a selected generation, print a couple of lines of text, then no-HDMI screen blank, after which complete non-responsiveness from the device until it's power cycled.

Of course, if the mounting above step truly is problematic as it is, then perhaps that is where (part of ) the solution lies.

sanderhollaar, 2024-10-14:
> Yeah IIRC the first time I had to mkdir that directory manually on `/mnt-nixos` after mounting it, which makes the mkdir in the compound mount command redundant altogether. But all I really (think [...]

Thing is, `/mnt-nixos/boot` is not available for `NIXBOOT` to be mounted on, after `NIXROOT` is mounted at `/mnt-nixos`.

Create directory:
```
$ sudo mkdir -p /mnt-nixos/boot
```
List contents of `/mnt-nixos`:
```
$ sudo ls -l /mnt-nixos/
total 4
drwx--x--x 2 root root 3452 Oct 14 17:25 boot
```
Mount filesystem at `/mnt-nixos`:
```
$ sudo mount -t tmpfs tmpfs /mnt-nixos
```
List contents of `/mnt-nixos`:
```
$ sudo ls -l /mnt-nixos/
total 0
```
The directory `boot` is now hidden underneath the mount.

To make `/mnt-nixos/boot` available for `NIXBOOT`, `mkdir` it in the empty filesystem mounted at `/mnt-nixos` as you apparently did the very first time.
```
$ sudo mkdir /mnt-nixos
$ sudo mount -t tmpfs tmpfs /mnt-nixos
$ sudo mkdir /mnt-nixos/boot
$ sudo mount -t tmpfs tmpfs /mnt-nixos/boot
$ df -h /mnt-nixos/ /mnt-nixos/boot
Filesystem      Size  Used Avail Use% Mounted on
tmpfs            32G     0   32G   0% /mnt-nixos
tmpfs            32G     0   32G   0% /mnt-nixos/boot
```
So I'm really just nitpicking as you kept posting the command sequence that created a directory that you could not use.

> Bafflingly, and the strongest hint yet something else might be going on here: This minimal config (after adding my own `users.users` block) yields the same observable results as my own: It'll appear [...]

In Armdebian, can you post the output of `journalctl -D /mnt-nixos/var/log/journal/` ?

jjpe, 2024-10-14:
> In Armdebian, can you post the output of journalctl -D /mnt-nixos/var/log/journal/ ?

Sure, here's the output:
```
root@CM3588:~# journalctl -D /mnt-nixos/var/log/journal/ | cat
Oct 14 21:52:50 hamster kernel: Booting Linux on physical CPU 0x0000000000 [0x412fd050]
Oct 14 21:52:50 hamster kernel: Linux version 6.11.2 (nixbld@localhost) (gcc (GCC) 13.3.0, GNU ld (GNU Binutils) 2.42) #1-NixOS SMP Fri Oct  4 14:38:57 UTC 2024
Oct 14 21:52:50 hamster kernel: KASLR enabled
Oct 14 21:52:50 hamster kernel: Machine model: FriendlyElec CM3588 NAS
Oct 14 21:52:50 hamster kernel: efi: EFI v2.7 by EDK II
Oct 14 21:52:50 hamster kernel: efi: SMBIOS=0xeac30000 SMBIOS 3.0=0xeac10000 MEMATTR=0xef2e0018 ACPI 2.0=0xeafcb018 RNG=0xeafcbe98 INITRD=0xea90cf98 MEMRESERVE=0xea90d118
Oct 14 21:52:50 hamster kernel: random: crng init done
Oct 14 21:52:50 hamster kernel: NUMA: No NUMA configuration found
Oct 14 21:52:50 hamster kernel: NUMA: Faking a node at [mem 0x0000000000200000-0x00000003ffffffff]
Oct 14 21:52:50 hamster kernel: NUMA: NODE_DATA [mem 0x3fdcb48c0-0x3fdcb7fff]
Oct 14 21:52:50 hamster kernel: Zone ranges:
Oct 14 21:52:50 hamster kernel:   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
Oct 14 21:52:50 hamster kernel:   DMA32    empty
Oct 14 21:52:50 hamster kernel:   Normal   [mem 0x0000000100000000-0x00000003ffffffff]
Oct 14 21:52:50 hamster kernel:   Device   empty
Oct 14 21:52:50 hamster kernel: Movable zone start for each node
Oct 14 21:52:50 hamster kernel: Early memory node ranges
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x0000000000200000-0x00000000006fffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000007c0000-0x00000000007effff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000007f0000-0x00000000083fffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x0000000008400000-0x00000000093fffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x0000000009400000-0x00000000ea05ffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000ea060000-0x00000000ea8fffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000ea900000-0x00000000eab3ffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000eab40000-0x00000000eafbffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000eafc0000-0x00000000effaffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000effb0000-0x00000000effdffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000000effe0000-0x00000000efffffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x0000000100000000-0x00000003fbffffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000003fc000000-0x00000003fc4fffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000003fc500000-0x00000003ffefffff]
Oct 14 21:52:50 hamster kernel:   node   0: [mem 0x00000003fff00000-0x00000003ffffffff]
Oct 14 21:52:50 hamster kernel: Initmem setup node 0 [mem 0x0000000000200000-0x00000003ffffffff]
Oct 14 21:52:50 hamster kernel: On node 0, zone DMA: 512 pages in unavailable ranges
Oct 14 21:52:50 hamster kernel: On node 0, zone DMA: 192 pages in unavailable ranges
Oct 14 21:52:50 hamster kernel: cma: Reserved 32 MiB at 0x00000000ed200000 on node -1
Oct 14 21:52:50 hamster kernel: psci: probing for conduit method from DT.
Oct 14 21:52:50 hamster kernel: psci: PSCIv1.1 detected in firmware.
Oct 14 21:52:50 hamster kernel: psci: Using standard PSCI v0.2 function IDs
Oct 14 21:52:50 hamster kernel: psci: Trusted OS migration not required
Oct 14 21:52:50 hamster kernel: psci: SMC Calling Convention v1.2
Oct 14 21:52:50 hamster kernel: percpu: Embedded 55 pages/cpu s100824 r8192 d116264 u225280
Oct 14 21:52:50 hamster kernel: pcpu-alloc: s100824 r8192 d116264 u225280 alloc=55*4096
Oct 14 21:52:50 hamster kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 [0] 4 [0] 5 [0] 6 [0] 7
Oct 14 21:52:50 hamster kernel: Detected VIPT I-cache on CPU0
Oct 14 21:52:50 hamster kernel: CPU features: detected: GIC system register CPU interface
Oct 14 21:52:50 hamster kernel: CPU features: detected: Virtualization Host Extensions
Oct 14 21:52:50 hamster kernel: CPU features: kernel page table isolation forced ON by KASLR
Oct 14 21:52:50 hamster kernel: CPU features: detected: Kernel page table isolation (KPTI)
Oct 14 21:52:50 hamster kernel: CPU features: detected: ARM errata 1165522, 1319367, or 1530923
Oct 14 21:52:50 hamster kernel: alternatives: applying boot alternatives
Oct 14 21:52:50 hamster kernel: Kernel command line: initrd=\EFI\nixos\82hnfbja1a1xfsnpmd9mdp9x78n7j8j0-initrd-linux-6.11.2-initrd.efi init=/nix/store/k53qfcxcr9dbmnhxafvfgfgr3c2b80ap-nixos-system-hamster-24.11.20241006.c31898a/init loglevel=4
Oct 14 21:52:50 hamster kernel: Dentry cache hash table entries: 2097152 (order: 12, 16777216 bytes, linear)
Oct 14 21:52:50 hamster kernel: Inode-cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
Oct 14 21:52:50 hamster kernel: Fallback order for Node 0: 0
Oct 14 21:52:50 hamster kernel: Built 1 zonelists, mobility grouping on.  Total pages: 4128064
Oct 14 21:52:50 hamster kernel: Policy zone: Normal
Oct 14 21:52:50 hamster kernel: mem auto-init: stack:all(zero), heap alloc:on, heap free:off
Oct 14 21:52:50 hamster kernel: software IO TLB: area num 8.
Oct 14 21:52:50 hamster kernel: software IO TLB: mapped [mem 0x00000000e4cc0000-0x00000000e8cc0000] (64MB)
Oct 14 21:52:50 hamster kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=8, Nodes=1
[... cut ...]
Oct 14 21:54:43 hamster systemd[1]: Finished System Power Off.
Oct 14 21:54:43 hamster systemd[1]: Reached target System Power Off.
Oct 14 21:54:43 hamster systemd[1]: Shutting down.
Oct 14 21:54:43 hamster systemd-shutdown[1]: Syncing filesystems and block devices.
Oct 14 21:54:43 hamster systemd-shutdown[1]: Sending SIGTERM to remaining processes...
Oct 14 21:54:43 hamster systemd-journald[481]: Received SIGTERM from PID 1 (systemd-shutdow).
Oct 14 21:54:43 hamster systemd-journald[481]: Journal stopped
root@CM3588:~#
```

sanderhollaar, 2024-10-15:
> > In Armdebian, can you post the output of journalctl -D /mnt-nixos/var/log/journal/ ?
> 
> Sure, here's the output:
> 
> ```
> root@CM3588:~# journalctl -D /mnt-nixos/var/log/journal/ | cat
> [..]
> Oct 14 21:52:51 hamster systemd[1]: Finished CPU Frequency Setup.
> Oct 14 21:52:51 hamster systemd[1]: Reached target Network.
> Oct 14 21:52:51 hamster systemd[1]: Reached target Host and Network Name Lookups.
> Oct 14 21:52:51 hamster systemd[1]: Reached target User and Group Name Lookups.
> Oct 14 21:52:51 hamster systemd[1]: Starting D-Bus System Message Bus...
> Oct 14 21:52:51 hamster systemd[1]: Starting SSH Daemon...
> Oct 14 21:52:52 hamster systemd[1]: Starting User Login Management...
> Oct 14 21:52:52 hamster systemd[1]: Started Nix Daemon.
> Oct 14 21:52:52 hamster systemd[1]: Started D-Bus System Message Bus.
> Oct 14 21:52:52 hamster systemd-logind[738]: New seat seat0.
> Oct 14 21:52:52 hamster systemd-logind[738]: Watching system buttons on /dev/input/event2 (rk805 pwrkey)
> Oct 14 21:52:52 hamster systemd[1]: Started User Login Management.
> Oct 14 21:52:52 hamster nix-daemon[746]: accepted connection from pid 730, user j
> Oct 14 21:52:52 hamster hm-activate-j[705]: Activating checkFilesChanged
> Oct 14 21:52:52 hamster hm-activate-j[705]: Activating checkLinkTargets
> Oct 14 21:52:52 hamster hm-activate-j[705]: Activating writeBoundary
> Oct 14 21:52:52 hamster hm-activate-j[705]: Activating installPackages
> Oct 14 21:52:52 hamster hm-activate-j[705]: Activating linkGeneration
> Oct 14 21:52:52 hamster hm-activate-j[705]: Creating profile generation 1
> Oct 14 21:52:52 hamster nix-daemon[746]: accepted connection from pid 776, user j
> Oct 14 21:52:52 hamster nix-daemon[746]: accepted connection from pid 780, user j
> Oct 14 21:52:52 hamster hm-activate-j[705]: Creating home file links in /home/j
> Oct 14 21:52:54 hamster kernel: r8169 0004:41:00.0 enP4p65s0: Link is Up - 1Gbps/Full - flow control off
> Oct 14 21:52:54 hamster systemd-networkd[560]: enP4p65s0: Gained carrier
> Oct 14 21:52:54 hamster systemd-timesyncd[665]: Network configuration changed, trying to establish connection.
> Oct 14 21:52:54 hamster hm-activate-j[705]: Activating onFilesChange
> Oct 14 21:52:54 hamster systemd-resolved[634]: Switching to fallback DNS server 1.1.1.1#cloudflare-dns.com.
> Oct 14 21:52:54 hamster hm-activate-j[705]: Activating reloadSystemd
> Oct 14 21:52:54 hamster hm-activate-j[705]: User systemd daemon not running. Skipping reload.
> Oct 14 21:52:54 hamster systemd[1]: Finished Home Manager environment for j.
> Oct 14 21:52:54 hamster systemd[1]: Starting Permit User Sessions...
> Oct 14 21:52:54 hamster systemd[1]: Finished Permit User Sessions.
> Oct 14 21:52:54 hamster systemd[1]: Started Getty on tty1.
> Oct 14 21:52:54 hamster systemd[1]: Started Serial Getty on ttyS2.
> Oct 14 21:52:54 hamster systemd[1]: Reached target Login Prompts.
> ```

This seems a successful boot. Can you try to logon?

> ```
> Oct 14 21:52:55 hamster systemd-networkd[560]: enP4p65s0: Gained IPv6LL
> Oct 14 21:52:55 hamster systemd-timesyncd[665]: Network configuration changed, trying to establish connection.
> Oct 14 21:52:56 hamster systemd-timesyncd[665]: Network configuration changed, trying to establish connection.
> Oct 14 21:52:58 hamster systemd-timesyncd[665]: Network configuration changed, trying to establish connection.
> Oct 14 21:52:58 hamster sshd-pre-start[745]: Generating public/private rsa key pair.
> Oct 14 21:52:58 hamster sshd-pre-start[745]: Your identification has been saved in /etc/ssh/ssh_host_rsa_key
> Oct 14 21:52:58 hamster sshd-pre-start[745]: Your public key has been saved in /etc/ssh/ssh_host_rsa_key.pub
> [..]
> Oct 14 21:52:58 hamster sshd-pre-start[1177]: Generating public/private ed25519 key pair.
> Oct 14 21:52:58 hamster sshd-pre-start[1177]: Your identification has been saved in /etc/ssh/ssh_host_ed25519_key
> Oct 14 21:52:58 hamster sshd-pre-start[1177]: Your public key has been saved in /etc/ssh/ssh_host_ed25519_key.pub
> [..]
> Oct 14 21:52:58 hamster systemd[1]: Started SSH Daemon.
> Oct 14 21:52:58 hamster sshd[1178]: Server listening on 0.0.0.0 port 22.
> Oct 14 21:52:58 hamster sshd[1178]: Server listening on :: port 22.
> ```

Network:

> ```
> Oct 14 21:52:50 hamster systemd-networkd[560]: /etc/systemd/network/enP4p65s0.network:6: Address '192.168.1.145' is specified without prefix length. Assuming the prefix length is 32. Please [...]

Could it be that your address is missing trailing `/24` like `address = [ "192.168.178.3/24" ];` ?

> ```
> Oct 14 21:52:51 hamster systemd-networkd[560]: eth0: Interface name change detected, renamed to enP4p65s0.
> Oct 14 21:52:51 hamster kernel: r8169 0004:41:00.0 enP4p65s0: renamed from eth0
> ```

> ```
> Oct 14 21:54:42 hamster systemd-logind[738]: Power key pressed short.
> Oct 14 21:54:42 hamster systemd-logind[738]: Powering off...
> Oct 14 21:54:42 hamster systemd-logind[738]: System is powering down.
> [..]
> Oct 14 21:54:43 hamster systemd[1]: Reached target System Shutdown.
> Oct 14 21:54:43 hamster systemd[1]: Reached target Late Shutdown Services.
> Oct 14 21:54:43 hamster systemd[1]: systemd-poweroff.service: Deactivated successfully.
> Oct 14 21:54:43 hamster systemd[1]: Finished System Power Off.
> Oct 14 21:54:43 hamster systemd[1]: Reached target System Power Off.
> Oct 14 21:54:43 hamster systemd[1]: Shutting down.
> Oct 14 21:54:43 hamster systemd-shutdown[1]: Syncing filesystems and block devices.
> Oct 14 21:54:43 hamster systemd-shutdown[1]: Sending SIGTERM to remaining processes...
> Oct 14 21:54:43 hamster systemd-journald[481]: Received SIGTERM from PID 1 (systemd-shutdow).
> Oct 14 21:54:43 hamster systemd-journald[481]: Journal stopped
> root@CM3588:~#
> ```

jjpe, 2024-10-15:
> This seems a successful boot. Can you try to logon?

When I try to log into the root account and then execute `shutdown now`, it actually takes effect, indicating success. So that's definitely progress.

However, after I log in, I execute `ip a a 192.168.178.3/24 dev $(ip a s | grep -m1 MULTICAST | awk -F ':' '{print $2}')`, but that doesn't seem to make any difference to my ability to ping the device (I ping the hostname as I have added a static route to my router for the device):
```
j@~❯ ping 192.168.1.145                                                                                            09:13:34.232301834
PING 192.168.1.145 (192.168.1.145) 56(84) bytes of data.
From 192.168.1.150 icmp_seq=1 Destination Host Unreachable
From 192.168.1.150 icmp_seq=2 Destination Host Unreachable
From 192.168.1.150 icmp_seq=3 Destination Host Unreachable
From 192.168.1.150 icmp_seq=4 Destination Host Unreachable
From 192.168.1.150 icmp_seq=5 Destination Host Unreachable
From 192.168.1.150 icmp_seq=6 Destination Host Unreachable
^C
--- 192.168.1.145 ping statistics ---
8 packets transmitted, 0 received, +6 errors, 100% packet loss, time 7201ms
```
So something is still off with the network config in particular.
I just had a thought: Since the eMMC device is (for some reason) consistently listed as `/dev/mmcblk2`, perhaps the network interface is enumerated differently too? The only problem with that is that IIUC the `ip` oneliner should automatically fetch the correct interface.

> Could it be that your address is missing trailing /24 like address = [ "192.168.178.3/24" ]; ?

I tried adding the `/24` subnet mask and reinstalling nixos, but that seems to make no difference in my ability to log into the device using ssh.

What I'll try next is adding the subnet mask to both the `address` and `gateway` attributes, see if that helps.

jjpe, 2024-10-15:
Turns out I already had those blocks in my config.
So I rebooted the CM3588, tried ssh'ing in, and... it works!

If I ever meet you in meatspace, I definitely owe you a beer :)

jjpe, 2024-10-16:
There's 1 final issue I need to tackle: The nixos config provides LAN access to I can ssh into it, but no internet access, making updating the system a bit difficult.

So I create a default route using `ip`, which does work:
```
[j@hamster:~]$ ping google.com
ping: connect: Network is unreachable

[j@hamster:~]$ sudo ip route add default via 192.168.1.1 dev enP4p65s0
[sudo] password for j:

[j@hamster:~]$ ping google.com
PING google.com (172.217.23.206) 56(84) bytes of data.
64 bytes from prg03s05-in-f206.1e100.net (172.217.23.206): icmp_seq=1 ttl=56 time=3.25 ms
64 bytes from prg03s05-in-f206.1e100.net (172.217.23.206): icmp_seq=2 ttl=56 time=2.74 ms
64 bytes from prg03s05-in-f206.1e100.net (172.217.23.206): icmp_seq=3 ttl=56 time=2.73 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 2.725/2.904/3.254/0.247 ms
```

The only issue is, at the moment it's necessary that I execute it on every reboot, which is undesirable.
Ive tried adding `networking.interfaces.enP4p65s0.ipv4.routes`:
```nix
  networking = {
    dhcpcd.enable = false;
    firewall.enable = false;
    hostName = "cumulus";
    useDHCP = false;
    useNetworkd = true;
    interfaces.enP4p65s0.ipv4.routes = [{ # I was hoping this would do the trick
      address = "192.168.1.145/24";
      prefixLength = 24;
      via = "192.168.1.1/24";
    }];
  };
```
But that just borks the LAN connection, necessitating a reinstall from Armbian.

Any idea how I can "nixify" the `ip` command, i.e. any idea how I can capture that default route in my nixos config files?

[37 more replies]

closed 2025-02-28 by mariobalanica