edk2-porting/edk2-rk3588 #143: Synchronous Exception when loading kernel from eMMC (works on USB/SD)

vsnrain, 2024-06-16:
Regardless whether it's in ACPI mode or DTB mode (with or without dtb overrides), if I load kernel (rockchip 5.10 or 6.1+) from eMMC it fails with Synchronous Exception. Bootloader (systemd-boot) works fine.
If I put the exactly same eMMC card into USB or SD adapter and boot from USB/SD it works perfectly. 
Tested on Rock Pi 5B and Orange Pi 5 Plus, releases 0.8~0.9.1 and nightly, multiple eMMC chip manufacturers, same issue.
I will attach exception log from Orange Pi 5 Plus.
[opi5plus.txt](https://github.com/user-attachments/files/15858738/opi5plus.txt)

BigfootACA, 2024-06-16:
Can you try using `llvm-symbolizer` or `addr2line` to get the exact location of the crash?

mariobalanica, 2024-06-20:
> If I put the exactly same eMMC card into USB or SD adapter and boot from USB/SD it works perfectly.

You mean plugging the eMMC module into the onboard SD card slot?

vsnrain, 2024-06-22:
@BigfootACA thanks, I'll try building and finding where the crash is.

@mariobalanica yes, that's correct. So it leads me to believe it's not the issue with the kernel or the eMMC module itself.

mariobalanica, 2024-06-22:
Could you send me a disk .img with what you're trying to boot so I can take a look?

I haven't been able to repro this issue with the Armbian or Arch Linux images linked under the latest release.

vsnrain, 2024-06-22:
sorry, by disk .img what exactly do you mean? If the EDK2 SPI image, I tried nightly builds and 0.8, 0.9, 0.9.1 from this repo.
If you mean the OS, then there is no .img per se, I am trying to boot just the kernel off fat32 ESP partition. Tried [6.9.3-1-ARCH](https://archlinuxarm.org/packages/aarch64/linux-aarch64) kernel from Arch ARM and [5.10.160-1](https://drive.google.com/drive/folders/1_UAZKe0u2t_Zp52fNCY5Fo2wbm0KSTWS) from the image provided by orangepi. Both of them seems to work fine when booting off USB or SD, but get synchronous exception on eMMC.

mariobalanica, 2024-06-22:
Turns out it's a bug in the eMMC controller. I've pushed a workaround here: 02d150251c1e4ef6e4aa68336d2c65db0c7d2e7c

Let me know if you're still having issues.

vsnrain, 2024-06-23:
Yeah, looks like this fixes the issue, it boots properly now, thank you!

closed 2024-06-23 by vsnrain