edk2-porting/edk2-rk3588 #146: Install ACPI to platform

phanbagiabao2001, 2024-06-28:
How did you install ACPI tables to boot your platform?
Do you know how to debug ACPI in linux kernel?

binarymaster, 2024-06-28:
ACPI tables will be completely useless on Linux on this group of devices, as drivers lack bindings for devices listed in the ACPI tables.

phanbagiabao2001, 2024-06-28:
https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#in-acpi-mode
Here Supported OSes in ACPI mode, what does this mean?
Can these platforms use ACPI to boot OS like Raspberry Pi?

mariobalanica, 2024-06-28:
All the ACPI-related bits can be found in `AcpiPlatformDxe` and the `AcpiTables` directory.

> Here Supported OSes in ACPI mode, what does this mean?
Can these platforms use ACPI to boot OS like Raspberry Pi?

That's a list of OSes that have been tested to boot in ACPI mode, exactly. 

But for Linux, like mentioned in the readme, you should be using a Device Tree. ACPI is no longer being supported/tested here for anything other than Windows (because it's frankly a time sink).

phanbagiabao2001, 2024-06-29:
Can you tell me what method did you use to construct the ACPI for linux kernel? Or did we construct only the necessary tables mentioned here https://docs.kernel.org/arch/arm64/arm-acpi.html#booting-using-acpi-tables in EDK2 and linux will load it?
And how did you use to debug the kernel message before booting successfully?

mariobalanica, 2024-06-29:
There isn't a universal "method" per se. It all depends on what you're trying to achieve, what kind of devices you want to bring up...

Generally, you can look at other platforms (like this one, the RPi port, etc.) and read the ACPI specification along to get a hang of it. You can find other helpful resources online as well.

The Pi 5 port I did should be a good starting point: https://github.com/worproject/edk2-platforms/tree/rpi5-dev/Platform/RaspberryPi/RPi5, particularly this commit is pretty self-contained: https://github.com/worproject/edk2-platforms/commit/0f5d150305d726ec55de7ee53490ed2b63c15a23 with the bare minimum tables and devices exposed in DSDT to boot any ACPI-compatible OS, including Linux.

Obviously, these tables should be changed to reflect your actual hardware.

closed 2024-06-29 by mariobalanica

phanbagiabao2001, 2024-06-29:
> There isn't a universal "method" per se. It all depends on what you're trying to achieve, what kind of devices you want to bring up...

My goal is to install ACPI to STM32MP25 platform so that it can boot linux with ACPI.
However, after creating the necessary tables I can not boot the OS. Of course my tables are not initially exact and I want to find where are the mistakes by debugging. The problem here is that I can't not extract any kernel messages to verify because the boot process stop when it just displays EFI stub.

This is the way I did to install ACPI in my platform. Do you know any ways to debug or is the method I am using not practical?