edk2-porting/edk2-rk3588 #152: Support for Radxa ROCK 5C

ondrej-ilcik, 2024-08-05:
Good afternoon,
Since Radxa ROCK 5A and 5B are supported, could you please add support of ROCK 5C as well?
https://radxa.com/products/rock5/5c/
Thank you very much

kwankiu, 2024-08-07:
I have added Radxa Rock 5C on my fork

https://github.com/kwankiu/edk2-rk3588

I found Rock 5C and Radxa CM5 almost identical to the Rock 5A

But I dont have a rock 5c with me, so you will have to test it out, let me know what's working and what's not. It has been tested that it currently works well on device tree mode, but have not tested on ACPI with any Linux or Windows.

Ph42oN, 2024-08-11:
I tested booting the rock 5A image on my rock 5C lite and it boots fine.

ondrej-ilcik, 2024-08-11:
Good afternoon and sorry for late response.
Tested on:
VMware ESXi ARM Edition
- USB2: Working
- USB3: Working
- NIC: Detected by system, Link status detected(Up/Down), Communication not Working (probably driver issue)
- SATA (Penta SATA HAT): Controller detected, disk detected (disk presence delays disk boot significantly), any attempt to use the disk fails

Windows 11
- USB2: Not working 
- USB3: Working
- NIC: couldn't install system due to missing SD/eMMC so couldnt test
- SD/eMMC: Not detected, possibly driver issue
- SATA (Penta SATA HAT): Not detected

FreeBSD 14.1
- USB2: Working
- USB3: Working
- NIC: Not detected

Please note that I'm kind of new in this and still learning, so all tests were done just with default setup and issues may not be caused by your firmware but just by lack of knowledge on my side.

Anyway as my goal was to have mainly ESXi ARM to play with, your firmware helped me alot.
Thank you very much!

lin72h, 2024-08-12:
I haven't know any success story about FreeBSD on RK3588, @ondrej-ilcik Can you make FreeBSD 14.1 work at all beside NIC driver?

Ph42oN, 2024-08-17:
I tested the rock 5c fork on my 5c lite, so far everything appears to be working fine using 6.1 kernel from https://github.com/radxa/kernel. I tested USB ports, ethernet and wifi, they all work. Wifi requires this driver: https://github.com/radxa-pkg/aic8800

This UEFI also unlocked CPU cores, now i have 8 cores working instead of default 6 on 5c lite. GPU is not working, but that might be just faulty on my board, i haven't been able to get anything but software rendering to run.

closed 2025-02-28 by mariobalanica