edk2-porting/edk2-rk3588 #157: Orange PI 5 PRO

Kosa53, 2024-09-01:
I am eagerly awaiting support for Orange PI 5 Pro because the image for Orange PI 5 (and plus) unfortunately do not work on my version. Orange PI 5 version - zero reaction. In the version with Orange PI 5 Plus, at least the green diode lights up but nothing else happens. I kindly ask you to add support for Orange PI 5 Pro. Thank you.

MAHDTech, 2024-09-25:
Hi @Kosa53,

I was able to get Orange PI 5 Pro to boot with UEFI and then I loaded the [live ISO](https://github.com/amazingfate/rk3588-live-iso) but it's missing the NIC, EMMC and NVMe at this stage.

The last hurdle is I need to get a device tree for the Pro so these will work then I can submit a working [PR](https://github.com/edk2-porting/edk2-rk3588/pull/162).

It looks like I need a modern 6.8+ kernel from what I can tell but still looking into it.

ndenev, 2024-09-29:
Would the Orange Pi 5 Max be suppored by this?

Marcinoo97, 2024-09-29:
no until someone adds specific dtb and platform file with correct config data for the board

mariobalanica, 2024-12-05:
Last time I checked, OPi 5 Pro had some obscure Ethernet controller (why?) that lacks support from any OS aside from their downstream Linux kernel. So until that changes, this board will not be supported here, as I believe it adds no value to the project.

I would suggest looking into other hardware that's already supported.

closed 2024-12-05 by mariobalanica