edk2-porting/edk2-rk3588 #165: WoR Images

macooper59, 2024-10-07:
Hello Everyone,

Which on of the images is for the Orange Pi5 Plus? [WoR Images]https://github.com/edk2-porting/edk2-rk3588/releases

Rippanda12, 2024-10-07:
https://objects.githubusercontent.com/github-production-release-asset-2e65be/567314232/d2ab0fe1-4ed1-43e2-b0f5-1acde0362b84?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F20241007%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20241007T074149Z&X-Amz-Expires=300&X-Amz-Signature=bbe82a2505a98607ea6a6d90bf95eea7a9eebe704c2f1e974a7162426086c832&X-Amz-SignedHeaders=host&response-content-disposition=attachment%3B%20filename%3D&response-content-type=application%2Foctet-stream

orangepi-5plus_UEFI_Release_v0.11.2.img

macooper59, 2024-10-07:
> https://objects.githubusercontent.com/github-production-release-asset-2e65be/567314232/d2ab0fe1-4ed1-43e2-b0f5-1acde0362b84?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=releaseassetproduction%2F2 [...]

Thank you sir

macooper59, 2024-10-08:
It says the link doesn't work?

[image: image.png]

Thanks,

Michael Cooper
*Certified Linux Professional*
*Docker/Kubernetes Trained*
RPi Hobby Blog <http://www.rpihobby.us>

mariobalanica, 2024-12-05:
GitHub may truncate the list of assets. There's a link at the bottom to show all of them.

closed 2024-12-05 by mariobalanica