edk2-porting/edk2-rk3588 #168: Cannot detect SATA or eMMC in AHCI

Xvatten, 2024-11-14:
## The Problem

Cannot detect my SATA disks, nor the eMMC disk.

## Information about my setup

### AHCI MODE
I'm running Fedora Server 41 on the Rock 5 ITX

`$ uname -r
6.11.7-300.fc41.aarch64`

`$ lsblk`
`NAME           MAJ:MIN RM   SIZE RO TYPE MOUNTPOINTS`
`zram0           252:0    0     8G  0 disk [SWAP]`
`nvme0n1         259:0    0 465.8G  0 disk `
`├─nvme0n1p1     259:1    0   600M  0 part /boot/efi`
`├─nvme0n1p2     259:2    0     1G  0 part /boot`
`└─nvme0n1p3     259:3    0 464.2G  0 part `
`  └─fedora-root 253:0    0    15G  0 lvm  /`

Boots but cannot see my SATA disks nor the eMMC.

### DEVICETREE MODE
With devicetree enabled doesn't boot anything. Black screen, no ssh, router doesn't assign an IP to the machine so i assume it just doesn't boot.

Found `rk3588-rock-5-itx.dtb` file in `/boot/dtb*/rockchip/` folders

Xvatten, 2024-11-19:
Thanks for your comment zh-jq.

I put the dtb at `/boot/rk3588-rock-5-itx.dtb` 
and added it to `/etc/default/grub` 
looking like this: `GRUB_DEFAULT_DTB=rk3588-rock-5-itx.dtb`

In grub it looks like this when editing the command list:
`devicetree ($root)/rk3588-rock-5-itx.dtb`

grub prints this messages when choosing a grub entry:
```
EFI stab: Decompressing Linux Kernel...
EFI stab: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stab: Using DTB from configuration table
EFI stab: Exiting boot services...
```

After that my keyboard LEDs powers down and nothing happens.

Any other suggestions on what I can try is most welcome.

UnicodeMan, 2024-11-19:
Base dtb has to be in a fat32 efi partition, /boot/efi/dtb/base/rk3588-rock-5-itx.dtb

The weird thing is, at least with uboot, if you boot kernel image with wrong/missing dtb you can fry your nvme ssd

At least it happened to me.
One ssd i got replaced under warranty after that
The other bounced back in about half a year. At first it would drop connection, later it no longer showed up (with an enclosure), and half a year when i tried it again dropping storage device gradually improved until it started working stable.

zh-jq, 2024-11-20:
> Any other suggestions on what I can try is most welcome.

you can use serial port to login

Xvatten, 2024-11-28:
I install Armbian instead couldn't get SATA to work on Fedora Server.

closed 2024-11-28 by Xvatten