edk2-porting/edk2-rk3588 #169: subhook submodule is 404

UnicodeMan, 2024-11-15:
`git clone https://github.com/edk2-porting/edk2-rk3588.git --recurse-submodules`
fails to clone `https://github.com/Zeex/subhook/` submodule, when manually accessing the submodule through web browser, Github returns 404.

binarymaster, 2024-11-15:
Same problem as in https://github.com/microsoft/mu_basecore/issues/1208

It should be possible to fix by replacing URL `https://github.com/Zeex/subhook` with `https://github.com/tianocore/edk2-subhook`

binarymaster, 2024-11-15:
This has been addressed in upstream, git submodule should be updated.

bicebird, 2024-11-26:
Tried building for rock5b plus with this branch but build.py failed with error 7000
(sorry if not the right place / not enough detail as still getting my head around github)

closed 2024-12-05 by mariobalanica