edk2-porting/edk2-rk3588 #170: BSA ACS hangs after ECAM info

hrw, 2024-11-20:
I think that https://github.com/ARM-software/bsa-acs/issues/397 belongs here rather than to bsa-acs project.

Decided to run BSA ACS on something other than usual. Took RK3588 SBC, put EDK2 on it and ran bsa.efi:

```
FS2:\> bsa.efi -v 1

 BSA Architecture Compliance Suite
          Version 1.0.8

 Starting tests with print level :  1

 Creating Platform Information Tables
 Using SMC as PSCI conduit
 Creating PE INFO table
  MADT is at EAFCB098 and length is 2D4
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 0 PE num 0
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 100 PE num 1
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 200 PE num 2
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 300 PE num 3
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 400 PE num 4
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 500 PE num 5
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 600 PE num 6
  Flags 1
  PE Enabled 1, Online Capable 0
 MPIDR 700 PE num 7
 PE_INFO: Number of PE detected       :    8
 PE_INFO: Primary PE index       :    0
 Creating GIC INFO table
  MADT is at EAFCB098 and length is 2D4
  GIC DIS base FE600000
  GICR RD base FE680000
  GICR RD Length 100000

       gic version from info table = 3  GIC INFO: GIC version                :    v3
 GIC_INFO: Number of GICD             :    1
 GIC_INFO: Number of GICR RD          :    1
 GIC_INFO: Number of ITS              :    0
 Creating TIMER INFO table
  GTDT is at EAFCBD98 and length is 60
  GTDT revision is at 2
 TIMER_INFO: Number of system timers  :    0
 Creating WATCHDOG INFO table
 WATCHDOG_INFO: Number of Watchdogs   :    0
 Creating PCIe INFO table
  Ecam Index = 0
  Base Address = 0x900000000
  Segment   = 0x0
  Start Bus = 0x1
  End Bus   = 0x1
  Ecam Index = 1
  Base Address = 0x940000000
  Segment   = 0x1
  Start Bus = 0x1
  End Bus   = 0x1
  Ecam Index = 2
  Base Address = 0x980000000
  Segment   = 0x2
  Start Bus = 0x1
  End Bus   = 0x1
  Ecam Index = 3
  Base Address = 0x9C0000000
  Segment   = 0x3
  Start Bus = 0x1
  End Bus   = 0x1
  Ecam Index = 4
  Base Address = 0xA00000000
  Segment   = 0x4
  Start Bus = 0x1
  End Bus   = 0x1
  Ecam Index = 5
  Base Address = 0xA40000000
  Segment   = 0x0
  Start Bus = 0x0
  End Bus   = 0x0
  Ecam Index = 6
  Base Address = 0xA40400000
  Segment   = 0x1
  Start Bus = 0x0
  End Bus   = 0x0
  Ecam Index = 7
  Base Address = 0xA40800000
  Segment   = 0x2
  Start Bus = 0x0
  End Bus   = 0x0
  Ecam Index = 8
  Base Address = 0xA40C00000
  Segment   = 0x3
  Start Bus = 0x0
  End Bus   = 0x0
  Ecam Index = 9
  Base Address = 0xA41000000
  Segment   = 0x4
  Start Bus = 0x0
  End Bus   = 0x0
 PCIE_INFO: Number of ECAM regions    :    10
```

And it hangs here. Added some 'val_print()' calls into code and it goes to `val_pcie_enumerate()` and then to `val_pcie_create_device_bdf_table()` and ends there.

All PCIe devices work fine when I boot Linux in ACPI mode.

[ACPI tables printed with 'acpiview' command](https://github.com/user-attachments/files/17830338/tables.log)

OK, got to the point when it hangs on reading 0x40100000 address in pal_mmio_read().

Next step: check ACPI tables again.

hrw, 2024-11-20:
Dropped second entry in MCFG and it worked a bit better.

hrw, 2024-11-20:
ACPI tables are in terrible state. Without MCFG entries for segment 1 (pcie30x2 which is not present on NanoPC-T6) BSA ACS starts and do the job.

Still weird to see 1032 PCIe devices (8 existing ones + 1024 not existing root ports).

hrw, 2024-11-20:
[log of whole BSA ACS run](https://github.com/user-attachments/files/17832426/bsa-acs.log)

mariobalanica, 2024-12-05:
Correct, nothing can be done to make ECAM look right on this platform. I have added some compatibility modes to make it work in Windows and Linux, but that's about as good as it's going to get with ACPI...

closed 2024-12-05 by mariobalanica