edk2-porting/edk2-rk3588 #173: Wifi support on orange pi 5 plus (Windows 11)

awesomefroggy29, 2024-12-08:
When I boot into windows 11 (or any uefi os), the wifi doesn’t work at all. The only workaround for that was Ethernet, but I don’t always have Ethernet wherever I go. Is there any way to fix this? I got my orange pi from here: https://www.amazon.com/Rockchip-Computer-Frequency-Bluetooth-Android/dp/B0CKP3YCG4/ref=sr_1_2?crid=WK1MODKGGOA2&dib=eyJ2IjoiMSJ9.05X4H8ckU5S2IHb3fnlLBlr-yoQ_EN_9QuyAnw-8U3QVTzHU54LnExWhSCK2fdMKnSTNh9695aeAPJxxcA5JIPSy8INKyF8yxFTWIhWRHppDqaICjkNw2NoiF7tnMJ1qjVDwHIK6y1ceRWRRf0FiRBjAyorzqJniPKRfP-TbPhCFxScgGF4juEiCbFNdvMpiJEvOemQgKCn6DcR2XhnfqX3OEiv4u_OwMFfOLZdgrxY.RIDhCkwhrq5KjImVSVvLvZypY4Qv4IhvC6EpJzH25Y8&dib_tag=se&keywords=orange%2Bpi%2B5%2Bplus%2B32gb%2Bwith%2B256gb&qid=1733620993&sprefix=Orange%2Bpi%2B5%2Bplus%2B256gb%2B%2Caps%2C275&sr=8-2&th=1

mariobalanica, 2024-12-08:
Look for AW-CB178NF (Marvell 88W8897 chipset), this is the only Wi-Fi card with a working driver under Windows on the RK3588.

As for OSes other than Linux, they may also lack drivers for that chipset.

closed 2024-12-08 by mariobalanica

awesomefroggy29, 2024-12-09:
Ok, thanks!

MadGoatHaz, 2024-12-09:
lurk ninja:
https://www.amazon.com/gp/product/B07FMJDK1N/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

awesomefroggy29, 2024-12-09:
and one more question,

do any of you know when hardware acceleration will be supported?