edk2-porting/edk2-rk3588 #174: Support fo Orange Pi 5 Max

DummyTitan, 2024-12-13:
Before starting, thanks for the project, we really need a UEFI for the RK3588, now i want to make a branch for the Orange Pi 5 Max, im a programmer but have no experience with this type of things. I already tried the uefi for the Orange Pi 5 plus since it has similar things but it didn't work (I imagine that is because of the dtb and drivers). As such i want to request support for the Orange Pi 5 Max if possible, if you guys are just to occupied with other projects, I would love some information on where to start so that I can at least attemp to make it my self. I really don't know anything about where to start so I would need a bit of help to know the structure and what to add. Thanks.

DummyTitan, 2024-12-13:
Just in case the dtb has the av1d turned off by default in this board, opened a support in the git of Orange pi and it has been ignored for months, I fixed mine but could not provide the dtb since I am a noob with this (I am graduating next semester from computer science).

mariobalanica, 2024-12-14:
Hey,

Thanks for your interest.

Orange Pi have not reached out to support any of their newer hardware, therefore no plans from us.

Generally we've accepted board addition pull requests from community, but now I'm reluctant to continue doing so for a few reasons:
- lots of new boards being pushed out with less and less interest from their vendors to actually support them. As you've discovered yourself with that DTB issue, technical debt is now placed entirely on community with essentially nothing to gain from it (unless headaches count), which I don't believe is the right way to approach things. Particularly since we're talking about something as fundamental as boot firmware and OS. 

- some hardware have less than ideal design choices that make them not so great of an option for UEFI usage (e.g. OPi 5 Pro with that odd Ethernet controller unsupported by any upstream OS and obviously no UEFI SNP driver, no SPI NOR for firmware storage and runtime variables, etc.).

- with no access to that hardware, it can be difficult to attest both existing and new functionality. These platforms are usually left behind with no interest to carry them over, yet are still listed as being "Supported" here. People often tend to stop there with research, and get disappointed when the board arrives and turns out to be less supported than other options.

I'm soon going do add support for booting generic mainline Linux distros with Device Tree. **This will be a hard requirement for new platforms here**, because the net goal of this firmware is to make these boards look more like a regular PC capable of booting almost everything you throw at it, as opposed to an embedded device tied to some vendor legacy kernel. Such a compatible Device Tree either has to already exist upstream or somebody (again, preferably the vendor) do the effort to write one and eventually also upstream it so it can be useful beyond this UEFI port.

Wrapping up, I'm not going to say "no" if you really want to work on it, as it doesn't look like a bad board (from what I can tell, it's essentially a smaller OPi 5 Plus and has a proper Realtek NIC), but I'm unable to help with this endeavor other than reviewing your final patches. My suggestion is to look at how enablement for other similar devices has been handled and attempt to replicate that. 

Also, all peripherals currently supported by this firmware and mainline Linux must be confirmed to work, both in UEFI and the OS. Things like M.2 PCIe slots, HDMI output, onboard Wi-Fi, Ethernet, USB, SD/eMMC, RTC, voltage regulators and so on. Of course, each where applicable - Wi-Fi isn't going to work in UEFI but must do in Linux as it has drivers for that. Testing ACPI boot with Windows 11 and latest driver package would also be recommend, but is optional. 

If nobody's able to validate all of that, then I'm afraid the changes cannot be accepted. The alternative is getting the hardware vendor (Orange Pi) to fund this effort.

DummyTitan, 2024-12-14:
I actually have an Orange Pi 5 Max. Currently used as an NAS, so if I need to make any test, I will be able to try them over there if necessary. I currently have no knowledge about UEFI, and any or all information on the internet is really scarce. So I welcome any advice that you can give me about that. Also I am going to try to achieve this If possible. I am going to try to look for all information and yes, the Orange Pi 5 Max is similar to the Orange Pi 5 plus, basically has almost the same things. Only a few things are not the same.

DummyTitan, 2024-12-14:
I just wrote to the Zhao Steven email provided by orange pi to obtain the proper hardware info and dtb. And for God sake I expect them to answer this time.

yswtrue, 2025-01-06:
[rk3588-orangepi-5-max.zip](https://github.com/user-attachments/files/18320816/rk3588-orangepi-5-max.zip)
This is the opi5max dtd file copy from official Debian 12 ROM

yswtrue, 2025-01-06:
https://github.com/orangepi-xunlong/linux-orangepi/blob/orange-pi-5.10-rk35xx/arch/arm64/boot/dts/rockchip/rk3588-orangepi-5-max.dts
and this is the dts file

littlefish12345, 2025-02-11:
I tried to compile a version support OrangePi 5 Max with a configuration mostly copied from OrangePi 5 Plus and the original device tree from latest mainline kernel. I was able to enter configuration menu but only one of the two USB3.0 ports works which is the top one. USB2.0 didn't work at all. I tried to boot archboot, GRUB menu showed up, I selected the option but it ended up stuck with no display output. Before it stuck I saw a log saying exiting bootdevice so I assume that perhaps there is a need to fix the original device tree like you guys did to OrangePi 5 Plus. Maybe I will give it a try in the future. If anyone wants I can probide the janky firmware I just compiled or my configuration.

closed 2025-02-28 by mariobalanica

frozen-cdn, 2026-08-27:
Why is this marked Not Planned?
I am a Orange Pi 5 Max owner also seeking this out

DummyTitan, 2026-08-27:
Just letting you guys know. Someone made a prototype of this. He is called amazing fate. It works, I'm using it. It still has stuff to improve but for those that want to test it it works good enough. I do mention that the project appears to be abandoned so if anyone wants to pursue updating it (since it has an older version) then I would recomend using it as a basis. I don't attempt this cause I have many things to do now compared to prior (got a new job and have many projects at home) and also would not know where to start. But hey give it a try.

DummyTitan, 2026-08-27:
Also in my case the 2 3.0 usb work if the os is the armbian. I tried to make a offshoot armbian that allowed the unfinished using grub but still had all the proper drivers. It worked. That's why currently, even if it is a beta it works enough for my purposes (would prefer full functionality but better then nothing).