edk2-porting/edk2-rk3588 #176: Rock 5 ITX - missing NVME, SATA with Debian 13 Trixie + edk2 v0.12

PinkFreud, 2024-12-28:
After having my Rock 5 ITX running Radxa's bullseye image off sd for a few months, I've decided to go for as mainstream of an installation as I can get: Debian Trixie with kernel 6.12.6.

To that end, I have edk2 v0.12 flashed to SPI flash, and uboot removed from the onboard eMMC.  I found that the Debian installer wasn't finding the nvme drive (a Samsung 990 EVO), so I forged ahead with installing to a micro sd card in the board's internal sd slot.

Now that I have Trixie booting, I did some digging, only to find out that both the onboard SATA and NVME drives are missing.

Dumping dmesg from the Radxa bullseye image, I see the following printed by Radxa's rk-pcie driver:

```
[   13.205485] rk-pcie fe180000.pcie: invalid prsnt-gpios property in node
[   13.205496] rk-pcie fe180000.pcie: Looking up vpcie3v3-supply from device tree
[   13.205703] rk-pcie fe150000.pcie: invalid prsnt-gpios property in node
[   13.205713] rk-pcie fe150000.pcie: Looking up vpcie3v3-supply from device tree
[   13.205797] rk-pcie fe160000.pcie: invalid prsnt-gpios property in node
[   13.205802] rk-pcie fe160000.pcie: Looking up vpcie3v3-supply from device tree
[   13.205897] rk-pcie fe180000.pcie: missing legacy IRQ resource
[   13.205899] rk-pcie fe170000.pcie: invalid prsnt-gpios property in node
[   13.205903] rk-pcie fe170000.pcie: Looking up vpcie3v3-supply from device tree
[   13.205922] rk-pcie fe180000.pcie: IRQ msi not found
[   13.205959] rk-pcie fe180000.pcie: use outband MSI support
[   13.205966] rk-pcie fe180000.pcie: Missing *config* reg space
[   13.205997] rk-pcie fe180000.pcie: host bridge /pcie@fe180000 ranges:
[   13.206028] rk-pcie fe180000.pcie:      err 0x00f3000000..0x00f30fffff -> 0x00f3000000
[   13.206049] rk-pcie fe180000.pcie:       IO 0x00f3100000..0x00f31fffff -> 0x00f3100000
[   13.206073] rk-pcie fe180000.pcie:      MEM 0x00f3200000..0x00f3ffffff -> 0x00f3200000
[   13.206091] rk-pcie fe180000.pcie:      MEM 0x09c0000000..0x09ffffffff -> 0x09c0000000
[   13.206122] rk-pcie fe180000.pcie: Missing *config* reg space
[   13.206167] rk-pcie fe180000.pcie: invalid resource
[   13.211361] rk-pcie fe170000.pcie: missing legacy IRQ resource
[   13.211370] rk-pcie fe170000.pcie: IRQ msi not found
[   13.211379] rk-pcie fe170000.pcie: use outband MSI support
[   13.211382] rk-pcie fe170000.pcie: Missing *config* reg space
[   13.222120] rk-pcie fe170000.pcie: host bridge /pcie@fe170000 ranges:
[   13.222132] rk-pcie fe170000.pcie:      err 0x00f2000000..0x00f20fffff -> 0x00f2000000
[   13.222139] rk-pcie fe170000.pcie:       IO 0x00f2100000..0x00f21fffff -> 0x00f2100000
[   13.222146] rk-pcie fe170000.pcie:      MEM 0x00f2200000..0x00f2ffffff -> 0x00f2200000
[   13.222151] rk-pcie fe170000.pcie:      MEM 0x0980000000..0x09bfffffff -> 0x0980000000
[   13.222176] rk-pcie fe170000.pcie: Missing *config* reg space
[   13.222203] rk-pcie fe160000.pcie: missing legacy IRQ resource
[   13.222206] rk-pcie fe150000.pcie: missing legacy IRQ resource
[   13.222209] rk-pcie fe170000.pcie: invalid resource
[   13.222220] rk-pcie fe160000.pcie: IRQ msi not found
[   13.222224] rk-pcie fe150000.pcie: IRQ msi not found
[   13.222228] rk-pcie fe150000.pcie: use outband MSI support
[   13.222236] rk-pcie fe160000.pcie: use outband MSI support
[   13.222247] rk-pcie fe160000.pcie: Missing *config* reg space
[   13.222250] rk-pcie fe150000.pcie: Missing *config* reg space
[   13.222263] rk-pcie fe150000.pcie: host bridge /pcie@fe150000 ranges:
[   13.222270] rk-pcie fe160000.pcie: host bridge /pcie@fe160000 ranges:
[   13.222286] rk-pcie fe160000.pcie:      err 0x00f1000000..0x00f10fffff -> 0x00f1000000
[   13.222292] rk-pcie fe150000.pcie:      err 0x00f0000000..0x00f00fffff -> 0x00f0000000
[   13.222294] rk-pcie fe160000.pcie:       IO 0x00f1100000..0x00f11fffff -> 0x00f1100000
[   13.222304] rk-pcie fe160000.pcie:      MEM 0x00f1200000..0x00f1ffffff -> 0x00f1200000
[   13.222311] rk-pcie fe160000.pcie:      MEM 0x0940000000..0x097fffffff -> 0x0940000000
[   13.222315] rk-pcie fe150000.pcie:       IO 0x00f0100000..0x00f01fffff -> 0x00f0100000
[   13.222338] rk-pcie fe150000.pcie:      MEM 0x00f0200000..0x00f0ffffff -> 0x00f0200000
[   13.222341] rk-pcie fe160000.pcie: Missing *config* reg space
[   13.222355] rk-pcie fe150000.pcie:      MEM 0x0900000000..0x093fffffff -> 0x0900000000
[   13.222368] rk-pcie fe160000.pcie: invalid resource
[   13.222383] rk-pcie fe150000.pcie: Missing *config* reg space
[   13.222425] rk-pcie fe150000.pcie: invalid resource
[   13.429860] rk-pcie fe160000.pcie: PCIe Linking... LTSSM is 0x0
[   13.429890] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x0
[   13.455407] rk-pcie fe160000.pcie: PCIe Linking... LTSSM is 0x2
[   13.455430] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x210021
[   13.468767] rk-pcie fe180000.pcie: PCIe Link up, LTSSM is 0x130011
[   13.468959] rk-pcie fe180000.pcie: PCI host bridge to bus 0003:30
[   13.482096] rk-pcie fe150000.pcie: PCIe Linking... LTSSM is 0x210023
[   13.485407] rk-pcie fe170000.pcie: PCIe Link up, LTSSM is 0x30011
[   13.485479] rk-pcie fe170000.pcie: PCI host bridge to bus 0002:20
[   13.538740] rk-pcie fe160000.pcie: PCIe Link up, LTSSM is 0x230011
[   13.538811] rk-pcie fe160000.pcie: PCI host bridge to bus 0001:10
[   13.565434] rk-pcie fe150000.pcie: PCIe Link up, LTSSM is 0x230011
[   13.565601] rk-pcie fe150000.pcie: PCI host bridge to bus 0000:00
[   14.114170] rk-pcie fe190000.pcie: invalid prsnt-gpios property in node
[   14.114177] rk-pcie fe190000.pcie: Looking up vpcie3v3-supply from device tree
[   14.114897] rk-pcie fe190000.pcie: missing legacy IRQ resource
[   14.114906] rk-pcie fe190000.pcie: IRQ msi not found
[   14.114937] rk-pcie fe190000.pcie: use outband MSI support
[   14.114940] rk-pcie fe190000.pcie: Missing *config* reg space
[   14.114969] rk-pcie fe190000.pcie: host bridge /pcie@fe190000 ranges:
[   14.114987] rk-pcie fe190000.pcie:      err 0x00f4000000..0x00f40fffff -> 0x00f4000000
[   14.114994] rk-pcie fe190000.pcie:       IO 0x00f4100000..0x00f41fffff -> 0x00f4100000
[   14.115002] rk-pcie fe190000.pcie:      MEM 0x00f4200000..0x00f4ffffff -> 0x00f4200000
[   14.115007] rk-pcie fe190000.pcie:      MEM 0x0a00000000..0x0a3fffffff -> 0x0a00000000
[   14.115036] rk-pcie fe190000.pcie: Missing *config* reg space
[   14.115081] rk-pcie fe190000.pcie: invalid resource
[   14.375579] rk-pcie fe190000.pcie: PCIe Link up, LTSSM is 0x130011
[   14.375799] rk-pcie fe190000.pcie: PCI host bridge to bus 0004:40
```

Indeed, these lines are rather telling:
```
[   13.468959] rk-pcie fe180000.pcie: PCI host bridge to bus 0003:30
[   13.485479] rk-pcie fe170000.pcie: PCI host bridge to bus 0002:20
[   13.538811] rk-pcie fe160000.pcie: PCI host bridge to bus 0001:10
[   13.565601] rk-pcie fe150000.pcie: PCI host bridge to bus 0000:00
[   14.375799] rk-pcie fe190000.pcie: PCI host bridge to bus 0004:40
```

Looking for those addresses in dmesg for 6.12.6:
```
[    0.215728] /pcie@fe180000: Fixed dependency cycle(s) with /pcie@fe180000/legacy-interrupt-controller
[    0.216080] /pcie@fe190000: Fixed dependency cycle(s) with /pcie@fe190000/legacy-interrupt-controller
[    0.229585] /pcie@fe150000: Fixed dependency cycle(s) with /pcie@fe150000/legacy-interrupt-controller
[    0.229740] /pcie@fe170000: Fixed dependency cycle(s) with /pcie@fe170000/legacy-interrupt-controller
[    0.954918] rockchip-dw-pcie a40c00000.pcie: host bridge /pcie@fe180000 ranges:
[    1.164225] rockchip-dw-pcie a41000000.pcie: host bridge /pcie@fe190000 ranges:
[    1.390190] rockchip-dw-pcie a40800000.pcie: host bridge /pcie@fe170000 ranges:
[   18.148042] /pcie@fe150000: Failed to get clk index: 6 ret: -517
```

It appears that fe160000 has gone completely missing (pcie bus 0001:10), and fe150000 (pcie bus 0000:00) is failing for some reason.

According to Radxa's dmesg, nvme is pcie 0000:01:00.0, and ahci 0001:11:00.0, which would explain why those two devices are missing for me under mainline 6.12.6.

See attached for full dmesg logs for both distributions:

[radxa_bullseye_dmesg.txt](https://github.com/user-attachments/files/18264697/radxa_bullseye_dmesg.txt)
[debian_trixie_dmesg.txt](https://github.com/user-attachments/files/18264699/debian_trixie_dmesg.txt)

mariobalanica, 2024-12-28:
Please try this firmware build: https://github.com/edk2-porting/edk2-rk3588/actions/runs/12527262952 and let me know if everything shows up correctly in UEFI and Linux.

PinkFreud, 2024-12-28:
I wish I could tell you, but unfortunately, that build no longer recognized my sd card.

Debug log attached.
[edk2-ek3588-c9563d7.log](https://github.com/user-attachments/files/18267845/edk2-ek3588-c9563d7.log)

mariobalanica, 2024-12-28:
"UEFI Transcend/Samsung GD2S5 7E69C799 SD Device"

PinkFreud, 2024-12-28:
Yes, I know - but the 'debian' boot option no longer appears in the boot menu, and selecting UEFI Transcend/Samsung GD2S5 7E69C799 SD Device in the boot menu just causes it to go right back to the menu rather than booting.

Unfortunately, I think I may have borked this install further by attempting to re-flash v0.12 via rkdeveloptool, as I now see u-boot strings on partition 1.  I'll get this install fixed and then I'll try again.

mariobalanica, 2024-12-28:
Did you try to flash it to SD card this time? I do recall seeing some weird behavior with rkdevtool where flashing one device would overwrite a bit of the others as well. In case you have the firmware on SD/eMMC alongside an OS, care must be taken to flash only part of the image at a specific offset, to avoid overwriting the GPT: https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#updating-the-firmware

Anyway, I do see that PCIe 3x2 (hosting the SATA bridge) did not manage to link up, will look into that and send you another build to try in a few minutes.

mariobalanica, 2024-12-28:
https://github.com/edk2-porting/edk2-rk3588/actions/runs/12530647406

PinkFreud, 2024-12-29:
> Did you try to flash it to SD card this time?

No, it should have been to SPI flash.  Rockchip's documentation surrounding rkdeveloptool is a tad on the thin side, so I'm not precisely sure if it was writing to SPI or another dev.

I'm not sure what I spotted on the card, but once I managed to boot into rescue mode via the debian installer and mount the sd card, I was able to re-flash v0.12 and the `debian` boot option reappeared in the boot menu, so something is definitely off with c9563d7 and the internal sd card.

I'll try your new build momentarily - now that I have a working recovery process for a broken uefi build, I can roll back (relatively) easily.

PinkFreud, 2024-12-29:
Still no go with booting my existing Trixie install with 76d4032.  :(

See attached screenshot for how the boot manager menu appears (slightly wonky due to how it appears over serial!).  Note there's no `debian` option, which should appear first in the list.

I've also attached the serial debug log.

[edk2-ek3588-76d4032.log](https://github.com/user-attachments/files/18268077/edk2-ek3588-76d4032.log)
[image]

PinkFreud, 2024-12-29:
It occurs to me - with v0.12, the 990 EVO does show up in the boot list (even though there's nothing installed on it as of yet), but the installer doesn't see it due to the aforementioned issues.

mariobalanica, 2024-12-29:
So, the SD card does work in v0.12 but not the latest builds? Are you using rkdevtool with the SD card plugged in? I would suggest removing it during flashing just to be sure nothing gets overwritten there.

The only changes in these builds are PCIe-related, there's nothing that could impact the SD boot option.
I'm assuming you're using the RELEASE variant of v0.12, could you try using the DEBUG one from here instead: https://github.com/edk2-porting/edk2-rk3588/actions/runs/12484685405/artifacts/2359912236 to see if that makes a difference?

The SATA controller does show up with 76d4032bb001c2d4b0e0ba058dc20b6adf7ac3d0 now.

mariobalanica, 2024-12-29:
It would also be worth trying the same card in your TS-RDF5A USB card reader.

PinkFreud, 2024-12-29:
> Are you using rkdevtool with the SD card plugged in? I would suggest removing it during flashing just to be sure nothing gets overwritten there.

Yep, that thought occurred to me ... after I tried the initial re-flash with rkdeveloptool.  :(

> It would also be worth trying the same card in your TS-RDF5A USB card reader.

That thought occurred to me as well, but I haven't had any luck there, despite the debian installer image booting successfully via it.

> I'm assuming you're using the RELEASE variant of v0.12, could you try using the DEBUG one from here instead:

Yep, v0.12 RELEASE.  I can certainly try the debug build of that!

mariobalanica, 2024-12-29:
If the installer card boots fine but the one with Trixie installed does not - neither in the onboard slot nor the USB card reader - that suggests an issue with the image on the card. Probably because of that rkdevtool bug. You can actually compare the beginning of the card to the UEFI image and see if it matches with `cmp /dev/mmcblkX rock-5-itx_UEFI_Release_v0.12.img -n 2MB` or a hex editor.

The card does show up and I don't see any bus errors in the logs.

PinkFreud, 2024-12-29:
[edk2-ek3588-v0.12.log](https://github.com/user-attachments/files/18268217/edk2-ek3588-v0.12.log)

Well, that's unexpected.  No debian boot option with the debug build, but flashing v0.12 RELEASE definitely lets me see - and boot from - the sd card in the itx's internal sd card reader.

I guess I'll have to do a fresh installation again.  Sigh.

mariobalanica, 2024-12-29:
In this case, can you flash the RELEASE variant of 76d4032bb001c2d4b0e0ba058dc20b6adf7ac3d0? https://github.com/edk2-porting/edk2-rk3588/actions/runs/12530956203/artifacts/2368511391

PinkFreud, 2024-12-29:
Yep, will do.  I was just about to try a nuke & pave, but I'll give this a try first.

PinkFreud, 2024-12-29:
Nope, no `debian` boot option with 76d4032 RELEASE.

I'm going to try overwriting the beginning of this sd card with zeroes and doing a fresh install - I've used this card for a few things, including troubleshooting a RISC-V board somewhat recently, so I may as well make sure there's nothing extraneous screwing with edk2.  I suppose I'll re-flash 76d4032 DEUBG first.

I'll be gone much of tomorrow and back to work on Monday, so I probably won't have much more time to keep troubleshooting until the new year, but if I have time to do anything between now and then, I'll reply here - so don't stress yourself out with this too much.  :)

PinkFreud, 2024-12-29:
Reinstalled Trixie after wiping the first 1MB of the sd card (I probably should have done a bit more, but oh well, it worked... for now.).

Looks like fe160000 is now throwing the same error as fe150000, so that's a minor improvement:
```
[   21.218426] /pcie@fe160000: Failed to get clk index: 6 ret: -517
[   21.224910] /pcie@fe150000: Failed to get clk index: 6 ret: -517
```

See attached for debug log.
[edk2-rk3588-76d4032.log](https://github.com/user-attachments/files/18268369/edk2-rk3588-76d4032.log)

I just checked the log myself, and noticed:
```
U-Boot SPL board init
U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
Trying to boot from MMC2
```

... crap.  I swear I got rid of u-boot everywhere.  Looks like I missed one.

mariobalanica, 2024-12-29:
Turns out that PCIe on the 5 ITX depends on this change: https://github.com/torvalds/linux/commit/4940071d962827467f7297be3b874c96186ca0b7, which only exists in kernel 6.13-rc1 and newer.

If there's interest, I *could* add a small patch to not depend on that clock (for compatibility's sake with older kernels), but note that RK3588 upstream is a moving target and using the absolute latest kernel is preferred. E.g. native HDMI support is only supported in 6.13+ as well.

> I just checked the log myself, and noticed:
> 
> U-Boot SPL board init
> U-Boot SPL 2017.09-g5f53abfa1e-221223 #zzz (Dec 26 2022 - 09:10:09)
> Trying to boot from MMC2
> ... crap. I swear I got rid of u-boot everywhere. Looks like I missed one.

It's expected to see U-Boot SPL. Documentation only warns about having U-Boot SPL / firmware on multiple devices, which could screw up the boot device detection in UEFI, but with UEFI being on SPI NOR, it's all good.

PinkFreud, 2024-12-30:
> Turns out that PCIe on the 5 ITX depends on this change: torvalds/linux@4940071, which only exists in kernel 6.13-rc1 and newer.

Well, crud.  That's weird, though - both NICs are PCIe, and they appear to work fine (I've only tested one, but both show up w/o error).  Still, that's disappointing to hear, but I suppose not entirely surprising.

> If there's interest, I could add a small patch to not depend on that clock (for compatibility's sake with older kernels), but note that RK3588 upstream is a moving target and using the absolute latest kernel is preferred. E.g. native HDMI output is only supported in 6.13+ as well.

My plan with this machine was to install Trixie and then compile my own 6.13 kernel.  Iirc, 6.12 was announced as the new LTS recently, and as Trixie is on 6.12 now, I don't think it's ever officially going to run 6.13.

Since Trixie is likely to come out around late summer 2025, with Debian 14 another two years behind that, that means we have years to go before Debian officially supports the rk3588 - so it might be a good idea to add that patch, given that I don't see the Rockchip platform significantly decreasing in popularity in the next few years (it's hardly the most widely used thing *now*, but as a fairly cheap and capable ARM platform with active development to bring mainline support, I suspect popularity will increase!).

> It's expected to see U-Boot SPL. Documentation only warns about having U-Boot SPL / firmware on multiple devices, which could screw up the boot device detection in UEFI, but with UEFI being on SPI NOR, it's all good.

*Whew*, ok, I won't worry about that, then.  Thank you!

PinkFreud, 2024-12-30:
[edk2-rk3588-b789894.log](https://github.com/user-attachments/files/18277269/edk2-rk3588-b789894.log)

We're right back to the debian installation not being detected on the internal sd card.  Doing a diff between the earlier edk2-rk3588-76d4032.log and this log doesn't make any of the differences immediately stand out - I see different load addresses for some of the modules, but that seems normal for different builds.

I really don't want to have to reinstall Trixie every time I flash firmware.

mariobalanica, 2024-12-30:
Have you removed the SD card while flashing the firmware?

> and this log doesn't make any of the differences immediately stand out

The changes are entirely for Linux this time.

PinkFreud, 2024-12-30:
> Have you removed the SD card while flashing the firmware?

That's a bit hard to do, since I'm using flashcp to write the firmware to the spi chip from the OS, which is running from that same sd card.  Unlike rkdeveloptool, though, I'm reasonably certain that flashcp isn't modifying the sd card!

mariobalanica, 2024-12-30:
Is flashcp erasing the whole memory before or something like that? Maybe use dd instead like described here: https://github.com/edk2-porting/edk2-rk3588?tab=readme-ov-file#updating-the-firmware

At worst I would expect the Debian option to disappear (if flash gets fully erased), but the SD boot device option should still remain bootable. If it doesn't boot in an USB card reader either, then you can be sure the card is getting corrupted by something else and it's not related to the firmware version.

PinkFreud, 2024-12-30:
Hmm, I am having it erase flash (flashcp -A) before writing the new image - that's typically the safest thing to do, iirc.

I wouldn't think that would affect it's ability to find the debian boot option - those options are detected dynamically at boot, correct?

PinkFreud, 2024-12-30:
Well, one bit of good news - the Trixie installer's rescue mode now detects the nvme drive, so that patch does appear to work!

Based on what you said above... I'm guessing that not all options are detected dynamically, and are instead written into the flash at some point.  I have a way to test this, at least, and I'll let you know if it resolves my issue with the boot option disappearing.

PinkFreud, 2024-12-30:
> I wouldn't think that would affect it's ability to find the debian boot option - those options are detected dynamically at boot, correct?

Well, there it is - I can re-add it via the boot maintenance menu.  Apparently that one is NOT detected dynamically at boot.  I wasn't expecting that.

Solution: Don't perform full eraseflash when upgrading edk2.  Damn - I wasn't expecting that.

mariobalanica, 2024-12-30:
> Well, one bit of good news - the Trixie installer's rescue mode now detects the nvme drive, so that patch does appear to work!
Nice! Can you confirm that the SATA controller appears as well? 

> Based on what you said above... I'm guessing that not all options are detected dynamically, and are instead written into the flash at some point. I have a way to test this, at least, and I'll let you know if it resolves my issue with the boot option disappearing.

Right, OS-specific boot options are written by grub-install (I believe) to the variable store on flash, there's no way to discover them automatically. Erase the variable store and they're gone, along with other settings you might have made.

Regardless of that, the SD Device option should still allow you to boot into the Grub menu.

PinkFreud, 2024-12-30:
> Nice! Can you confirm that the SATA controller appears as well?

```
[    1.438374] ahci 0001:11:00.0: version 3.0
[    1.438388] ahci 0001:11:00.0: enabling device (0000 -> 0002)
[    1.438585] ahci 0001:11:00.0: SSS flag set, parallel bus scan disabled
[    1.448858] ahci 0001:11:00.0: AHCI vers 0001.0301, 32 command slots, 6 Gbps, SATA mode
[    1.448868] ahci 0001:11:00.0: 20/24 ports implemented (port mask 0xffff0f)
[    1.448871] ahci 0001:11:00.0: flags: 64bit ncq sntf stag pm led only pio sxs deso sadm sds apst 
[    1.452064] scsi host0: ahci
[    1.452292] scsi host1: ahci
[    1.452397] scsi host2: ahci
[    1.452499] scsi host3: ahci
[    1.452601] scsi host4: ahci
[    1.452702] scsi host5: ahci
[    1.452810] scsi host6: ahci
[    1.452906] scsi host7: ahci
[    1.453008] scsi host8: ahci
[    1.453110] scsi host9: ahci
[    1.453210] scsi host10: ahci
[    1.453317] scsi host11: ahci
[    1.453417] scsi host12: ahci
[    1.453515] scsi host13: ahci
[    1.453623] scsi host14: ahci
[    1.453738] scsi host15: ahci
[    1.453838] scsi host16: ahci
[    1.454039] scsi host17: ahci
[    1.454254] scsi host18: ahci
[    1.454362] scsi host19: ahci
[    1.454460] scsi host20: ahci
[    1.454567] scsi host21: ahci
[    1.454665] scsi host22: ahci
[    1.454767] scsi host23: ahci
```

Well, that's a good sign.  lspci also shows:
```
0000:00:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0000:01:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller PM9C1a (DRAM-less)
0001:10:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0001:11:00.0 SATA controller: ASMedia Technology Inc. ASM1164 Serial ATA AHCI Controller (rev 02)
0002:20:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0002:21:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8852BE PCIe 802.11ax Wireless Network Controller
0003:30:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0003:31:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
0004:40:00.0 PCI bridge: Rockchip Electronics Co., Ltd RK3588 (rev 01)
0004:41:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8125 2.5GbE Controller (rev 05)
```

> Right, OS-specific boot options are written by grub-install (I believe) to the variable store on flash

Indeed they are - I was able to fix the boot menu (I had two `debian` options after my initial fix when I added the sd card entry in the efi boot manager) by re-running grub-install.

PinkFreud, 2024-12-30:
Now, for one last issue:
During this process, I've found that my usb keyboard isn't working after booting for some reason, though it does work in the installer, and it seems to work if I unplug it and reconnect it to a usb port (the same usb port it used during boot seems to work here).  I've been able to work around it by using the serial console or ssh (depending on whether I was booted into my installed OS at the time), but I really do need to have a working console!

lsusb shows:
```
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 002: ID 1a40:0101 Terminus Technology Inc. Hub
Bus 002 Device 003: ID 13d3:3570 IMC Networks Bluetooth Radio
Bus 002 Device 004: ID 05ac:1001 Apple, Inc. Keyboard Hub [ALPS]
Bus 002 Device 005: ID 05ac:0201 Apple, Inc. USB Keyboard [Alps or Logitech, M2452]
Bus 002 Device 006: ID 046d:c016 Logitech, Inc. Optical Wheel Mouse
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 005 Device 002: ID 05e3:0610 Genesys Logic, Inc. Hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 006 Device 002: ID 05e3:0620 Genesys Logic, Inc. GL3523 Hub
Bus 006 Device 003: ID 8564:4000 Transcend Information, Inc. microSD/SD/CF UHS-II Card Reader [RDF8, RDF9]
```

I'm not entirely certain this is edk2-related, but I'm finding myself lacking other explanations for it not working until reconnection.

mariobalanica, 2024-12-30:
The USB keyboard issues are in UEFI or the OS?

PinkFreud, 2024-12-30:
OS, oddly enough, but I haven't had them appear anywhere but here.  If you think it's not UEFI-related, then I'll keep digging to see if I can figure out what's going wrong.

mariobalanica, 2024-12-30:
Hmm, I have not encountered this on my ROCK 5B. 

If you compare UEFI + official Debian to Radxa's images, do note that the mainline/generic kernel is quite different from the one shipped in vendor distros. Many drivers had to be reworked for upstream inclusion and may currently lack code that deals with certain edge cases. It could also just be some omission in the device tree around regulators.

Since you said that the keyboard works fine while in UEFI, it's most certainly kernel/DTB-related. UEFI resets these peripherals before handing-off to the OS, and I'm pretty sure Linux will also do so before trying to access the hardware.

At any rate, I'm going to close this one given that PCIe is now fully functional. Thanks for testing!

closed 2024-12-30 by mariobalanica