edk2-porting/edk2-rk3588 #177: Support for Radxa CM5

Entity1011, 2024-12-31:
Is there any chance the CM5 can be supported? From what I found, it's basically a 5B, but on a smaller board. With some elbow grease, I've managed to make the EFI boot, but USB didn't work. I managed to hook up a UART interface so I could navigate to a bootable OS, but no matter what distro I try to run, it always prints: 
```
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Exiting boot services...
```
and then becomes completely unresponsive.
Am I doing something wrong? Is there a way for this to be fixed or for me to fix it? This is pretty much my first "real" SBC, so I don't know a lot about it yet

peltho, 2025-01-11:
If CM5 is also based on rk3582 I would be interested for my 5C lite as well 🙏 
_(keeping an eye on this)_

peltho, 2025-01-22:
My bad I was thinking about the CM5 lite.

closed 2025-02-28 by mariobalanica

kwankiu, 2025-04-03:
For anyone whose still looking for this, check
https://github.com/edk2-porting/edk2-rk3588/pull/186