edk2-porting/edk2-rk3588 #178: Fedora ISO fails to boot in ACPI mode on Orange Pi 5 Plus

Integral-Tech, 2025-01-02:
I found Fedora Server 41 ISO failed to boot in ACPI mode on Orange Pi 5 Plus. On the contrary, in ACPI mode it boots successfully.

- Kernel version: 6.11.4
- edk2 version: 0.12

[image]

In the screenshot I am trying to boot with Ventoy 1.0.99, but without ventoy it still fails.

mariobalanica, 2025-01-02:
Use the latest artifact build from here: https://github.com/edk2-porting/edk2-rk3588/actions/workflows/build.yml

I'm guessing you're booting from USB.

Integral-Tech, 2025-01-04:
@mariobalanica It works, thanks you :)

Integral-Tech, 2025-03-29:
@mariobalanica It seems that this issue occurs again in the latest build :(

mariobalanica, 2025-03-29:
Yeah it's caused by 48d52ef954551456b63170443d1bcfeb6f2ef841.

The device tree included in the kernel for this board is *still* incomplete...

Integral-Tech, 2025-03-29:
@mariobalanica Is it because enabling USB port triggers other problems?

mariobalanica, 2025-03-29:
No. We provide additional device-tree fixups for boards that have not received much attention upstream and lack the definitions for things like audio, some USB ports, HDMI and other things. The fixup file was removed in the commit above because it appeared that those changes had been merged upstream and were no longer needed. However, it turns out that at least the definition for the USB3 ports is still missing.

It will be fixed in the next firmware version after kernel 6.15-rc1 gets released.

Integral-Tech, 2025-04-03:
@mariobalanica Therefore what does commit 48d52ef954551456b63170443d1bcfeb6f2ef841 do?

Integral-Tech, 2025-04-03:
@mariobalanica Which version of Linux kernel are you booting?

mariobalanica, 2025-04-03:
Fedora Rawhide with the latest "6.15-rc0" build. But it doesn't matter because as far as USB support is concerned, there haven't been any changes between these firmware versions, so older kernels should continue to work.

Integral-Tech, 2025-04-03:
> Fedora Rawhide with the latest "6.15-rc0" build. But it doesn't matter because as far as USB support is concerned, there haven't been any changes between these firmware versions, so older kernels should continue to work.

But the device tree might change between different kernel versions

mariobalanica, 2025-04-03:
Can you confirm that downgrading the firmware actually fixes the issue?

mariobalanica, 2025-04-03:
Ok, I have just tried booting the Fedora Server 41 aarch64 raw image from an USB3 drive - no problems with the latest firmware.

> But the upstream device tree might change between different kernel versions.

It has changed and will continue to do so given that RK3588 porting is still work-in-progress. However, there have not been any breaking changes so far. It's reasonably backwards/forwards compatible.

closed 2025-04-07 by mariobalanica