edk2-porting/edk2-rk3588 #180: rockpi 5b+ doesn't work in device tree mode

simper, 2025-01-08:
Installed 0.12.2 into SPI for rockpi 5b+ board:
- if selecting ACPI mode, it can boot armbian with 6.6 kernel;
- Then install the 6.1.75 vendor kernel under ambian, copy the rk3588 5b+ base dtb file from relevant kernel folder into dtb/base folder, under the boot partition, switch to device tree mode after the reboot.
The boot options are all available in grub boot menu, but if choosing 6.1.75 vendor kernel to boot, the board goes without any video output after a while, then just stuck...

mariobalanica, 2025-01-09:
1. Have you enabled the `Support DTB override & overlays` option?
2. Does the DTB have the correct name? I.e. `rk3588-rock-5bp.dtb` for the 5B+.

If it still doesn't appear to boot, then you'd need an UART adapter to see what's actually going on.

simper, 2025-01-10:
> 1. Have you enabled the `Support DTB override & overlays` option?
> 2. Does the DTB have the correct name? I.e. `rk3588-rock-5bp.dtb` for the 5B+.
> 
> If it still doesn't appear to boot, then you'd need an UART adapter to see what's actually going on.

yes, both points are checked, no signal from hdmi port after choosing grub menu kernel, even tried to install a compiled 6.1.84 kernel from radxa bsp build tool, with the 5b plus dtb file put into relevant folder, it seems the fan is working (under ACPI mode, fan is not triggered after booting the kernel), I suspect the GPU user space driver issue? 
In Armbian UEFI apt repository, mesa version is very low as 22.x.x, while in radxa shipped images, it is 24.x.x onwards already.

simper, 2025-01-12:
Most probably a ed2k UEFI firmware issue specific for rock 5b plus, since bredOS using UEFI doesn't boot into desktop environment as well, after reading device tree information, the screen die.

closed 2025-02-28 by mariobalanica